Koenkk/zigbee2mqtt #23763: [New device support]: Direct Signs Controller DS-Z-001DE

wolfganghuse, 2024-08-28:
### Link

https://www.directsigns.cz/web/content/4784?unique=50c2521e1aa8aa4f5b534ef1cb6c17edf8cc77e8&download=true

### Database entry

0xa4c138168feb4d66

### Comments

Find all details reported by Zigbee-Adapter in iobroker:
modelZigbee:DS-Z-001DE
type:Router
nwk:38719
manuf id:4687
manufacturer:DIRECTSIGNS
power:Mains (single phase)
app version:1
hard version:2
zcl version:3
stack version:2
date code:20230911
build:20451203
interviewed:false
configured:false
endpoint:11
profile:260
input clusters:genBasic (0)
genIdentify (3)
genGroups (4)
genScenes (5)
genOnOff (6)
genLevelCtrl (8)
lightingColorCtrl (768)
touchlink (4096)
output clusters:genOta (25)
endpoint:242

### External definition

```shell
dont have one
```

### What does/doesn't work with the external definition?

dont know

Si3g3l, 2024-12-18:
Hello, is there a .js for the device that already provides the function?

murmlos, 2025-01-02:
It works in ZHA - and kinda works in z2m. I've only tested with white lights dont know about the rgb.

IEEE Address    0xa4c138aa15c6f8c8

const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['DS-Z-001DE'],
    model: 'DS-Z-001DE',
    vendor: 'DIRECTSIGNS',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[158,500]}})],
    meta: {},
};

module.exports = definition;

Rico81, 2025-01-25:
I'd like to add a device file to the z2m repository on githuib, but I can't.
If there's anyone around that can, please create a "DS-Z-001DE.ts" file out of the code below.

`const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['DS-Z-001DE'],
    model: 'DS-Z-001DE',
    vendor: 'DIRECTSIGNS',
    description: 'Automatically generated definition',
    extend: [light()],
    meta: {},
};

module.exports = definition;`

closed 2025-04-04 by github-actions[bot]