Koenkk/zigbee2mqtt #23767: Aqara Z1 Pro ZNQBKG45LM decoupled mode

DrEvil35, 2024-08-28:
### What happened?

Hello! I have problems with this switch. Only the top button goes into decoupled mode. The rest of the buttons work with a connection to the relay and do not respond to attempts to change the mode from Z2M.It was also noticed that in some cases changing the middle button mode has an impact on the top button mode.
[image]

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.3

### Adapter

Ember

### Setup

docker image Z2M on  rasperryPi4

MadMirza, 2024-09-02:
+1
I have the same issue as well, was working well in the past version of z2m, after updating to 1.39.1, the switch has improper behavior. Top Button works for decoupling, Middle and Bottom can't be decoupled. Now, Changing Middle button operation mode changes the top button operation mode as well.
Also, could someone help me to understand the difference between, Lock relay vs Operation Mode, seems to be doing the same thing in terms of control :)

z2m version 1.39.1
Coordinator: Zigbee Dongle-P, Firmware version 20221226
Home Assistant: 2024.7.4

[image]

Logs when trying to change the operating mode attached.

MadMirza, 2024-09-02:
@DrEvil35 Just updated to z2m 1.40.0, seems to have solved the decoupling problem. Now in decouple mode, single_top, single_center, single_bottom works.

But https://www.zigbee2mqtt.io/devices/ZNQBKG45LM.html
shows the top, middle and bottom gangs has got more exposes which are not seen when tried on.

closed 2024-12-20 by github-actions[bot]