Koenkk/zigbee2mqtt #23768: Sonoff Zigbee 3.0 USB-Dongle Plus E (EFR32MG21) unable to start

wiedikerli, 2024-08-28:
### What happened?

Hi!
Dongle in question is this one: https://de.aliexpress.com/item/1005006316807136.html
I tried to connect to my new ZBDongle-E but i get this error message:
``` log
Attaching to zigbee2mqtt
zigbee2mqtt  | Using '/app/data' as data directory
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt  | [2024-08-29 00:10:50] info:      z2m: Logging to console, file (filename: log.log)
zigbee2mqtt  | [2024-08-29 00:10:50] debug:     z2m: Loaded state from file /app/data/state.json
zigbee2mqtt  | [2024-08-29 00:10:50] info:      z2m: Starting Zigbee2MQTT version 1.39.1 (commit #e132316)
zigbee2mqtt  | [2024-08-29 00:10:50] info:      z2m: Starting zigbee-herdsman (0.55.3)
zigbee2mqtt  | [2024-08-29 00:10:50] debug:     z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":5716,\"extendedPanID\":[69,26,151,253,94,102,234,233],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\"},\"adapter\":{\"concurrent\":null,\"delay\":null,\"disableLED\":false}}"'
zigbee2mqtt  | [2024-08-29 00:10:51] debug:     zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":5716,"extendedPanID":[69,26,151,253,94,102,234,233],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
zigbee2mqtt  | [2024-08-29 00:10:51] info:      zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
zigbee2mqtt  | [2024-08-29 00:10:51] info:      zh:zstack:znp: Serialport opened
zigbee2mqtt  | [2024-08-29 00:10:51] debug:     zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
zigbee2mqtt  | [2024-08-29 00:10:51] debug:     zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
zigbee2mqtt  | [2024-08-29 00:10:51] info:      zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
zigbee2mqtt  | [2024-08-29 00:10:51] debug:     zh:zstack:unpi:writer: --> buffer [239]
zigbee2mqtt  | [2024-08-29 00:10:52] debug:     zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
zigbee2mqtt  | [2024-08-29 00:10:52] debug:     zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
zigbee2mqtt  | [2024-08-29 00:10:53] info:      zh:zstack:znp: Skip bootloader for CC2652/CC1352
zigbee2mqtt  | [2024-08-29 00:10:53] debug:     zh:zstack:unpi:parser: <-- [17]
zigbee2mqtt  | [2024-08-29 00:10:53] debug:     zh:zstack:unpi:parser: --- parseNext [17]
zigbee2mqtt  | [2024-08-29 00:10:53] debug:     zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
zigbee2mqtt  | [2024-08-29 00:10:53] debug:     zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
zigbee2mqtt  | [2024-08-29 00:10:54] debug:     zh:zstack:unpi:parser: <-- [26,193,2,2,155,123,126]
zigbee2mqtt  | [2024-08-29 00:10:54] debug:     zh:zstack:unpi:parser: --- parseNext [17,26,193,2,2,155,123,126]
zigbee2mqtt  | [2024-08-29 00:10:59] debug:     zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
zigbee2mqtt  | [2024-08-29 00:10:59] debug:     zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
zigbee2mqtt  | [2024-08-29 00:11:05] debug:     zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
zigbee2mqtt  | [2024-08-29 00:11:05] debug:     zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
zigbee2mqtt  | [2024-08-29 00:11:11] error:     z2m: Error while starting zigbee-herdsman
zigbee2mqtt  | [2024-08-29 00:11:11] error:     z2m: Failed to start zigbee
zigbee2mqtt  | [2024-08-29 00:11:11] error:     z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
zigbee2mqtt  | [2024-08-29 00:11:11] error:     z2m: Exiting...
zigbee2mqtt  | [2024-08-29 00:11:11] error:     z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
zigbee2mqtt  |     at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:119:27)
zigbee2mqtt  |     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
zigbee2mqtt  |     at Zigbee.start (/app/lib/zigbee.ts:63:27)
zigbee2mqtt  |     at Controller.start (/app/lib/controller.ts:139:27)
zigbee2mqtt  |     at start (/app/index.js:154:5)
zigbee2mqtt exited with code 0
```

The configuration file is the default one created after starting the docker image. Here is my compose file:
``` log
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: unless-stopped
    volumes:
        - ./zigbee2mqtt/data:/app/data
        - /run/udev:/run/udev:ro
    ports:
        - 8124:8080
    environment:
        - TZ=Europe/Berlin
    devices:
        - /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_ce8a5334c112ef1186336db8bf9df066-if00-port0:/dev/ttyACM0
``` 
I tried flashing it with the webflasher (https://darkxst.github.io/silabs-firmware-builder/) which could not connect. Then i tried flashing it with universal-silabs-flasher. Which did not work as well.
Firmware: ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl (i hope i found the correct one?!?).
Probe output:
``` log
universal-silabs-flasher --device /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_ce8a5334c112ef1186336db8bf9df066-if00-port0 probe
2024-08-28 23:28:13.737 raz401 universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-08-28 23:28:15.748 raz401 universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2024-08-28 23:28:20.066 raz401 universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 115200 baud
2024-08-28 23:28:24.388 raz401 universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 230400 baud
2024-08-28 23:28:28.716 raz401 universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2024-08-28 23:28:29.986 raz401 universal_silabs_flasher.flasher INFO Detected ApplicationType.EZSP, version '6.10.3.0 build 297' (6.10.3.0.297) at 115200 baudrate (bootloader baudrate None)
2024-08-28 23:28:29.987 raz401 universal_silabs_flasher.flash INFO Dumping EmberZNet Config
CONFIG_PACKET_BUFFER_COUNT=250
CONFIG_NEIGHBOR_TABLE_SIZE=26
CONFIG_APS_UNICAST_MESSAGE_COUNT=32
CONFIG_BINDING_TABLE_SIZE=32
CONFIG_ADDRESS_TABLE_SIZE=32
CONFIG_MULTICAST_TABLE_SIZE=8
CONFIG_ROUTE_TABLE_SIZE=16
CONFIG_DISCOVERY_TABLE_SIZE=8
CONFIG_STACK_PROFILE=0
CONFIG_SECURITY_LEVEL=5
CONFIG_MAX_HOPS=30
CONFIG_MAX_END_DEVICE_CHILDREN=32
CONFIG_INDIRECT_TRANSMISSION_TIMEOUT=3000
CONFIG_END_DEVICE_POLL_TIMEOUT=8
CONFIG_TX_POWER_MODE=0
CONFIG_DISABLE_RELAY=0
CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE=0
CONFIG_SOURCE_ROUTE_TABLE_SIZE=200
CONFIG_FRAGMENT_WINDOW_SIZE=1
CONFIG_FRAGMENT_DELAY_MS=0
CONFIG_KEY_TABLE_SIZE=12
CONFIG_APS_ACK_TIMEOUT=1600
CONFIG_ACTIVE_SCAN_DURATION=3
CONFIG_END_DEVICE_BIND_TIMEOUT=60
CONFIG_PAN_ID_CONFLICT_REPORT_THRESHOLD=2
CONFIG_REQUEST_KEY_TIMEOUT=0
CONFIG_CERTIFICATE_TABLE_SIZE=0
CONFIG_APPLICATION_ZDO_FLAGS=0
CONFIG_BROADCAST_TABLE_SIZE=35
CONFIG_MAC_FILTER_TABLE_SIZE=0
CONFIG_SUPPORTED_NETWORKS=1
CONFIG_SEND_MULTICASTS_TO_SLEEPY_ADDRESS=0
CONFIG_ZLL_GROUP_ADDRESSES=1
CONFIG_ZLL_RSSI_THRESHOLD=128
CONFIG_MTORR_FLOW_CONTROL=1
CONFIG_RETRY_QUEUE_SIZE=16
CONFIG_NEW_BROADCAST_ENTRY_THRESHOLD=29
CONFIG_TRANSIENT_KEY_TIMEOUT_S=300
CONFIG_BROADCAST_MIN_ACKS_NEEDED=255
CONFIG_TC_REJOINS_USING_WELL_KNOWN_KEY_TIMEOUT_S=300
CONFIG_CTUNE_VALUE=133
CONFIG_ASSUME_TC_CONCENTRATOR_TYPE=1
```

### Zigbee2MQTT version

latest

### Adapter firmware version

6.10.3.0 build 297 (not sure)

### Adapter

i dont know

### Setup

Docker on Raspberry pi 4

clockbrain, 2024-08-29:
Did you include the following in your configuration as per the guide [https://www.zigbee2mqtt.io/guide/adapters/emberznet.html](https://www.zigbee2mqtt.io/guide/adapters/emberznet.html)
 
```
serial:
    adapter: ember

```

Yukyniss, 2024-08-30:
Zigbee-Dongle-E uses ember (higher version) or ezsp (lower version). From the log, I see that z2m uses the default zstack to start it. Please specify whether the adapter is ember or ezsp in your configuration.

wiedikerli, 2024-08-30:
with the adapter set to ezsp it works! Thanks a lot. With ember it didnt but i guess the reason is older firmware, right?

Fywqdk, 2024-09-19:
This finally did it for me too. Same story; Didn't work with ember but works with ezsp. 

It's weird though: Brand new dongle and thought I had flashed the new 7.4.3 firmware on it, but the webflasher didn't work. I did manage to flash in secureCRT but it didn't tell me that it was actually successful in doing so it just finished.

didius62, 2024-10-06:
hello i confirm that secureCRT don't indicate the successfull update !!

the zdongle-e has 2 uart variant :
The “ZBDongle-E” adapter variant is known to use either a WCH CH9102F USB-to-Serial converter chip or Silabs CP2102N USB-to-UART bridge chip so if using Windows OS or Mac OS then need to install CH9102/CH343 device drivers from WCH and CP210x USB to UART Bridge VCP Drivers from Silicon Labs:
[http://www.wch-ic.com/products/CH9102.html 421](http://www.wch-ic.com/products/CH9102.html)
[https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers 31](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers)

we don't know when we buy !!!

i have another question about the flash Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl , i flashed a Zdongle-E as a router but i am not able to associate the router under zigbee2mqtt.
my question is , is this fw is compatible with a zdongle-E as a coordinator with a fw 7.4.3.0 ?

i tried with Ezsp and Ember  driver , same bad result !!!

the green and red leds on the zdongle-e are fixed on ! i don't know the signification !

Any help ?

regards

chevdor, 2024-11-19:
I ran into the issue as well and I can confirm, I got a new dongle and the driver that I had to use on MacOS is:
` CH9102/CH343 device drivers from WCH`. A reboot later....

After that, picking:
[image]

The web updater could connect and confirm that I got a ancient firmware:
[image]

and the upgrade worked:
[image]

closed 2025-01-27 by github-actions[bot]