Koenkk/zigbee2mqtt #23770: Motion vs Occupancy class on devices

McGiverGim, 2024-08-29:
### What happened?

I've observed that the "icon" in Home Assistant for some motion detection devices (like the RTCGQ11LM from Aqara) is wrong. There are two "similar" [classes ](https://www.home-assistant.io/integrations/binary_sensor/#device-class) for this kind of devices, `motion` and `occupancy`. For a motion detection device, like Aqara, the correct one must be the first, but it's using the second one.
Here you have a sample:
[image]

The first one is the Aqara with "occupancy", and the second one has the "motion". I don't know when it changed but some time ago it was ok.

### What did you expect to happen?

To have the correct icon based on the device.

### How to reproduce it (minimal and precise)

Simply look at the icon in Home Assistant for this kind of motion devices.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20230507

### Adapter

egony ebyte

### Setup

Addon Home Assistant on HAOS on Raspberry Pi

McGiverGim, 2024-09-30:
I've found the culprit of that: https://github.com/Koenkk/zigbee2mqtt/pull/22896

It seems it was an intended change. So I suppose it will not be a fix for it.

AndreKR, 2024-10-06:
Just to be sure that you know: When you open the entity settings there is a "Show as" setting that you can switch to "Motion":

[image]

This will bring back the running man.

McGiverGim, 2024-10-06:
Yes, that's the workaround I used, but I continue thinking the correct fix is to change it at the beginning.

ste-fle, 2024-12-13:
I also noticed this issue when I added a presence sensor to a room that already had a motion sensor and suddenly HA showed presence information twice. 
I now changed this manually as suggested above but in my opinion the current motion sensor implementation is wrong. 
Presence information could be an optional sensor based on the available timeout setting but the main sensor should display motion again.

ste-fle, 2025-02-13:
Still relevant

closed 2025-04-23 by github-actions[bot]