Koenkk/zigbee2mqtt #23774: Z2M started freezing after upgrading to 20240710 coordinator FW with SLZB-06

rodrigogbs, 2024-08-29:
### What happened?

After upgrading to 20240710 coordinator FW with SLZB-06 I realized that 2 erros started happening in the logs:

```
- z2m: MQTT error: Keepalive timeout
- z2m: MQTT error: write EPIPE
```

Sometimes Z2M recover by itself, but othr times no and it is necessary to restart the Dongle and Z2M.

I upgraded the FW in 08/27/24 and since then I faced 3 episodes of freezing and needed to restart the Dongle and Z2M.

This is a separeted zigbee network just for sensors, I have 19 presence sensors and 1 energy monitor sensor connected to this Z2M installation.

### What did you expect to happen?

No erros and a stable execution.

### How to reproduce it (minimal and precise)

1. Upgrade the FW of the dongle to 20240710 coordinator
2. Start Z2M and leave it running for some days

### Zigbee2MQTT version

1.39.1-1

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Add-on on HASS OS running in a x86-64 mini pc

### Debug log

[log.log](https://github.com/user-attachments/files/16799001/log.log)
[log1.log](https://github.com/user-attachments/files/16799003/log1.log)

najjadas, 2024-09-02:
I have the same issue 😕

rodrigogbs, 2024-09-08:
Just to contributed a little bit more, I have 2 installations, onde with a Sonoff Dongle-P, where the FW 20240710 is working perfectly, the problem is just with the SLZB-06 adapter with the new FW.

ounyai, 2024-11-19:
I can confirm I also am having the same issues since upgrading to 20240710. I have around 45 sensors; and was not experiencing issues until the firmware upgrade. I will try rolling back also, thanks.

closed 2025-01-27 by github-actions[bot]