Koenkk/zigbee2mqtt #23775: CC1352P2_CC2652P_launchpad_coordinator doesn't work with rtscts: true on Sonoff zbdongle-p regardless of dip switch position

evelant, 2024-08-29:
### What happened?

I installed the latest coordinator firmware from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin on my zbdongle-p. I enabled the hardware flow control dip switch on the stick. I tried to start z2m with `rtscts: true` but it fails to start with no communication from the stick. Turning off hardware flow control works.

### What did you expect to happen?

Hardware flow control should work with the zbdongle-p per the table on https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

### How to reproduce it (minimal and precise)

See above -- install latest firmware, turn dip switch on, set `rtscts: true`, and observe z2m fail to start

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20240710

### Adapter

Sonoff zbdongle-p

### Setup

Hass addon, rpi5

Koenkk, 2024-08-29:
The firmware doesn't support hardware flow control, that's why it doesn't work.

evelant, 2024-08-29:
Ah well it makes sense then 😄 I thought that the firmware supported it since it's listed in the table.

closed 2024-08-29 by evelant