Koenkk/zigbee2mqtt #23785: Z2M not Starting with Sonoff Dongle-P

xBorsti, 2024-08-31:
### What happened?

on Homeassistant, Zigbee2MQTT-addon stopped working on my Sonoff Dongle-E. 
So i got an Sonoff Dongle-P, deleted all the old Z2M files and diretories, uninstalled and started new. 

First i updated the Dongle-P to the latest coordinator firmware (20240710) with the bsl-tool without any error. then tried to start Z2M

my configuration.yaml: 
`homeassistant: true
advanced:
  log_level: debug 
  network_key:
    - 151
    - 149
    - 208
    - 212
    - 177
    - 193
    - 69
    - 166
    - 42
    - 145
    - 11
    - 17
    - 42
    - 72
    - 11
    - 154
  pan_id: 30670
  ext_pan_id:
    - 126
    - 61
    - 225
    - 49
    - 32
    - 151
    - 1
    - 31
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: XXXX
serial:
  port: /dev/ttyUSB0
#  baudrate: 115200
  adapter: zstack
frontend:
  port: 8099
`

the log- output: 
`[16:54:10] INFO: Preparing to start...
[16:54:11] INFO: Socat not enabled
[16:54:13] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-08-31 16:54:20] info: 	z2m: Logging to console, file (filename: log.txt)
[2024-08-31 16:54:20] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2024-08-31 16:54:20] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
[2024-08-31 16:54:20] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-31 16:54:20] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":30670,\"extendedPanID\":[126,61,225,49,32,151,1,31],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"concurrent\":null,\"delay\":null,\"disableLED\":false}}"'
[2024-08-31 16:54:21] debug: 	zh:adapter: Failed to validate path: 'Error: spawn udevadm ENOENT'
[2024-08-31 16:54:21] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":30670,"extendedPanID":[126,61,225,49,32,151,1,31],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
[2024-08-31 16:54:21] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-08-31 16:54:21] info: 	zh:zstack:znp: Serialport opened
[2024-08-31 16:54:21] debug: 	zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-08-31 16:54:21] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-08-31 16:54:21] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-08-31 16:54:21] debug: 	zh:zstack:unpi:writer: --> buffer [239]
[2024-08-31 16:54:22] debug: 	zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-08-31 16:54:22] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-08-31 16:54:23] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-08-31 16:54:23] debug: 	zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-08-31 16:54:23] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-08-31 16:54:29] debug: 	zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-08-31 16:54:29] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-08-31 16:54:35] debug: 	zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-08-31 16:54:35] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-08-31 16:54:35] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,1,89,6,61]
[2024-08-31 16:54:35] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,1,89,6,61]
[2024-08-31 16:54:35] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
[2024-08-31 16:54:35] debug: 	zh:zstack:znp: SRSP: <-- SYS - ping - {"capabilities":1625}
[2024-08-31 16:54:35] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-08-31 16:54:35] debug: 	zh:zstack:znp: SREQ: --> SYS - version - {}
[2024-08-31 16:54:35] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,2,35]
[2024-08-31 16:54:41] debug: 	zh:zstack: Failed to get zStack version, assuming 1.2
[2024-08-31 16:54:41] debug: 	zh:zstack: Adapter concurrent: 2
[2024-08-31 16:54:41] debug: 	zh:zstack: Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":0,"maintrel":0,"revision":""})
[2024-08-31 16:54:41] debug: 	zh:adapter:zstack:manager: beginning znp startup
[2024-08-31 16:54:41] debug: 	zh:zstack:znp: SREQ: --> SYS - osalNvLength - {"id":130}
[2024-08-31 16:54:41] debug: 	zh:zstack:unpi:writer: --> frame [254,2,33,19,130,0,178]
[2024-08-31 16:54:47] debug: 	zh:zstack:znp: SREQ: --> SYS - osalNvLength - {"id":130}
[2024-08-31 16:54:47] debug: 	zh:zstack:unpi:writer: --> frame [254,2,33,19,130,0,178]
[2024-08-31 16:54:53] debug: 	zh:zstack:znp: SREQ: --> SYS - osalNvLength - {"id":130}
[2024-08-31 16:54:53] debug: 	zh:zstack:unpi:writer: --> frame [254,2,33,19,130,0,178]
[2024-08-31 16:54:59] error: 	z2m: Error while starting zigbee-herdsman
[2024-08-31 16:54:59] error: 	z2m: Failed to start zigbee
[2024-08-31 16:54:59] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-31 16:54:59] error: 	z2m: Exiting...
[2024-08-31 16:54:59] error: 	z2m: TypeError: Cannot read properties of null (reading 'length')
    at AdapterNvMemory.init (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-nv-memory.ts:28:42)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:54:9)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)`

Any Idea why it suddenly stops and fails to start?

### What did you expect to happen?

Z2M starting correctly

### How to reproduce it (minimal and precise)

update Sonoff Dongle-P to latest firmware, start Z2M

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20240710

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus P

### Setup

Add-on on Homeassistant

tbrasser, 2024-09-02:
same here

```
[13:19:54] INFO: Preparing to start...
[13:19:54] INFO: Socat not enabled
[13:19:55] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-09-02 13:19:56] info: 	z2m: Logging to console, file (filename: log.log)
[2024-09-02 13:19:56] info: 	z2m: Starting Zigbee2MQTT version 1.40.0 (commit #unknown)
[2024-09-02 13:19:56] info: 	z2m: Starting zigbee-herdsman (0.57.1)
[2024-09-02 13:19:56] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-09-02 13:19:56] info: 	zh:zstack:znp: Serialport opened
[2024-09-02 13:20:03] error: 	z2m: Error while starting zigbee-herdsman
[2024-09-02 13:20:03] error: 	z2m: Failed to start zigbee
[2024-09-02 13:20:03] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-09-02 13:20:03] error: 	z2m: Exiting...
[2024-09-02 13:20:03] error: 	z2m: Error: SRSP - SYS - stackTune after 6000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:300:45
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:26)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1111:13
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:165:13)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:138:29)
    at Zigbee.start (/app/lib/zigbee.ts:64:27)
    at Controller.start (/app/lib/controller.ts:140:27)
    at start (/app/index.js:154:5)
```

flashed the dongle with:

```
sudo podman run --rm --device=/dev/ttyUSB0 --privileged --group-add "$(stat --format=%g /dev/ttyUSB0)" -v /dev/:/dev/ -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip ckware/ti-cc-tool:2024-03-19 -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb
Resolved "ckware/ti-cc-tool" as an alias (/var/cache/containers/short-name-aliases.conf)
Trying to pull docker.io/ckware/ti-cc-tool:2024-03-19...
Getting image source signatures
Copying blob e0183997aaa7 done   | 
Copying blob 4abcf2066143 skipped: already exists  
Copying blob 2aa9be932802 done   | 
Copying blob c3cdf40b8bda skipped: already exists  
Copying blob 09f577b6635e done   | 
Copying blob ae52dd4ac952 done   | 
Copying blob 48d1cb2b9803 done   | 
Copying blob 4377ce580b2c done   | 
Copying blob 4f4fb700ef54 skipped: already exists  
Copying config 88726657f6 done   | 
Writing manifest to image destination
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip
Firmware source: 'CC1352P2_CC2652P_launchpad_coordinator_20240710.zip'
Firmware file:   'CC1352P2_CC2652P_launchpad_coordinator_20240710.hex'
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20240710.hex
Firmware file: Intel Hex
Connecting to target...
CC1350 PG2.1 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:29:DE:23:D8
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xd9dd0124)
```

tbrasser, 2024-09-02:
Flashed back to previous firmware, but still having the same issue when starting!!! :(

closed 2024-12-20 by github-actions[bot]