Koenkk/zigbee2mqtt #23793: MiBoxer FUT037Z+ effects not working

keteflips, 2024-09-01:
### What happened?

I can turn on/off, change lights colors and brightness, but the effects simply do nothing, no mqtt message or effect working. Nothing happens when you change the effect.

### What did you expect to happen?

Just play the effects.

### Zigbee2MQTT version

1.39.1-dev commit: 6ed9b266

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Add-on on Home Assistant

keteflips, 2024-09-07:
Upgraded to 1.40 and the problem persists

grezco, 2024-09-07:
Same problem...

grezco, 2024-12-12:
Is a correction possible?

reneposthumus, 2025-03-06:
Please remove stele label, problem not yet solved.

giogua, 2025-04-29:
Same issue for me, with a Miboxer E3-ZR

closed 2025-09-06 by github-actions[bot]

giogua, 2025-09-06:
Please reopen, still not fixed