Koenkk/zigbee2mqtt #23794: When I control the Zigbee device, MQTT will send 2 command messages, which will cause the problem that some dimmable lamps suddenly go off after being turned on.

yuanxin32323, 2024-09-01:
### What happened?

All along, my Zigbee lamps have occasionally had the situation where they suddenly go off after being controlled to turn on, occurring once or twice every day. This experience is very bad. Later, I monitored the MQTT messages and found that each command of the control device was repeatedly issued twice.
 By calling the services "turn_on" and "turn_off" of HomeAssistant, MQTT would receive 2 commands. 
By calling the service "toggle" of HomeAssistant, MQTT would receive 3 commands. Repeated commands have little impact on switch-type devices, but the impact on dimmable lamps is fatal, resulting in the situation where they suddenly go off after being turned on. 

[image]
[image]

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

 20221226

### Adapter

CC2652P

### Setup

x86-64

frag-duino, 2024-09-03:
I'm experiencing the same problem since the update to 1.40.0 yesterday:
Every message is being sent two times, which makes toggeling lights annoying:

[image]
[image]
[image]

I'm using:
Zigbee2MQTT version
[1.40.0] commit: 6fa1dabe
Coordinator type: ConBee2/RaspBee2
Coordinator revision: 0x26780700
Openhabian with openhab 4.2.1

frag-duino, 2024-09-09:
Still facing the same problem with 1.40.1.
Found a workaround: Set a debounce of 0.1

closed 2024-12-20 by github-actions[bot]