Koenkk/zigbee2mqtt #23795: [New device support]: Can’t get Zemismart tuya blind motor to work in Z2MQTT, (ZM25-ZB)

DJAMIRSAM, 2024-09-01:
### Link

https://www.zemismart.com/products/zm25-zb

### Database entry

No idea where to find this (Checked using Vscode in HA)

### Comments

Hey so i just purchased one of these ZM25-ZB from them and in Z2MQTT it shows as Zigbee Model
TS0601
Zigbee Manufacturer
_TZE284_fzo2pocs

and says unsupported, does anyone have any idea on how to add support?

What I have in based on the [Z2MQTT Documentation](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-1-extending-the-external-definition) and [ZM25 Config](https://github.com/Koenkk/zigbee-herdsman-converters/blob/cea2c45bd37d4408702e52a476dea5c2bab730d4/src/devices/zemismart.ts)

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: 'Zemismart',
    description: 'Tubular motor',
    fromZigbee: [legacy.fz.tuya_cover, fz.ignore_basic_report],
    toZigbee: [legacy.tz.tuya_cover_control, legacy.tz.tuya_cover_options, legacy.tz.tuya_data_point_test],
    exposes: [e.cover_position().setAccess('position', ea.STATE_SET)],
    extend: [forcePowerSource({powerSource: 'Mains (single phase)'})],
};

module. Exports = definition;
```

What i have based on the add support for tuya documentation

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE284_fzo2pocs',
        },
    ],
    zigbeeModel: ['TS0601\u0000'],
    model: 'TS0601',
    vendor: '_TZE284_fzo2pocs',
    description: 'Tubular motor',
    fromZigbee: [legacy.fz.tuya_cover, fz.ignore_basic_report, legacy.fromZigbee.tuya_cover, tuya.fz.datapoints],
    toZigbee: [legacy.tz.tuya_cover_control, legacy.tz.tuya_cover_options, legacy.tz.tuya_data_point_test, legacy.toZigbee.tuya_cover_control, tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.enum('motor_direction', ea.STATE_SET, ['normal', 'reversed']).withDescription('Motor direction').withCategory('config'),
        e
            .enum('motor_working_mode', ea.STATE_SET, ['continuous', 'intermittently'])
            .withDescription('Motor operating mode')
            .withCategory('config'),
        e.enum('remote_pair', ea.STATE_SET, ['on', 'off']).withDescription('Remote control pairing mode').withCategory('config'),
        e.enum('upper_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set / Reset the upper stroke limit').withCategory('config'),
        e
            .enum('middle_stroke_limit', ea.STATE_SET, ['SET', 'RESET'])
            .withDescription('Set / Reset the middle stroke limit')
            .withCategory('config'),
        e.enum('lower_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set / Reset the lower stroke limit').withCategory('config'),
    ],
    extend: [forcePowerSource({powerSource: 'Mains (single phase)'})],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [5, 'motor_direction', tuya.valueConverter.tubularMotorDirection],
            [101, 'remote_pair', tuya.valueConverterBasic.lookup({on: true, off: false})],
            [103, 'upper_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
            [104, 'middle_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
            [105, 'lower_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
            [106, 'motor_working_mode', tuya.valueConverterBasic.lookup({continuous: tuya.enum(0), intermittently: tuya.enum(1)})],
        ],
    },

};

module.exports = definition;`

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_fzo2pocs',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Z2M won't detect the device as supported

DJAMIRSAM, 2024-09-01:
The Update today allows me to control it although the image and some other stuff are incorrect, I also cannot set the up, lower, middle limits using this

numminorih-git, 2024-12-03:
same here with 1.42.0-2: 
1. No settings for setup limits
2. incorrect shows state in HA (in log it shows opened, but on state - closed, and "open" button is inactive)

DJAMIRSAM, 2024-12-03:
Yep the inverse state is something that has personally been bugging me with my automations. The set state does not exist for the ones that got property recognize. Ended up getting the remote for pairing.

numminorih-git, 2024-12-03:
I managed to work with automation (with state ignore) with set position, for example:

```
metadata: {}
data:
  position: 100
target:
  entity_id:
    - cover.attic_projector_screen
action: cover.set_cover_position
```

Can anyone help with external converter? As I understand it should be same as _TZE200_1vxgqfba model that fully work with z2mqtt

hubaksis, 2025-01-24:
This external converter worked for me. 
The device is detected, can be controlled and the top and bottom limits can be set.

Please note: I am running on Raspberry PI, so your folder may be different. Also I am running v2.0.0

1. Create a folder `/opt/zigbee2mqtt/data/external_converters`
2. Create a file there `Zemismart_mains_blinds.js`
3. Paste the following code:
```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: tuya.fingerprint('TS0601', [
            '_TZE284_fzo2pocs'
        ]),
        model: 'TS0601_cover_8',
        vendor: 'Tuya',
        description: 'Cover motor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetTime,
        options: [exposes.options.invert_cover()],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            e.enum('reverse_direction', ea.STATE_SET, ['forward', 'back']).withDescription('Reverse the motor direction'),
            e.binary('motor_fault', ea.STATE, true, false),
			e.enum('upper_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set or Reset the upper stroke limit').withCategory('config'),
			e.enum('middle_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set or Reset the middle stroke limit').withCategory('config'),
			e.enum('lower_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set or Reset the lower stroke limit').withCategory('config'),
			e.enum('motor_working_mode', ea.STATE_SET, ['continuous', 'intermittently']).withDescription('Motor operating mode').withCategory('config'),
        ],
        whiteLabel: [
            tuya.whitelabel('Zemismart', ' ZM25TQ-2/25', 'Cover motor', ['_TZE284_fzo2pocs']),
        ],
        meta: {
            // All datapoints go in here
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverterBasic.lookup({OPEN: tuya.enum(0), STOP: tuya.enum(1), CLOSE: tuya.enum(2)})],
                [2, 'position', tuya.valueConverter.coverPosition],
                [3, 'position', tuya.valueConverter.coverPosition],
                [5, 'reverse_direction', tuya.valueConverterBasic.lookup({forward: tuya.enum(0), back: tuya.enum(1)})],
                [12, 'motor_fault', tuya.valueConverter.trueFalse0],                
				[103, 'upper_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
				[104, 'middle_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
				[105, 'lower_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
				[106, 'motor_working_mode', tuya.valueConverterBasic.lookup({continuous: tuya.enum(0), intermittently: tuya.enum(1)})]
            ],
        },
    };

module.exports = definition;

```
4. Restart Zigbee2mqtt. When it starts, you will see a message in mqtt topic zigbee2mqtt/bridge/converters with your file name and the saved code.

numminorih-git, 2025-01-24:
Confirm, [hubaksis] thx, all work + fixed incorrect shows state. Only little issue, that's no "image" in z2mqtt.

hubaksis, 2025-01-27:
> "Motor working mode / Motor operating mode", I don't understand what it is responsible for

This is a motor working more. 
Continuous - this is regular (default) mode, when blinds go up or down from top to bottom without stopping.
Intermittent - when motor turns to a specific angle once and stops. I suppose that this mode can be useful for setting up limits,  where you don't have to press up/down and stop immediately to find the perfect position. However this option is useless without another option that can set the 'angle' for this mode.

closed 2025-04-06 by github-actions[bot]