Koenkk/zigbee2mqtt #23802: Device quirk handling

RubenKelevra, 2024-09-01:
### What happened?

@Koenkk I was wondering if z2m is able to handle device quirks. I got several different light models which have weird turn-on behavior on certain parameters, like brightness and transition, where the light will flicker and then turn off again, either report this back as "turning back off" or not reporting back at all.

Is this a thing z2m would be interested in adding as rules, to avoid this, or is this left to the users?

Should we document such weird behavior in the hardware database?

Thanks in advance for advice!

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652RB stick

### Setup

Add-on in HomeAssistant, Rasberry Pi 4B

Koenkk, 2024-09-02:
@RubenKelevra is this something that could be fixed with custom converters? E.g. sending a different command instead?

RubenKelevra, 2024-09-02:
Yeah I think so.

Some examples:

#### Lidl HG08010
[HG08010](https://www.zigbee2mqtt.io/devices/HG08010.html#lidl-hg08010) will sometimes flicker just shortly, then turn off again and not update the state (showing as on) if you send

```
data:
  brightness: 1
  transition: 0
  kelvin: *any*
```

If you set no transition (it uses then it's default of 1 second) it does not happen.

##### Müller-Licht 45311

Müller-Licht 45311 (identified in z2m as [404037/404038](https://www.zigbee2mqtt.io/devices/404037_404038.html#m%25C3%25BCller%2520licht-404037%252F404038)) will often turn off after the transition duration or flicker for a sub second duration and update it's status again as "off" if you send the following command when it's off:

```
  brightness: *anything below 43*
  transition: *anything non-zero*
  kelvin: *any*
```

This also applies for turning it just on with a transition, if the brightness previously was below 43.

---

So it's stuff you would expect that a vendor is fixing in software updates. So I thought maybe adding documentation would help users to avoid these issues (or these vendors). Both vendors do not publish Firmware updates AFAIK.

Koenkk, 2024-09-03:
@RubenKelevra I think it's something to put on the device page, please make a PR by going to the device page, on the bottom you will find a link to edit the file.

closed 2024-12-20 by github-actions[bot]