Koenkk/zigbee2mqtt #23804: [Wrong device]: Ledvance Smart+ Classic Dimmable A60 9W E27 / 4058075208377 / AC33897 / A60 DIM Z3 wrongly identified as a Ledvance SMART+ candle E14 dimmable white / 405807520842

Trashwarez, 2024-09-02:
### Link

https://zigbee.blakadder.com/Ledvance_4058075208377.html

### Model

4058075208377 / AC33897 / A60 DIM Z3

### Description

Ledvance Smart+ Classic Dimmable A60 9W E27

### Vendor

Ledvance

### Picture (link)

https://www.ledvance.com/en/tech/products/index?productId=4058075208377&category=6387&documentId=165

### Database entry

{"id":62,"type":"Router","ieeeAddr":"0xf0d1b800001d2211","nwkAddr":45601,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"LEDVANCE DIM","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LEDVANCE DIM","manufacturerName":"LEDVANCE","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"02176550"}},"genLevelCtrl":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"02176550","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1725227746592}

### Notes

Device wrongly identified as Ledvance SMART+ candle E14 dimmable white / 4058075208421 (https://www.zigbee2mqtt.io/devices/4058075208421.html#ledvance-4058075208421).
Works fine with zigbee2mqtt except that it as been identified as a 4058075208421 model. So wrong picture and wrong OTA updates are offered.

Trashwarez, 2024-09-13:
This is an update of the upper external converter. Unfortunately the endpoints array doesn't seem to work with the most recent version of z2m although the values should be okay.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const ledvance_1 = require('zigbee-herdsman-converters/lib/ledvance');

const definition = {
    fingerprint: [{
        applicationVersion: 0,
        manufacturerID: 4489,
        type: 'Router',
//     dateCode: '20240428',
        hardwareVersion: 2,
        manufacturerName: 'LEDVANCE',
        modelID: 'LEDVANCE DIM',
        powerSource: 'Mains (single phase)',
//     softwareBuildID: '03177310',
        stackVersion: 0,
//     zclVersion: 8,
        endpoints: [{
            ID: 1,
            profileID: 260,
            deviceID: 257,
            inputClusters: [0,3,4,5,6,8,2821],
            outputClusters: [25],
        }],
    }],
    zigbeeModel: ['A60 DIM Z3'],
    model: '4058075208377',
    vendor: 'LEDVANCE',
    description: 'Smart+ Classic Dimmable A60 9W E27',
    icon: '/icons/Ledvance_33897.png',
    extend: [(0, ledvance_1.ledvanceLight)({})],
};

module.exports = definition;
```

michelbalzer, 2025-02-25:
Same with the GU10. (Providing more info soon.)

closed 2025-07-27 by github-actions[bot]