Koenkk/zigbee2mqtt #23805: Change parameters for reporting will fail on Xiaomi's GZCGQ01LM

RubenKelevra, 2024-09-02:
### What happened?

I noticed that the GZCGQ01LM is quite "choppy" on updated of the brightness. They will happened very quickly, but only if they are large enough. Otherwise, there won't be an update.

While this might not be an update for binary decisions if a light should be turned on or not, I like to use them as feedback for the light's brightness in a room, to meet specific brightness targets without wasting energy.

So I naturally tried to adjust the reporting parameters, which - if I understand this correctly - would adjust the minimal reporting interval and the minimal reporting differences in brightness (in their internal unit, not Lux).

But this just leads to an error:

[Screencast_20240902_022745.webm](https://github.com/user-attachments/assets/c1f50c01-4e0b-4d44-bfee-fb541cf094a1)

### What did you expect to happen?

No errors.

### How to reproduce it (minimal and precise)

- Update reporting specs in 1.40.0 for a Xiaomi GZCGQ01LM

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652RB stick

### Setup

Add-on in HomeAssistant, Rasberry Pi 4B

Claude2666, 2024-09-03:
Have you tried to wake up the sensor before updating the reporting values?

My experience with these sensors is that they are not 100% reliable. They sometimes don't report anything for an hour, sometimes the reported lux value drops drastically while it is in full sun.  The newer version GZCGQ11LM is more stable.

RubenKelevra, 2024-09-03:
For me those sensors are very reliable. Just not responsive enough for small changes of brightness.

I'm currently doing a proof of concept an upcoming Adaptive Lighting feature I like to write to see if they work as feedback for brightness adjustments in a room.

I'll try hitting the button before sending the config values. Thanks for the suggestion!

RubenKelevra, 2024-09-03:
Weird.

Here's the TCS34725 (named "Ambient light") vs an GZCGQ01LM:

[image]

Maybe it's a power issue? I got one of my GZCGQ01LM wired up to a power supply and actually planned to do that with the rest as well, as they most likely eat through batteries pretty quickly if I change the settings.

> Anyway, I could adjust reporting by waking the sensor up and updating the reporting values. (I am still on 1.39.0-dev)

I'll try that, thanks for reporting back!

RubenKelevra, 2024-09-03:
Works! Man what a difference!
[image]

Thanks for the quick help.

closed 2024-09-03 by RubenKelevra