Koenkk/zigbee2mqtt #23807: LEDVANCE 4058075729247 SMART+ Filament Globe125 52 E27 Amber dimmable OTA update failed with size mismatch

NickTheRipper, 2024-09-02:
### What happened?

OTA Update show new firmware version for my lamps, but update failed with error „Size mismatch“

error line:
("installed_version":51656288,"latest_version":52442720,"state":"available) but OTA firmware update failed with error „Size mismatch“.)

### How to reproduce it (minimal and precise)

Checked OTA updates and tried to update new showed version for my lamps

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

7.4.3 [GA]

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Setup

Add-on on Home Assistant on Home Assistant Green

### Debug log

[log.log](https://github.com/user-attachments/files/16835720/log.log)

itchensen, 2024-09-02:
Same here regarding a LEDVANCE [4058075729322](https://www.zigbee2mqtt.io/devices/4058075729322.html#ledvance-4058075729322) PLUG COMPACT EU T (SMART+ Compact Outdoor Plug EU) as commented [here](https://github.com/Koenkk/zigbee2mqtt/issues/22687#issuecomment-2324839574)

But one of my two plugs worked nicely.

Koenkk, 2024-09-03:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/22687

closed 2024-09-03 by Koenkk