Koenkk/zigbee2mqtt #23809: Self-test not working on smoke detector Aqara JTYJ-GD-01LM/BW

McGiverGim, 2024-09-02:
### What happened?

Since some time ago, the self-test option for the smoke detector is not working at all.

### What did you expect to happen?

To make a self-test of the device.

### How to reproduce it (minimal and precise)

Press the self-test button on the mqtt UI for the device or use the instructions provided here https://www.zigbee2mqtt.io/devices/JTYJ-GD-01LM_BW.html#self-test

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20240710

### Adapter

Egony E72

### Setup

Add-on on Home Assistant, Raspberry Pi 4b

### Debug log

In the log I can see a:
```
[2024-09-02 12:08:27] debug: 	zh:zstack: Data confirm error (0x00158d0003612ca2:13228,205,3)
```

This is the log, if you need more please tell me:
```
[2024-09-02 12:08:19] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/detector_humo/set' with data '{"selftest": ""}'
[2024-09-02 12:08:19] debug: 	z2m: Publishing 'set' 'selftest' to 'detector_humo'
[2024-09-02 12:08:19] debug: 	zh:controller:endpoint: ZCL command 0x00158d0003612ca2/1 ssIasZone.write({"65521":{"value":50397184,"type":35}}, {"timeout":35000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":4447,"writeUndiv":false})
[2024-09-02 12:08:19] debug: 	zh:controller:endpoint: Request Queue (0x00158d0003612ca2/1): send request
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: <-- [254,33,68,129,0,0,0,239,117,189,1,1,0,142,0,26,19,86,0,0,13,9,62,2,0,2,20,2,0,4,0,0]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,117,189,1,1,0,142,0,26,19,86,0,0,13,9,62,2,0,2,20,2,0,4,0,0]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: <-- [0,3,117,189,29,236]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,117,189,1,1,0,142,0,26,19,86,0,0,13,9,62,2,0,2,20,2,0,4,0,0,0,3,117,189,29,236]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,117,189,1,1,0,142,0,26,19,86,0,0,13,9,62,2,0,2,20,2,0,4,0,0,0,3,117,189,29] - 236
[2024-09-02 12:08:22] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":48501,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":142,"securityuse":0,"timestamp":5640986,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,62,2,0,2,20,2,0,4,0,0,0,3]}}
[2024-09-02 12:08:22] debug: 	zh:controller: Received payload: clusterID=61184, address=48501, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=142, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":62,"commandIdentifier":2},"payload":{"seq":512,"dpValues":[{"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,3]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
[2024-09-02 12:08:22] debug: 	zh:controller:endpoint: ZCL command 0xa4c138ded93a2357/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":62,"writeUndiv":false})
[2024-09-02 12:08:22] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c138ded93a2357:48501/1 (0,0,3)
[2024-09-02 12:08:22] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":48501,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":9,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,62,11,2,0]}}
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,117,189,1,1,0,239,9,0,30,5,16,62,11,2,0,56]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-09-02 12:08:22] debug: 	z2m: Received Zigbee message from 'calidad_aire_comedor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,3],"type":"Buffer"},"datatype":2,"dp":20}],"seq":512}' from endpoint 1 with groupID 0
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-09-02 12:08:22] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,9,207]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,9,207]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,9] - 207
[2024-09-02 12:08:22] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":9}
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: <-- [254,33,68,129,0,0,0,239,117,189,1,1,0,142,0,126,32,86,0,0,13,9,63,2,0,2,21,2,0,4,0,0]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,117,189,1,1,0,142,0,126,32,86,0,0,13,9,63,2,0,2,21,2,0,4,0,0]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: <-- [0,6,117,189,29,190]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,117,189,1,1,0,142,0,126,32,86,0,0,13,9,63,2,0,2,21,2,0,4,0,0,0,6,117,189,29,190]
[2024-09-02 12:08:22] debug: 	zh:zstack:unpi:parser: --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,117,189,1,1,0,142,0,126,32,86,0,0,13,9,63,2,0,2,21,2,0,4,0,0,0,6,117,189,29] - 190
[2024-09-02 12:08:22] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":48501,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":142,"securityuse":0,"timestamp":5644414,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,63,2,0,2,21,2,0,4,0,0,0,6]}}
[... 167 lines cut ...]
[2024-09-02 12:08:27] debug: 	z2m: Received Zigbee message from 'enchufe_dormitorio_lamparas', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":232}' from endpoint 1 with groupID 0
[2024-09-02 12:08:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/enchufe_dormitorio_lamparas', payload '{"child_lock":"UNLOCK","countdown":0,"current":0,"energy":55.55,"indicator_mode":"on/off","last_seen":"2024-09-02T12:08:27+02:00","linkquality":25,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":67,"latest_version":67,"state":"idle"},"update_available":false,"voltage":232}'
[2024-09-02 12:08:27] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,205,1,16,27]
[2024-09-02 12:08:27] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,205,1,16,27]
[2024-09-02 12:08:27] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [205,1,16] - 27
[2024-09-02 12:08:27] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":16}
[2024-09-02 12:08:27] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-09-02 12:08:27] debug: 	zh:zstack: Data confirm error (0x00158d0003612ca2:13228,205,3)
[2024-09-02 12:08:27] debug: 	zh:zstack: Wait 2000ms
[2024-09-02 12:08:28] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,151,232,1,61,127,184,254,33,68,129,0,0,0,239,117,189,1,1,0,138,0,151,196,91,0,0,13,9]
[2024-09-02 12:08:28] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,151,232,1,61,127,184,254,33,68,129,0,0,0,239,117,189,1,1,0,138,0,151,196,91,0,0,13,9]
[2024-09-02 12:08:28] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [151,232,1,61,127] - 184
[2024-09-02 12:08:28] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":59543,"relaycount":1,"relaylist":[32573]}
[2024-09-02 12:08:28] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,117,189,1,1,0,138,0,151,196,91,0,0,13,9]
[2024-09-02 12:08:28] debug: 	zh:zstack:unpi:parser: <-- [68,2,0,0,18,2,0,4,0,0,1,25,117,189,29,218]
[2024-09-02 12:08:28] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,117,189,1,1,0,138,0,151,196,91,0,0,13,9,68,2,0,0,18,2,0,4,0,0,1,25,117,189,29,218]
[2024-09-02 12:08:28] debug: 	zh:zstack:unpi:parser: --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,117,189,1,1,0,138,0,151,196,91,0,0,13,9,68,2,0,0,18,2,0,4,0,0,1,25,117,189,29] - 218
[... cut ...]
[2024-09-02 12:08:28] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c138ded93a2357:48501/1 (0,0,3)
[2024-09-02 12:08:28] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":48501,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":17,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,68,11,2,0]}}
[2024-09-02 12:08:28] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,117,189,1,1,0,239,17,0,30,5,16,68,11,2,0,90]
[2024-09-02 12:08:28] debug: 	zh:zstack:unpi:parser: --- parseNext []
```

McGiverGim, 2024-09-02:
I close this. After removing the battery of the device and connected again, it seems to be working.

closed 2024-09-02 by McGiverGim

877dev, 2025-02-01:
Self test fails quite regularly for me as well, but no error. Maybe I have to enable debugging.

I'd be interested to sniff the traffic but have never done that before. 

I need to re-pair it every time.