Koenkk/zigbee2mqtt #23812: [New device support]: WKD-M8 Pro

Wolo76, 2024-09-02:
### Link

https://aliexpress.ru/item/1005007508844507.html?spm=a2g2w.orderdetail.0.0.481d4aa6eaWkv8&sku_id=12000041070109359

### Database entry

{"id":4,"type":"router","ieeeAddr":"0xa4c13808e25ae3aa

### Comments

couldn't find anything like it
[image]

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_atuj3i0w',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

does not meet the definition. 
There should be a presence sensor, 4 light buttons and a display panel that displays information: date, weather.

turbofex, 2024-09-28:
I've just created a basic version for this switch, switches only,no date nor weather & switch name change.
See if you need it.Save it as js and load it in the frontend of Z2M should do.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
    {
    modelID: 'TS0601', manufacturerName: '_TZE284_atuj3i0w',
       },
     ],
    model: 'M8Pro',
    vendor: 'Tuya',
    description: 'M8Pro 4 gang switch with LCD.KaneLau basic ver',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.switch().withEndpoint('l1'),
        tuya.exposes.switch().withEndpoint('l2'),
        tuya.exposes.switch().withEndpoint('l3'),
        tuya.exposes.switch().withEndpoint('l4'),
    ],
    endpoint: (device) => {
        return {l1: 1, l2: 1, l3: 1, l4: 1};
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [24, 'state_l1', tuya.valueConverter.onOff],
            [25, 'state_l2', tuya.valueConverter.onOff],
            [26, 'state_l3', tuya.valueConverter.onOff],
            [27, 'state_l4', tuya.valueConverter.onOff],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;

Koenkk, 2024-09-29:
@turbofex feel free to make a PR!

Wolo76, 2024-09-30:
SUPER

Wolo76, 2024-10-17:
Good afternoon

Can this be added to the z2m update?
I couldn't do it myself(

Koenkk, 2024-10-18:
Added it.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `M8Pro.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

Dimmers81, 2024-11-01:
Hi I actually ordered this item and will get it in a few weeks time. Regarding the time and weather option, what can I do to help with the weather/time feature. I don't pretend to know what to do but is there anything I can do to assist with this?

Wolo76, 2024-12-03:
time is working.
We need to make the weather. It doesn't show the weather.

Dimmers81, 2024-12-08:
Hi, @Koenkk is there anything I can do to add the weather aspect? As I have this device any suggestions, happy to help and so would @sebekhtc

fueryon, 2024-12-22:
working on zha implementation, some datapoints:
```
// backlight (LED's on buttons)
TuyaDatapointData(dp=101, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))

// show weather/clock or just switch names
TuyaDatapointData(dp=36, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))

// (?) maybe weather
TuyaDatapointData(dp=102, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x03', *payload=3))

// switch mode (?) switch_1, scene_1, smart_light_1
TuyaDatapointData(dp=21, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
TuyaDatapointData(dp=20, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
TuyaDatapointData(dp=19, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
TuyaDatapointData(dp=18, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))

// relay status (?)
TuyaDatapointData(dp=42, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
TuyaDatapointData(dp=41, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
TuyaDatapointData(dp=40, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
TuyaDatapointData(dp=39, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))

// indicator status (?)
TuyaDatapointData(dp=37, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))

// reset state (?)
TuyaDatapointData(dp=38, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))

// timers (?)
TuyaDatapointData(dp=33, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
TuyaDatapointData(dp=32, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
TuyaDatapointData(dp=31, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
TuyaDatapointData(dp=30, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))```

hoped weather uses:
https://developer.tuya.com/en/docs/iot/weather-function-description
but i don't see command 0x20 anywhere in logs

fueryon, 2025-01-01:
Figured out switch names:
datapoints 103, 104, 105, 106 - names displayed in switch mode
datapoints 107, 108, 109, 110 - names displayed in scene mode
those are string endpoints, but device doesn't use strings to set display names - we have to send 2 byte hex representations of characters (UTF-8):
to set switch name to "abc" i sent "006100620063"

Dimmers81, 2025-02-05:
@fueryon, thanks for this. I hope this will be implemented soon, names and weather.

daksh1305, 2025-02-24:
@Koenkk I am having the same issue i am able to connect the device successfully but i am unable to fetch the weather not able to change the switch name to custom name.

Please help me with that

GeorgeWatch, 2025-04-19:
@fueryon 
what does the full code look like in Mqtt explorer to rename the ones displayed in switch mode 103 ... etc?

**I managed to synchronize the time using automation:**
alias: M8Pro Time Sync
description: ""
triggers:
  - trigger: time_pattern
    hours: /20
conditions: []
actions:
  - data:
      topic: zigbee2mqtt/CUSTOM_NAME/set
      payload: "{\"time_remaining\": {{ now().timestamp() | int }}}"
      qos: 1
      retain: false
    action: mqtt.publish
mode: single

Mqtt Explorer: 
topic: zigbee2mqtt/CUSTOM_NAME/set
json: { "time_remaining": 1715000000 }

reineruhry, 2025-05-27:
Added support for Labeling the Buttons!

Later I'll add more functions to it!

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const { TextEncoder, TextDecoder } = require("util");

function bytesToHex(bytes) {
  return Array.from(bytes, byte => byte.toString(16).padStart(4, '0')).join('');
}
function hexToBytes(hex) {
  const bytes = [];
  for (let i = 0; i < hex.length; i += 4) {
    bytes.push(parseInt(hex.slice(i, i + 4), 16));
  }
  return Uint8Array.from(bytes);
}

function stringToUTF8Bytes(string) {
  console.log('stringToUTF8Bytes: '+new TextEncoder().encode(string));
  return new TextEncoder().encode(string);
}

function utf8BytesToString(bytes) {
  console.log('utf8BytesToString: '+new TextDecoder('utf-8').decode(bytes));
  return new TextDecoder('utf-8').decode(bytes);
}

const valueConverterLocal = {
  name: {
    to: (v, meta) => {
      const stringValue = String(v ?? "");
      const limitedString = stringValue.slice(0, 12);
      return bytesToHex(stringToUTF8Bytes(limitedString));
    },
    from: (v, meta, options, publish) => {
      return utf8BytesToString(hexToBytes(v));
    },
  },
}
const definition = {
    fingerprint: [{
        modelID: 'TS0601',
        manufacturerName: '_TZE284_atuj3i0w',
    }, ],
    model: 'M8Pro',
    vendor: 'Tuya',
    description: 'M8Pro 4 gang switch with LCD.',
    fromZigbee: [tuya.fz.datapoints], //
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint) => {
      const endpoint = device.getEndpoint(1);
      await tuya.configureMagicPacket(device, coordinatorEndpoint);
      await endpoint.command('manuSpecificTuya', 'mcuVersionRequest', {seq: 0x0002});
      await device.getEndpoint(1).command("manuSpecificTuya", "dataQuery", {});
    },
    exposes: [
        tuya.exposes.switch().withEndpoint('l1'),
        tuya.exposes.switch().withEndpoint('l2'),
        tuya.exposes.switch().withEndpoint('l3'),
        tuya.exposes.switch().withEndpoint('l4'),
        e.text("name", ea.STATE_SET).withEndpoint("l1").withDescription("Name for Switch 1"),
        e.text("name", ea.STATE_SET).withEndpoint("l2").withDescription("Name for Switch 2"),
        e.text("name", ea.STATE_SET).withEndpoint("l3").withDescription("Name for Switch 3"),
        e.text("name", ea.STATE_SET).withEndpoint("l4").withDescription("Name for Switch 4"),
    ],
    endpoint: (device) => {
        return {
            l1: 1,
            l2: 1,
            l3: 1,
            l4: 1
        };
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [24, 'state_l1', tuya.valueConverter.onOff],
            [25, 'state_l2', tuya.valueConverter.onOff],
            [26, 'state_l3', tuya.valueConverter.onOff],
            [27, 'state_l4', tuya.valueConverter.onOff],
            [103, 'name_l1', valueConverterLocal.name],
            [104, 'name_l2', valueConverterLocal.name],
            [105, 'name_l3', valueConverterLocal.name],
            [106, 'name_l4', valueConverterLocal.name],
         
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
`

Koenkk, 2025-06-03:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

reineruhry, 2025-06-04:
> Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Done!
https://github.com/Koenkk/zigbee-herdsman-converters/pull/9450

fueryon, 2025-06-13:
[image]
[image]
Got weather working (in zha custom quirk).
To set weather we need to respond with command 0x61 to weather request (command 0x60) - I just copied the time setting logic. The responding part probably doesn't matter (not tested, but i think just sending command 0x61 will also set weather). Although when gateway responds with correct weather payload, switch will poll for weather quite often.
Now for a payload - we need to send packet formed exactly as module requested (if module requested  3 days forecast, we need to send 3 days forecast). Probably the weather request is just hardcoded in the module and response values are read from hardcoded offsets (not tested)
Response format follows [this](https://developer.tuya.com/en/docs/mcu-standard-protocol/mcusdk-zigbee-uart-protocol?id=Kdg17v4544p37#0x3b) specification - switch will display today forecast values (not current weather)
what i sent exactly to set weather (notice no data length bytes, just rada right after command id 0x61):
`\x11 \x00 - header + location
\x12 \x03 - constant + forecast days
\x13 \x01 - constat + current weather flag
\x01   \x00 \x33    \x00 \x12    \x00 \x23     \x00 \x27 - temperature (current + 3 forecast, thats signed int so we can go negative)
\x02   \x00 \x3C    \x00 \x28    \x00 \x14     \x00 \x0B - humidity (current + 3 forecast)
\x03   \x78            \x67            \x6D             \x66 - condition`
as I mentioned, module will display today's forecast (so 2 column) - temp 0x0012 = 18, hum is ignored, condition 0x67 = 103 - sandstorm

Good luck with implementing it (I will publish my quirk soon, don't know how useful will it be for z2mqtt)

nigels0, 2025-06-19:
Where will the quirk be published?

fueryon, 2025-06-19:
> Where will the quirk be published?

https://github.com/fueryon/zha-device-handlers/blob/m8pro_switch/zhaquirks/tuya/m8pro.py
its messy and definitely unfinished, But kindof sortof works ;)

nigels0, 2025-06-19:
Thanks! It’s a big improvement - day/date and switches are working - but I’ve yet to work out ahow to get the weather correctly. Any tips?

nigels0, 2025-06-19:
That’s really cool?where did you put the yaml?

fueryon, 2025-06-24:
Datapoint 36 (bool) controls what is displayed (weather and clock if true, button names if false)
Don't know what datapoint 37 (enum) does - can't see any difference in behaviour, tested all 3 values
There is also Datapoint 101 (bool) that controls "backlight" - button leds (white/orange) will be off for false
Bonus - datapoint 102 (int, 32bit) seems to control "LCD screensaver timeout" - how long button names/weather are visible after press/movement detection (Default 3, 0~=11s, 3~=15s, 6~=19s, 9~=23s) needs more testing.
I'm pretty sure i've managed to disable pir sensor alltogether (no screensaver, button names always visible) in the past, not sure how
hope that helps ;)

pkhodak, 2025-06-25:
I'm curious if you have code for 101 endpoint and if you've managed to get backlight turned on and off? In my M8-4 device I have the endpoint 36 described as "backlight_switch" and type of bool but no matter what I send to it the backlight state doesn't change. Tuya Smart Life app works fine and changes it without a problem so I know it could work if I can figure out the right approach.

fueryon, 2025-06-25:
There is working code in my quirk for zha. And yes - changing datapoint 101 value turns on and off button LED's. Can't help You with z2mqtt code.

pkhodak, 2025-06-30:
Thanks for this, I have tried this code but sadly it doesn't work for the switch from the same manufacturer that I'm trying to use (https://github.com/Koenkk/zigbee2mqtt/issues/27791)

reineruhry, 2025-07-25:
Hi Luiz,

I'm still working converting the weather solution developed by **fueryon** which will support the display the correct weather.

Thanks, 
Reiner

luizcarlosnery, 2025-07-25:
> Olá Luiz,
> 
> Ainda estou trabalhando na conversão da solução climática desenvolvida pela **fueryon** , que permitirá a exibição do clima correto.
> 
> Obrigado, Reiner

Muito Obrigado @reineruhry , pela resposta rapida, voces são TOP !!!

lwator, 2025-08-16:
i create a quirk for HA (with AI help) like this, but when I add replacement it stop loading and is represent as:
"class": "zigpy.device.Device"

Any one cen help me to find out issue?
```
"""
Final, working Quirk for Tuya 4-gang switch (_TZE284_atuj3i0w).

Hypothesis: The replacement was failing because it discarded the essential
TuyaManufCluster. This version re-adds it to the primary endpoint, allowing
the TuyaSwitchBus and other components to function correctly.
"""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, OnOff, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufCluster, TuyaOnOff, TuyaSwitchBus

MANUFACTURER = "_TZE284_atuj3i0w"
MODEL = "TS0601"

class TuyaOnOffL1(TuyaOnOff):
    """Tuya OnOff cluster for datapoint 24."""
    dp_to_attribute = {24: "on_off"}

class TuyaOnOffL2(TuyaOnOff):
    """Tuya OnOff cluster for datapoint 25."""
    dp_to_attribute = {25: "on_off"}

class TuyaOnOffL3(TuyaOnOff):
    """Tuya OnOff cluster for datapoint 26."""
    dp_to_attribute = {26: "on_off"}

class TuyaOnOffL4(TuyaOnOff):
    """Tuya OnOff cluster for datapoint 27."""
    dp_to_attribute = {27: "on_off"}

class Tuya4GangSwitchFinalAttempt(CustomDevice):
    """Final attempt for the Tuya M8Pro 4-gang switch."""

    def __init__(self, *args, **kwargs):
        """Init."""
        self.switch_bus = TuyaSwitchBus(self)
        super().__init__(*args, **kwargs)

    signature = {
        MODELS_INFO: [(MANUFACTURER, MODEL)],
        ENDPOINTS: {
            1: {
                "profile_id": 260,
                "device_type": 81,
                "input_clusters": [0, 4, 5, 0xed00, 0xef00],
                "output_clusters": [10, 25],
            },
            242: {
                "profile_id": 41440,
                "device_type": 97,
                "input_clusters": [],
                "output_clusters": [33],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaOnOffL1,
                    TuyaManufCluster,  # THE CRITICAL ADDITION - KEEP THE "BRAIN"
                ],
                OUTPUT_CLUSTERS: [],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [TuyaOnOffL2],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [TuyaOnOffL3],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [TuyaOnOffL4],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

Hollako, 2025-09-17:
Hello Gents,

Is there any possibility to solve the weather and backlight issues with Z2M ? 
I bought 2 M8-Pro recently, everything is working perfectly except these 2 things.

Regards,

chris01150, 2025-09-29:
I have same problem for weather

devkaiwang, 2025-10-18:
@l1berat0r ,

Thanks for the reply! 

> Is there always a difference of "1"? For some reason, M8Pro for negative values, always displayed a value greater by 1 than one being sent.

A difference of "1" might be caused by writing the buffer with `writeUInt16BE` in your code for the temperature. According to the Tuya document, `writeInt16BE` should be used. (UINT16 vs INT16).
[image]  

> You can disable this behaviour by passing correctForNegativeValues = false to the converter constructor. Have you tried that?

Yes, I have tried, but still cannot display temp < 0.

> You can check if [weather condition mapping](https://github.com/l1berat0r/zigbee-herdsman-converters/blob/90afe5587d9790f617df3f1b3012ac8546477ca5/src/lib/tuya.ts#L62) is the same for F3-PRO. From what I've noticed there were many codes for the same icons, so I just put there single arbitrary choosen values. Especially there were a lot of codes that ended up, setting "Sunny" condition. If "100" doesn't work for F3-PRO, probably we should be able to choose something else that will work for both models.

I have updated the table and it works for me. Here is the [doc reference](https://developer.tuya.com/en/docs/mcu-standard-protocol/Bluetooth-LE-Weather-Service?id=Kd3wc6khl3423#title-3-Appendix). If it also works for you, we can have a unified one:

```TypeScript
export const WeatherCondition = {
    sunny: 120,
    heavy_rain: 101,
    thunderstorm: 102,
    sandstorm: 103,
    light_snow: 104,
    snow: 105,
    freezing_fog: 106,
    rainstorm: 107,
    isolated_shower: 108,
    dust: 109,
    thunder_and_lightning: 110,
    light_shower: 111,
    rain: 112,
    sleet: 113,
    dust_devil: 114,
    ice_pellet: 115,
    strong_sandstorm: 116,
    sand_blowing: 117,
    light_to_moderate_rain: 118,
    mostly_clear: 119,
    fog: 121,
    shower: 122,
    heavy_shower: 123,
    heavy_snow: 124,
    extraordinary_rainstorm: 125,
    blizzard: 126,
    hail: 127,
    light_to_moderate_snow: 128,
    partly_cloudy: 129,
    light_snow_shower: 130,
    moderate_snow: 131,
    overcast: 132,
    needle_ice: 133,
    downpour: 134,
    thundershower_and_hail: 136,
    freezing_rain: 137,
    snow_shower: 138,
    light_rain: 139,
    haze: 140,
    moderate_rain: 141,
    cloudy: 142,
    thundershower: 143,
    moderate_to_heavy_rain: 144,
    heavy_rain_to_rainstorm: 145,
    clear: 146,
};
```

markajm, 2025-11-07:
Hi,  

To fix your time, try adding  timeStart: "1970" to your converter.  or PR your device.  This fixed mine after the z2m 2.6.3 update.

```
        tuya.modernExtend.tuyaBase({
            timeStart: "1970",
            // ... existing code
        }),
```

I agree too that we should probably do a combined weather condition based on the tuya reference.  Some (future) devices might follow the table more faithfully.  But you can add your own custom override to l1berat0r's M8ProTuyaWeatherCondition table with an 'External Extension' (not a converter).   That way we don't need to fill up the z2m code with multiple variations of the weather condition values.

I added this as an extension and adjusted the text to fit my weather provider and choosing the best match number from the tuya values.  Working for my M9Pro external converter in HA- so I assume will still work once the PR is merged.  There might be a better way to do it - let me know.  I'm not very good with js.

tuya_weather_condition_override.js
```
const tuya = require('zigbee-herdsman-converters/lib/tuya');

class TuyaWeatherOverride {

    //constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
    constructor() {
       
        // delete all existing keys
        for (const key in tuya.M8ProTuyaWeatherCondition) {
          if (tuya.M8ProTuyaWeatherCondition.hasOwnProperty(key)) {
            delete tuya.M8ProTuyaWeatherCondition[key];
          }
        }

        // BOM 
        Object.assign(tuya.M8ProTuyaWeatherCondition, {
            // BOM values best-fit to available choices on M9Pro
                                    // Screen text
            clear: 120,             // Sunny
            cloudy: 133,            // Overcast day
            cyclone: 107,           // Torrential rain 
            dust: 103,              // Dust storm
            dusty: 109,             // Dust storm
            exceptional: 125,       // Heavy rainstorm
            fog: 121,               // Freezing fog        
            frost: 121,             // Freezing fog
            haze: 140,              // Smog
            hazy: 140,              // Smog
            heavy_shower: 101,      // Heavy rain
            heavy_showers: 101,     // Heavy rain
            light_rain: 139,        // Drizzle
            light_shower: 139,      // Drizzle
            light_showers: 139,     // Drizzle
            mostly_sunny: 119,      // Cloudy sunny
            partly_cloudy: 129,     // Cloudy sunny 
            rain: 101,              // Heavy rain
            shower: 141,            // Moderate rain
            showers: 141,           // Moderate rain
            snow: 105,              // Light snow
            storm: 102,             // Thunder shower
            storms: 102,            // Thunder shower
            sunny: 120,             // Sunny
            tropical_cyclone: 125,  // Heavy rainstorm
            wind: 117,              // Yangsha
            windy: 117,             // Yangsha
            None: 120,              // Sunny
        });
    }

        // // tuya mapping vs M9Pro display
        // // https://developer.tuya.com/en/docs/mcu-standard-protocol/Bluetooth-LE-Weather-Service?id=Kd3wc6khl3423#title-3-Appendix
        // Object.assign(tuya.M8ProTuyaWeatherCondition, { // screen text              icon notes
        //     heavy_rain: 101,                            // Heavy rain
[... 35 lines cut ...]
        //     freezing_rain: 137,                         // Overcast day             icon same as 133
        //     snow_shower: 138,                           // Heavy                    icon same as 128
        //     light_rain: 139,                            // Drizzle                  icon same as 112
        //     haze: 140,                                  // Smog
        //     moderate_rain: 141,                         // Moderate rain            icon same as 118
        //     cloudy: 142,                                // Cloudy sunny             icon same as 119
        //     thundershower: 143,                         // Thunder shower           icon same as 102
        //     moderate_to_heavy_rain: 144,                // Moderate rain            icon same as 118
        //     heavy_rain_to_rainstorm: 145,               // Heavy rain               icon same as 101
        //     clear: 146,                                 // Sunny
        // });

    // Required method
    start() {
    }

    // Required method
    stop() {
    }
}

module.exports = TuyaWeatherOverride;
```

[image]

Hollako, 2026-01-02:
Hello Gents,

First of all Happy New Year !!

My question is whether it is possible to implement fixed Time/Date functionality for the M8-Pro, along with weather integration, in an official and supported way. I previously used the external converter that functioned correctly with an earlier version, but it no longer works with the current release.

On the other hand, the M9-Pro supports Time/Date and Weather natively in the official Z2M release, without the need for any external converters, and it works perfectly.

Could you please advise on the feasibility of achieving the same level of functionality for the M8-Pro?

l1berat0r, 2026-01-02:
Hello. I needed to back off for a while from contributing and I thought someone would push a PR for fixing this issue. As I see none was created so I pushed mine. The issue should be fixed, when the change is accepted by @Koenkk .

@devkaiwang
> A difference of "1" might be caused by writing the buffer with `writeUInt16BE` in your code for the temperature. According to the Tuya document, `writeInt16BE` should be used. (UINT16 vs INT16). 

`writeInt16BE` is used for creating temperature value in the buffer. The issue occurs regardless of that.
Have you been able to find a solution? I will have some time in the following days so I will also try to do some trouble-shooting for F3-PRO.

closed 2026-03-14 by github-actions[bot]