Koenkk/zigbee2mqtt #23814: Enocean PTM 215Z Greenpower not working after upgrade 1.31.1=> 1.40.0

coendaar, 2024-09-02:
### What happened?

After upgrading from 1.31.2 to 1.40.0 the Enocean PTM215Z no longer works.
After downgrading back to 1.31.2 it works again.

I've tried
- factory reset on the ptm 215z
- repairing the ptm215z (join al)

### What did you expect to happen?

This is logging from version 1.31.2:
`
debug 2024-09-02 14:39:32: Received Zigbee message from '0x0000000001718f82', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":19,"frameCounter":9637,"options":0,"payloadSize":0,"srcID":24219522}' from endpoint 242 with groupID 2948
debug 2024-09-02 14:39:32: Device '0x0000000001718f82' reconnected
info  2024-09-02 14:39:32: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82/availability', payload 'online'
info  2024-09-02 14:39:32: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82', payload '{"action":"press_3","device":{"friendlyName":"0x0000000001718f82","ieeeAddr":"0x0000000001718f82","manufacturerID":null,"model":"PTM 215Z","networkAddress":24219522,"type":"GreenPower"},"last_seen":"2024-09-02T12:39:32.322Z","linkquality":127}'
info  2024-09-02 14:39:32: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82', payload '{"action":"","device":{"friendlyName":"0x0000000001718f82","ieeeAddr":"0x0000000001718f82","manufacturerID":null,"model":"PTM 215Z","networkAddress":24219522,"type":"GreenPower"},"last_seen":"2024-09-02T12:39:32.322Z","linkquality":127}'
info  2024-09-02 14:39:32: MQTT publish: topic 'homeassistant/device_automation/0x0000000001718f82/action_press_3/config', payload '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x0000000001718f82"],"manufacturer":"EnOcean","model":"Pushbutton transmitter module (PTM 215Z)","name":"0x0000000001718f82"},"payload":"press_3","subtype":"press_3","topic":"zigbee2mqtt/0x0000000001718f82/action","type":"action"}'
info  2024-09-02 14:39:32: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82/action', payload 'press_3'
debug 2024-09-02 14:39:32: Received MQTT message on 'homeassistant/device_automation/0x0000000001718f82/action_press_3/config' with data '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x0000000001718f82"],"manufacturer":"EnOcean","model":"Pushbutton transmitter module (PTM 215Z)","name":"0x0000000001718f82"},"payload":"press_3","subtype":"press_3","topic":"zigbee2mqtt/0x0000000001718f82/action","type":"action"}'
debug 2024-09-02 14:39:32: Received Zigbee message from 'Stokstaart', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[51,22,255],"type":"Buffer"}},"commandID":19,"frameCounter":9637,"gppGddLink":255,"gppNwkAddr":5683,"options":21640,"payloadSize":0,"srcID":24219522}' from endpoint 242 with groupID null
debug 2024-09-02 14:39:32: No converter available for 'HG06337' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[51,22,255],"type":"Buffer"}},"commandID":19,"frameCounter":9637,"gppGddLink":255,"gppNwkAddr":5683,"options":21640,"payloadSize":0,"srcID":24219522}'
info  2024-09-02 14:39:32: MQTT publish: topic 'zigbee2mqtt/Stokstaart', payload '{"device":{"applicationVersion":66,"dateCode":"","friendlyName":"Stokstaart","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffea30e8b","manufacturerID":4098,"manufacturerName":"_TZ3000_kdi2o9m6","model":"HG06337","networkAddress":5683,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"indicator_mode":null,"last_seen":"2024-09-02T12:39:32.369Z","linkquality":199,"power_on_behavior":null,"state":"OFF"}'
debug 2024-09-02 14:39:32: Received Zigbee message from 'Bollie', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[244,62,255],"type":"Buffer"}},"commandID":19,"frameCounter":9637,"gppGddLink":255,"gppNwkAddr":16116,"options":21672,"payloadSize":0,"srcID":24219522}' from endpoint 242 with groupID null
debug 2024-09-02 14:39:32: No converter available for 'LED1624G9' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[244,62,255],"type":"Buffer"}},"commandID":19,"frameCounter":9637,"gppGddLink":255,"gppNwkAddr":16116,"options":21672,"payloadSize":0,"srcID":24219522}'
info  2024-09-02 14:39:32: MQTT publish: topic 'zigbee2mqtt/Bollie', payload '{"brightness":176,"color":{"x":0.249,"y":0.632},"color_mode":"xy","color_temp":312,"device":{"applicationVersion":17,"dateCode":"20220816","friendlyName":"Bollie","hardwareVersion":1,"ieeeAddr":"0x842e14fffea67a91","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"LED1624G9","networkAddress":16116,"powerSource":"Mains (single phase)","softwareBuildID":"2.3.093","stackVersion":87,"type":"Router","zclVersion":3},"last_seen":"2024-09-02T12:39:32.393Z","linkquality":247,"power_on_behavior":"on","state":"OFF","update":{"installed_version":587806257,"latest_version":587806257,"state":"idle"},"update_available":false}'
debug 2024-09-02 14:39:32: Received Zigbee message from '0x0000000001718f82', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":23,"frameCounter":9638,"options":0,"payloadSize":0,"srcID":24219522}' from endpoint 242 with groupID 2948
info  2024-09-02 14:39:32: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82', payload '{"action":"release_3","device":{"friendlyName":"0x0000000001718f82","ieeeAddr":"0x0000000001718f82","manufacturerID":null,"model":"PTM 215Z","networkAddress":24219522,"type":"GreenPower"},"last_seen":"2024-09-02T12:39:32.493Z","linkquality":127}'
info  2024-09-02 14:39:32: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82', payload '{"action":"","device":{"friendlyName":"0x0000000001718f82","ieeeAddr":"0x0000000001718f82","manufacturerID":null,"model":"PTM 215Z","networkAddress":24219522,"type":"GreenPower"},"last_seen":"2024-09-02T12:39:32.493Z","linkquality":127}'
info  2024-09-02 14:39:32: MQTT publish: topic 'homeassistant/device_automation/0x0000000001718f82/action_release_3/config', payload '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x0000000001718f82"],"manufacturer":"EnOcean","model":"Pushbutton transmitter module (PTM 215Z)","name":"0x0000000001718f82"},"payload":"release_3","subtype":"release_3","topic":"zigbee2mqtt/0x0000000001718f82/action","type":"action"}'
info  2024-09-02 14:39:32: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82/action', payload 'release_3'
debug 2024-09-02 14:39:32: Received MQTT message on 'homeassistant/device_automation/0x0000000001718f82/action_release_3/config' with data '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x0000000001718f82"],"manufacturer":"EnOcean","model":"Pushbutton transmitter module (PTM 215Z)","name":"0x0000000001718f82"},"payload":"release_3","subtype":"release_3","topic":"zigbee2mqtt/0x0000000001718f82/action","type":"action"}'
debug 2024-09-02 14:39:32: Received Zigbee message from 'Bollie', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[244,62,255],"type":"Buffer"}},"commandID":23,"frameCounter":9638,"gppGddLink":255,"gppNwkAddr":16116,"options":21672,"payloadSize":0,"srcID":24219522}' from endpoint 242 with groupID null
debug 2024-09-02 14:39:32: No converter available for 'LED1624G9' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[244,62,255],"type":"Buffer"}},"commandID":23,"frameCounter":9638,"gppGddLink":255,"gppNwkAddr":16116,"options":21672,"payloadSize":0,"srcID":24219522}'
info  2024-09-02 14:39:32: MQTT publish: topic 'zigbee2mqtt/Bollie', payload '{"brightness":176,"color":{"x":0.249,"y":0.632},"color_mode":"xy","color_temp":312,"device":{"applicationVersion":17,"dateCode":"20220816","friendlyName":"Bollie","hardwareVersion":1,"ieeeAddr":"0x842e14fffea67a91","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"LED1624G9","networkAddress":16116,"powerSource":"Mains (single phase)","softwareBuildID":"2.3.093","stackVersion":87,"type":"Router","zclVersion":3},"last_seen":"2024-09-02T12:39:32.545Z","linkquality":255,"power_on_behavior":"on","state":"OFF","update":{"installed_version":587806257,"latest_version":587806257,"state":"idle"},"update_available":false}'
debug 2024-09-02 14:39:32: Received Zigbee message from 'Stokstaart', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[51,22,255],"type":"Buffer"}},"commandID":23,"frameCounter":9638,"gppGddLink":255,"gppNwkAddr":5683,"options":216 [... cut ...]
debug 2024-09-02 14:39:32: No converter available for 'HG06337' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[51,22,255],"type":"Buffer"}},"commandID":23,"frameCounter":9638,"gppGddLink":255,"gppNwkAddr":5683,"options":21640,"payloadSize":0,"srcID":24219522}'
`

Where it's clear that the PTM215Z with address 1718f82 sends a few button presses and they are directly read

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

0x26680700

### Adapter

conbee II

### Setup

docker on synology ds216+II

### Debug log

[log_v140_truncated.log](https://github.com/user-attachments/files/16838866/log_v140_truncated.log)

posixx, 2024-09-02:
I have a PTM 215Z on my network and it works without issues running 1.39.1-1. So i think i will wait upgrading to 1.40 for now..

chris-1243, 2024-09-02:
It might be related to `deconz` driver. 
I am using Z2M 1.40 on `zstack` and all my PTM215Z are working as they should

coendaar, 2024-09-03:
@chris-1243 
I paired as follows=>
- click permit join all
- Hold upper right button for 7 sec
- Press the 2 diagonal opposed buttons

I don’t reallly get how the log in v1.31 can be full of 
`no converter available for xxx with cluster greenpower` and yet if still works. However in v1.40 i  get the same errors and it doesn’t work.

chris-1243, 2024-09-03:
As I told you and from Z2M 1.31 to the latest 1.40, some changes might have occured in `deconz` driver...

Would you try to pair one of your PTM215Z as suggested in this link

https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices

posixx, 2024-09-03:
It has to be the deconz driver as i upgraded to 1.40 and no problems here using Tubezb controller.

coendaar, 2024-09-03:
> As I told you and from Z2M 1.31 to the latest 1.40, some changes might have occured in `deconz` driver...
> 
> Would you try to pair one of your PTM215Z as suggested in this link
> 
> https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices

As requested:
Using v 1.36.1:
- force remove the ptm215z
- permit join to specific device. An IKEA LED1732G11, firmware 2.3.095 (latest OTA from both normal and test URL using z2mqtt). The zigbee2mqtt gui indicates this light has both an inputcluster and outputcluster for greenpower under endpoint 242
- Switch now works, relevant parts of log "plafond_eettafel" is the router the switch is paired to:
```
debug 2024-09-03 13:27:15Received Zigbee message from '0x0000000001718f82', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":19,"frameCounter":9702,"options":0,"payloadSize":0,"srcID":24219522}' from endpoint 242 with groupID 2948
info 2024-09-03 13:27:15MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82', payload '{"action":"press_3","device":{"friendlyName":"0x0000000001718f82","ieeeAddr":"0x0000000001718f82","manufacturerID":null,"model":"PTM 215Z","networkAddress":24219522,"type":"GreenPower"},"last_seen":"2024-09-03T11:27:13.881Z","linkquality":127}'
debug 2024-09-03 13:27:15Received Zigbee message from 'Bollie', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[244,62,255],"type":"Buffer"}},"commandID":19,"frameCounter":9702,"gppGddLink":255,"gppNwkAddr":16116,"options":21672,"payloadSize":0,"srcID":24219522}' from endpoint 242 with groupID null
debug 2024-09-03 13:27:15No converter available for 'LED1624G9' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[244,62,255],"type":"Buffer"}},"commandID":19,"frameCounter":9702,"gppGddLink":255,"gppNwkAddr":16116,"options":21672,"payloadSize":0,"srcID":24219522}'
debug 2024-09-03 13:27:15Received Zigbee message from 'plafond_eettafel', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[105,145,255],"type":"Buffer"}},"commandID":19,"frameCounter":9702,"gppGddLink":255,"gppNwkAddr":37225,"options":21672,"payloadSize":0,"srcID":24219522}' from endpoint 242 with groupID null
debug 2024-09-03 13:27:15No converter available for 'LED1732G11' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{"raw":{"data":[105,145,255],"type":"Buffer"}},"commandID":19,"frameCounter":9702,"gppGddLink":255,"gppNwkAddr":37225,"options":21672,"payloadSize":0,"srcID":24219522}'
```

Next steps:
- shut down z2mqtt (docker down)
- update to v1.40
- try switch

The switch doesn't work.  Log in v 1.40:
```

[2024-09-03 13:45:08] debug: 	zh:deconz:frameparser: GP data notification
[2024-09-03 13:45:08] debug: 	zh:deconz:frameparser: GP_DATA_INDICATION - src id: 0x1718f82 cmd id: 0x10 frameCounter: 0x25e8
[2024-09-03 13:45:08] debug: 	zh:controller:greenpower: Received unhandled command '0x10' from '24219522'
[2024-09-03 13:45:08] debug: 	zh:controller: Data is from unknown device with address '24219522', skipping...
[2024-09-03 13:45:08] debug: 	zh:deconz:frameparser: GP data notification
[2024-09-03 13:45:08] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2024-09-03 13:45:08] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-03 13:45:08] debug: 	zh:deconz:driver: query aps data indication
[2024-09-03 13:45:08] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 61
[2024-09-03 13:45:08] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 61 srcAddr: 0x8f82 destAddr: 0xdd09 profile id: 0xa1e0 cluster id: 0x21 lqi: 95
[2024-09-03 13:45:08] debug: 	zh:deconz:frameparser: response payload: 17,45,0,160,84,130,143,113,1,232,37,0,0,16,0,143,44,255
[2024-09-03 13:45:08] debug: 	zh:controller:greenpower: Received unhandled command '0x10' from '36738'
[2024-09-03 13:45:08] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-03 13:45:08] debug: 	zh:controller: Data is from unknown device with address '36738', skipping...
[2024-09-03 13:45:08] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2024-09-03 13:45:08] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-03 13:45:08] debug: 	zh:deconz:driver: query aps data indication
[2024-09-03 13:45:08] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 62
[2024-09-03 13:45:08] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 62 srcAddr: 0x3ef4 destAddr: 0x0 profile id: 0xa1e0 cluster id: 0x21 lqi: 255
[2024-09-03 13:45:08] debug: 	zh:deconz:frameparser: response payload: 17,24,0,168,84,130,143,113,1,232,37,0,0,16,0,244,62,255
[2024-09-03 13:45:08] debug: 	zh:controller:greenpower: Received unhandled command '0x10' from '16116'
[2024-09-03 13:45:08] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-03 13:45:08] debug: 	zh:controller: Data is from unknown device with address '16116', skipping...
[2024-09-03 13:45:08] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[... cut ...]
- the log clearly shows that  press (0x10) and release (0x14) message of upper left button are received, but then discarded.
- either None of the lights (ikea, lidl, adurosmart) support the required translation, or the deconz driver doesn't allow greenpower anymore? 
- This makes me wonder... If my lights are incapable of translating greenpower, then it must have been the deconz / conbeeII driver. And this change has happened between v1.36.1 and v1.37.0. However, I can't find anything about a new deconz driver in the v1.37.0 changelog?

chris-1243, 2024-09-03:
I had a look in the thread you referenced and it seems related to `deconz` only. 

I guess something is not working/coded correctly in `deconz`. My main network is using `zstack` without issue and I checked `ember` and everything works as it should be now with this driver as well.

The best way to know if your PTM215Z are paired as `unicast` is to sniff trafic. A recent issue found some problem with HUE devices and latest firmware. The list provided is not exhaustive and has to be completed for sure. If no ZigbeeGreen clusters, it should not work as a proxy.

coendaar, 2024-09-03:
Assuming the  `deconz` driver is in zigbee-herdsman,
- z2m v1.36.1 uses herdsman 0.40.3
- z2m v1.37.0 uses herdsman 0.45.0

The changelogs show significant refactoring of Greenpower between these versions. Most of it from @Koenkk 

But i would have expected other users to have also noticed this.

Nerivec, 2024-09-03:
@Koenkk I suggest starting by re-writing
https://github.com/Koenkk/zigbee-herdsman/blob/19757aefe1229bd677cda0011fd3691ca8da6abd/src/adapter/deconz/adapter/deconzAdapter.ts#L1231-L1268
to something like this:
```typescript
    private checkReceivedGreenPowerIndication(ind: gpDataInd): void {
        const gpdHeader = Buffer.alloc(15); // applicationId === IEEE_ADDRESS ? 20 : 15
        gpdHeader.writeUInt8(0b00000001, 0); // frameControl: FrameType.SPECIFIC + Direction.CLIENT_TO_SERVER + disableDefaultResponse=false
        gpdHeader.writeUInt8(ind.seqNr!, 1);
        gpdHeader.writeUInt8(ind.id!, 2); // commandIdentifier
        gpdHeader.writeUInt16LE(0, 3); // options, only srcID present
        gpdHeader.writeUInt32LE(ind.srcId!, 5);
        // omitted: gpdIEEEAddr (ieeeAddr)
        // omitted: gpdEndpoint (uint8)
        gpdHeader.writeUInt32LE(ind.frameCounter!, 9);
        gpdHeader.writeUInt8(ind.commandId!, 13);
        gpdHeader.writeUInt8(ind.commandFrameSize!, 14);

        // TODO: change `commandFrame` parsing type to Buffer to avoid unnecessary re-conversion
        const payBuf = Buffer.concat([gpdHeader, Buffer.from(ind.commandFrame!)]);
        const payload: Events.ZclPayload = {
            header: Zcl.Header.fromBuffer(payBuf),
            data: payBuf,
            clusterID: Zcl.Clusters.greenPower.ID,
            address: ind.srcId! & 0xffff,
            endpoint: ZSpec.GP_ENDPOINT,
            linkquality: 127,// bogus
            groupID: ZSpec.GP_GROUP_ID,
            wasBroadcast: true,// TODO: take the codepath that doesn't require `gppNwkAddr` (true) or not (false)?
            destinationEndpoint: ZSpec.GP_ENDPOINT,
        };

        this.waitress.resolve(payload);
        this.emit('zclPayload', payload);
    }
```

It should clean things up and align behavior with `ember`.

Koenkk, 2024-09-05:
Thanks @Nerivec !

I was able to reproduce the issue (`Failed to find green power proxy device`) with my Deconz adapter

@coendaar can you see if this fixes the issue:

```bash
docker exec -it CONTAINER_NAME sh
apk add nano
echo > /app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js
nano /app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js
# Update with https://gist.github.com/Koenkk/3f8a69b8aefeefa27a3a1ed5d13801a6
# save and exit nano
exit
docker restart CONTAINER_NAME
```

Koenkk, 2024-09-05:
@coendaar try to send the pairing command again (so holding button of the corresponding channel + energy bar for 7 secs)

coendaar, 2024-09-05:
Ok. 
I’ll test it in about an hour..

coendaar, 2024-09-05:
Ok. 
Without re-pairing.. a button press still results in an error:
```
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:frameparser: GP data notification
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:frameparser: GP_DATA_INDICATION - src id: 0x1718f82 cmd id: 0x13 frameCounter: 0x261c
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:controller:greenpower: Received unhandled command '0x13' from '36738'
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:controller: Data is from unknown device with address '36738', skipping...
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:frameparser: GP data notification
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:frameparser: DEVICE_STATE changed: 10101010
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:driver: query aps data indication
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 102
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 102 srcAddr: 0x8f82 destAddr: 0xdd09 profile id: 0xa1e0 cluster id: 0x21 lqi: 239
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:frameparser: response payload: 17,14,0,168,84,130,143,113,1,28,38,0,0,19,0,105,145,255
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:controller:greenpower: Received unhandled command '0x13' from '36738'
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:controller: Data is from unknown device with address '36738', skipping...
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:frameparser: DEVICE_STATE changed: 10101010
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:driver: query aps data indication
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 103
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 103 srcAddr: 0x1633 destAddr: 0x0 profile id: 0xa1e0 cluster id: 0x21 lqi: 175
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:frameparser: response payload: 17,70,0,136,84,130,143,113,1,28,38,0,0,19,0,51,22,255
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:controller:greenpower: Received unhandled command '0x13' from '5683'
zigbee2mqtt                 | [2024-09-05 13:42:31] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
```

coendaar, 2024-09-05:
after deleting the switch and permit join to an ikea https://www.zigbee2mqtt.io/devices/LED1733G7.html#ikea-led1733g7

the log is as follow:
```
zigbee2mqtt                 | [2024-09-05 13:51:02] info:       zh:controller:greenpower: Received commissioning from '44786'
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:controller:greenpower: Payload.Options: 58728 wasBroadcast: false
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz:driver: send read parameter request from queue. seqNr: 17 paramId: 1
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz:frameparser: MAC: 0x00212effff07ad37
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz:driver: send read parameter request from queue. seqNr: 18 paramId: 7
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz:frameparser: NWKADDR: 0
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz: no response expected (34)
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:controller: Green power device '{"sourceID":24219522,"deviceID":2,"networkAddress":36738}' joined
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:controller: Data is from unknown device with address '44786', skipping...
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz:driver: DATA_REQUEST - destAddr: 0xaef2 EP:242 SeqNr. 19 request id: 46
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 46 status: 0
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz:frameparser: DEVICE_STATE changed: 10100110
zigbee2mqtt                 | [2024-09-05 13:51:02] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```

Koenkk, 2024-09-05:
Updated https://gist.github.com/Koenkk/3f8a69b8aefeefa27a3a1ed5d13801a6, can you try again?

Koenkk, 2024-09-05:
@coendaar to be sure, can you update your conbee to the latest versions?

coendaar, 2024-09-05:
@Koenkk  sure. I’ll try later tonight when I’m back home. 

Just to be sure: https://www.zigbee2mqtt.io/guide/adapters/deconz.html still mentions an issue:
>Warning: Conbee 2 firmware versions newer than 0x26580700 will result in an unstable network with devices dropping randomly, see [Issue 9554](https://github.com/Koenkk/zigbee2mqtt/issues/9554)

Is it safe to update to https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26780700.bin.GCF ?

Koenkk, 2024-09-05:
It seems you are already on a new firmware than the problematic one so I guess it should be fine

coendaar, 2024-09-05:
- Updated conbee firmware to 0x26780700, restarted zigbee2mqtt v1.36.1.
- Everything still works in v1.36.1, including the wall switch.

- Updated docker to normal v1.40.0 (clean, unpatched)
- Switch not working
- Apply patch from the gist
- Wall switch is not working.
- (partial) log from a single button press in patched v1.40.0:
```
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:frameparser: DEVICE_STATE changed: 10101010
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:driver: query aps data indication
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 142
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 142 srcAddr: 0xaef2 destAddr: 0x0 profile id: 0xa1e0 cluster id: 0x21 lqi: 255
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:frameparser: response payload: 17,88,0,168,84,130,143,113,1,83,38,0,0,23,0,242,174,255
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:controller:greenpower: Received unhandled command '0x17' from '44786'
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:controller: Data is from unknown device with address '44786', skipping...
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:frameparser: DEVICE_STATE changed: 10101010
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:driver: query aps data indication
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 143
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 143 srcAddr: 0x8f82 destAddr: 0xb84 profile id: 0xa1e0 cluster id: 0x21 lqi: 255
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:frameparser: response payload: 17,94,0,168,84,130,143,113,1,82,38,0,0,19,0,9,7,13
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:controller:greenpower: Received unhandled command '0x13' from '36738'
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 21:22:24] debug:      zh:controller: Data is from unknown device with address '36738', skipping...
zigbee2mqtt                 | [2024-09-05 21:22:26] debug:      zh:deconz:driver: send read device state from queue. seqNr: 144
zigbee2mqtt                 | [2024-09-05 21:22:26] debug:      zh:deconz:frameparser: device state: 10100010
zigbee2mqtt                 | [2024-09-05 21:22:26] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
``` 

- force-remove wall switch
- permit join all.
- pairing seems succesfull:
```
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz:frameparser: GP commissioning notification
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz:frameparser: GP_DATA_INDICATION - src id: 0x1718f82 cmd id: 0xe0 frameCounter: 0x2660
zigbee2mqtt                 | [2024-09-05 21:28:02] info:       zh:controller:greenpower: Received commissioning from '24219522'
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:controller:greenpower: Payload.Options: 58696 wasBroadcast: true
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz: zclFrame to all - zclFrame.payload:
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz: [object Object]
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz: sendZclFrameToAll - message send
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 80 request id: 58
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz:frameparser: GP commissioning notification
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz:frameparser: DEVICE_STATE changed: 10101010
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz:driver: query aps data indication
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 58 status: 0
zigbee2mqtt                 | [2024-09-05 21:28:02] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[... cut ...]
- Looking at the first lines from the above log, it looks promising. There is a source ID, and a command in the frame-parser.
- However the controller does nothing with it.  Note that src id: 0x1718f82 == decimal 24219522'
```
zigbee2mqtt                 | [2024-09-05 21:31:30] debug:      zh:deconz:frameparser: GP_DATA_INDICATION - src id: 0x1718f82 cmd id: 0x13 frameCounter: 0x2662
zigbee2mqtt                 | [2024-09-05 21:31:30] debug:      zh:controller:greenpower: Received unhandled command '0x13' from '24219522'
```

Nerivec, 2024-09-05:
@Koenkk Since your last gist revision reverts the srcID/nwkAddr masking in ZclPayload, it shows the received data seems to come in sometimes as plain GP (shows srcID instead since no longer masked in ZclPayload):
```
zh:controller:greenpower: Received unhandled command '0x13' from '24219522'
...
zh:controller: Data is from unknown device with address '24219522', skipping...
```
and sometimes as proxied GP (shows proper nwkAddr):
```
zh:controller:greenpower: Received unhandled command '0x13' from '36738'
...
zh:controller: Data is from unknown device with address '36738', skipping...
```
It's receiving the commissioning in duplicate too, once as plain, and once as proxied GP.
So `wasBroadcast` could be either true/false depending, and the codepath would change accordingly.

Either way, it's acting like the device never joined?? so it never processes regular cluster commands.

Koenkk, 2024-09-05:
@coendaar could you provide the debug logging again of pairing + a command with:
- `deconzAdapter.js`: https://gist.github.com/Koenkk/3f8a69b8aefeefa27a3a1ed5d13801a6
- `/app/node_modules/zigbee-herdsman/dist/controller/controller.js`: https://gist.github.com/Koenkk/2e49c53e0b4de5345b33edadac418d7b

coendaar, 2024-09-05:
ok. here we go:

pairing (join all) after a force remove:
```
zigbee2mqtt                 | [2024-09-05 22:51:00] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"h2dd4-12"}'
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: GP data notification
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: GP_DATA_INDICATION - src id: 0x1718f82 cmd id: 0x13 frameCounter: 0x2666
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller:greenpower: Received unhandled command '0x13' from '36738'
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: GP data notification
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller: Receiced ZCL GP: 24219522 - 36738, got it from router 36738
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller: Receiced ZCL GP resolved: undefined
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller: Data is from unknown device with address '36738', skipping...
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: GP data notification
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: GP data notification
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: GP data notification
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: GP data notification
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: DEVICE_STATE changed: 10101010
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: DEVICE_STATE changed: 10101010
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:driver: query aps data indication
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 142
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 142 srcAddr: 0x8f82 destAddr: 0xb84 profile id: 0xa1e0 cluster id: 0x21 lqi: 255
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: response payload: 17,35,0,160,84,130,143,113,1,102,38,0,0,19,0,109,74,255
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller:greenpower: Received unhandled command '0x13' from '36738'
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 142 srcAddr: 0x8f82 destAddr: 0xb84 profile id: 0xa1e0 cluster id: 0x21 lqi: 255
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: response payload: 17,35,0,160,84,130,143,113,1,102,38,0,0,19,0,109,74,255
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller:greenpower: Received unhandled command '0x13' from '36738'
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller: Receiced ZCL GP: 24219522 - 36738, got it from router 36738
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller: Receiced ZCL GP resolved: undefined
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller: Data is from unknown device with address '36738', skipping...
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller: Receiced ZCL GP: 24219522 - 36738, got it from router 36738
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller: Receiced ZCL GP resolved: undefined
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:controller: Data is from unknown device with address '36738', skipping...
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: DEVICE_STATE changed: 10101010
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:frameparser: DEVICE_STATE changed: 10101010
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:driver: query aps data indication
zigbee2mqtt                 | [2024-09-05 22:51:03] debug:      zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 143
[... cut ...]
zigbee2mqtt                 | [2024-09-05 22:55:02] debug:      zh:controller: Data is from unknown device with address '19053', skipping...
zigbee2mqtt                 | [2024-09-05 22:55:05] debug:      zh:deconz:driver: send read device state from queue. seqNr: 213
zigbee2mqtt                 | [2024-09-05 22:55:05] debug:      zh:deconz:frameparser: device state: 10100010
zigbee2mqtt                 | [2024-09-05 22:55:05] debug:      zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```

Koenkk, 2024-09-05:
I have the impression that there is a bug in the undelete code:

```
zh:controller: GP device joined: 0x0000000001718f82 - 36738
# GP device joins with network address 36738

 z2m: Device '0x0000000001718f82' joined
 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0000000001718f82"},"type":"device_connected"}'
 z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0000000001718f82/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x0000000001718f82/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x0000000001718f82"],"manufacturer":"EnOcean","model":"Pushbutton transmitter module (PTM 215Z)","name":"0x0000000001718f82","via_device":"zigbee2mqtt_bridge_0x00212effff07ad37"},"enabled_by_default":true,"entity_category":"diagnostic","icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x0000000001718f82","name":"Action","object_id":"0x0000000001718f82_action","origin":{"name":"Zigbee2MQTT","sw":"1.40.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0x0000000001718f82","unique_id":"0x0000000001718f82_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
 z2m: Successfully interviewed '0x0000000001718f82', device has successfully been paired
 z2m: Device '0x0000000001718f82' is supported, identified as: EnOcean Pushbutton transmitter module (PTM 215Z)
 z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Pushbutton transmitter module","friendly_name":"0x0000000001718f82","model":"PTM 215Z","supported":true,"vendor":"EnOcean"},"type":"pairing"}'

# Now it tries to resolve the device with address 36738, which doesn't return any result.
zh:controller: Receiced ZCL GP: 24219522 - 36738, got it from router 36738
zh:controller: Receiced ZCL GP resolved: undefined
zh:controller: Data is from unknown device with address '36738', skipping...

```

To prevent the undelete from happening, can you: force remove the device from the frontend, stop z2m, start z2m, try again and provide the log of this?

Koenkk, 2024-09-06:
> but it looks like it gets a lót of duplicate messages.. So that is maybe the unicast/broadcast joining...

Yes but I would say this is a feature and not a bug. It means that every GPP proxy will now forward messages of this GP device, which allows you to move the device around freely. Since if only 1 GPP proxy would forward messages and the GP device gets out of range, you will not get any messages at all.

Note that although you get multiple messages for the same click, only 1 action is published (`{"action":"press_1"}`) since the `frameCounter` is the same. The reason why you see many `z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82'` is because you have enabled the last seen functionality, this will update the last seen whenever a Zigbee message is received from the device.

I've merged https://github.com/Koenkk/zigbee-herdsman/pull/1175 so the GPP part is fixed now. That still leaves the undelete bug which I cannot reproduce.

Can you:
- Update to the latest dev branch (this contains the GP fixes, so no need to patch `deconzAdapter.js` anymore
- `/app/node_modules/zigbee-herdsman/dist/controller/model/device.js` -> https://gist.github.com/Koenkk/1d37f0504c299d1605056191525480af
- Force remove the GP device through the frontend
- Pair it again
- Provide the log of this.

coendaar, 2024-09-06:
@Koenkk  Nice!
As requested:
- running version docker:latest-dev:
`1.40.0-dev commit: [13a108d](https://github.com/Koenkk/zigbee2mqtt/commit/13a108d)`
`zigbee-herdsman-converters version 20.11.0`
`zigbee-herdsman version 0.57.2`
- switch works. :)
- apply patch to /app/node_modules/zigbee-herdsman/dist/controller/model/device.js
- docker restart
-  force remove wallswitch:
```
[2024-09-06 21:26:50] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":true,"id":"0x0000000001718f82","transaction":"jbxfd-2"}'
[2024-09-06 21:26:50] info:     z2m: Removing device '0x0000000001718f82' (block: false, force: true)
[2024-09-06 21:26:50] debug:    zh:controller:database: Writing database to '/app/data/database.db'
[2024-09-06 21:26:50] debug:    z2m: Clearing Home Assistant discovery for '0x0000000001718f82'
[2024-09-06 21:26:50] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0000000001718f82/action/config', payload 'null'
[2024-09-06 21:26:51] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82', payload ''
[2024-09-06 21:26:51] info:     z2m: Successfully removed device '0x0000000001718f82' (block: false, force: true)
[2024-09-06 21:26:51] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0000000001718f82/linkquality/config', payload 'null'
[2024-09-06 21:26:51] debug:    zh:deconz:driver: send read device state from queue. seqNr: 128
[2024-09-06 21:26:51] debug:    zh:deconz:frameparser: device state: 10100010
[2024-09-06 21:26:51] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-06 21:26:51] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x0000000001718f82"},"status":"ok","transaction":"jbxfd-2"}'
[2024-09-06 21:26:51] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0000000001718f82/last_seen/config', payload 'null'
[2024-09-06 21:26:51] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/device_automation/0x0000000001718f82/action_press_3/config', payload 'null'
[2024-09-06 21:26:51] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/device_automation/0x0000000001718f82/action_release_3/config', payload 'null'
[2024-09-06 21:26:52] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/device_automation/0x0000000001718f82/action_press_1/config', payload 'null'
[2024-09-06 21:26:52] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/device_automation/0x0000000001718f82/action_release_1/config', payload 'null'
[2024-09-06 21:26:59] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2024-09-06 21:26:59] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-06 21:26:59] debug:    zh:deconz:driver: query aps data indication
[2024-09-06 21:26:59] debug:    zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 129
[2024-09-06 21:26:59] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 129 srcAddr: 0xdb5e destAddr: 0x0 profile id: 0x104 cluster id: 0x0 lqi: 255
[2024-09-06 21:26:59] debug:    zh:deconz:frameparser: response payload: 8,8,10,223,255,66,30,0,0,0,0,12,0,0,0,0,17,0,0,0,0,12,0,0,0,0,17,0,0,0,0,12,0,0,0,0,17
[2024-09-06 21:26:59] debug:    zh:controller:device: DEVICE BY NETWORK ADDRESS 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: ALL DEVICES 0x00212effff07ad37,0x00158d0007df9b74,0x000d6ffffe7aba27,0x9035eafffe082ea2,0x14b457fffe7e0883,0x000d6ffffed3d16c,0x086bd7fffe24b079,0x842e14fffea67a91,0x00158d0007df959e,0x00158d0007d3c924,0x00158d0007f02abc,0x842e14fffe94b9f8,0x842e14fffe2a0629,0x00158d0007f02aaf,0x00158d0007f02aa7,0x00158d0006e184cd,0x00158d0006e18185,0xa4c138c6b601fc85,0x00158d0008c11d51,0x00158d0008cea280,0x00158d0008cec9a1,0x00158d0008977c49,0xcc86ecfffefa675d,0x2c1165fffe68fe70,0xb4e3f9fffea30e8b,0x0c4314fffe51a812,0xa4c138b134c329ff,0x001788010e19a22f,0x001788010e19a21c
[2024-09-06 21:26:59] debug:    zh:controller:device: ALL DELETED DEVICES 0x0000000001718f82
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x00212effff07ad37 - 0 == 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x00158d0007df9b74 - 15593 == 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x000d6ffffe7aba27 - 19053 == 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x9035eafffe082ea2 - 11407 == 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x14b457fffe7e0883 - 37225 == 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x000d6ffffed3d16c - 44786 == 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x086bd7fffe24b079 - 38 == 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x842e14fffea67a91 - 16116 == 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x00158d0008c11d51 - 51342 == 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x00158d0008cea280 - 63462 == 56158 - false
[2024-09-06 21:26:59] debug:    zh:controller:device: CHECK 0x00158d0008cec9a1 - 52261 == 56158 - false
[... cut ...]
[2024-09-06 21:30:58] debug:    zh:controller:device: CHECK 0x00158d0007f02aa7 - 22053 == 35777 - false
[2024-09-06 21:30:58] debug:    zh:controller:device: CHECK 0x00158d0006e184cd - 8312 == 35777 - false
[2024-09-06 21:30:58] debug:    zh:controller:device: CHECK 0x00158d0006e18185 - 1801 == 35777 - false
[2024-09-06 21:30:58] debug:    zh:controller:device: CHECK 0xa4c138c6b601fc85 - 35777 == 35777 - true
[2024-09-06 21:30:58] debug:    zh:controller:device: MATCH!
```

- switch still works :)

Koenkk, 2024-09-06:
Looks good, seems that the added logging changes the behaviour. @Nerivec just did a nice improvement in this area (https://github.com/Koenkk/zigbee-herdsman/pull/1176). Can you check again with the latest-dev if all works? (without any changed, also check the force remove -> repair case)

coendaar, 2024-09-06:
testing:
`1.40.0-dev commit: [0d46354](https://github.com/Koenkk/zigbee2mqtt/commit/0d46354)`
`20.11.0`
`0.57.3`

force remove:
```
[2024-09-07 00:15:09] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":true,"id":"0x0000000001718f82","transaction":"xlhha-2"}'
[2024-09-07 00:15:09] info:     z2m: Removing device '0x0000000001718f82' (block: false, force: true)
[2024-09-07 00:15:09] debug:    zh:controller:database: Writing database to '/app/data/database.db'
[2024-09-07 00:15:09] debug:    z2m: Clearing Home Assistant discovery for '0x0000000001718f82'
[2024-09-07 00:15:09] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0000000001718f82/action/config', payload 'null'
[2024-09-07 00:15:10] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82', payload ''
[2024-09-07 00:15:10] info:     z2m: Successfully removed device '0x0000000001718f82' (block: false, force: true)
[2024-09-07 00:15:10] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0000000001718f82/linkquality/config', payload 'null'
[2024-09-07 00:15:10] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x0000000001718f82"},"status":"ok","transaction":"xlhha-2"}'
[2024-09-07 00:15:10] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x0000000001718f82/last_seen/config', payload 'null'
```

re-pair join all:
```
[2024-09-07 00:18:12] info:     z2m: Zigbee: allowing new devices to join.
[2024-09-07 00:18:12] debug:    zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffc EP:0 SeqNr. 156 request id: 100
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 100 status: 0
[2024-09-07 00:18:12] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 100 status: 0
[2024-09-07 00:18:12] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10100110
[2024-09-07 00:18:12] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10100110
[2024-09-07 00:18:12] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-07 00:18:12] debug:    zh:deconz:driver: query aps data confirm
[2024-09-07 00:18:12] debug:    zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 157
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0xfffc request id: 100 confirm status: 0
[2024-09-07 00:18:12] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-07 00:18:12] debug:    zh:deconz:driver: send write parameter request from queue. seqNr: 158 paramId: 33 param: 254
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: write parameter response - parameter id: 33 - status: 0
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: write parameter response - parameter id: 33 - status: 0
[2024-09-07 00:18:12] debug:    zh:deconz: PERMIT_JOIN - 254 seconds
[2024-09-07 00:18:12] debug:    zh:deconz: zclFrame to all - zclFrame.payload:
[2024-09-07 00:18:12] debug:    zh:deconz: [object Object]
[2024-09-07 00:18:12] debug:    zh:deconz: sendZclFrameToAll - message send
[2024-09-07 00:18:12] debug:    zh:deconz:driver: DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 159 request id: 101
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 101 status: 0
[2024-09-07 00:18:12] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 101 status: 0
[2024-09-07 00:18:12] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10100110
[2024-09-07 00:18:12] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10100110
[2024-09-07 00:18:12] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-07 00:18:12] debug:    zh:deconz:driver: query aps data confirm
[2024-09-07 00:18:12] debug:    zh:deconz:driver: DATA_CONFIRM - sending data state request - SeqNr. 160
[2024-09-07 00:18:12] debug:    zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0xfffd request id: 101 confirm status: 0
[2024-09-07 00:18:12] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2024-09-07 00:18:12] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"xlhha-6"}'
[... cut ...]
[2024-09-07 00:19:45] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82', payload '{"action":null,"device":{"friendlyName":"0x0000000001718f82","ieeeAddr":"0x0000000001718f82","model":"PTM 215Z","networkAddress":36738,"type":"GreenPower"},"last_seen":"2024-09-06T22:19:45.530Z","linkquality":255}'
[2024-09-07 00:19:45] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0000000001718f82', payload '{"action":null,"device":{"friendlyName":"0x0000000001718f82","ieeeAddr":"0x0000000001718f82","model":"PTM 215Z","networkAddress":36738,"type":"GreenPower"},"last_seen":"2024-09-06T22:19:45.530Z","linkquality":255}'
```

everything seems to work. *yay*

Nerivec, 2024-09-07:
Thanks for the feedback :wink:
`zh:controller:greenpower: Received unhandled command` in this scenario means the command is passed back to controller for regular command parsing (i.e. state update). That is the expected behavior.

closed 2024-09-07 by Nerivec

tluethi71, 2024-09-07:
still not working at my end.. in zigbee2Mqtt it shows the correct payload 
`Info 2024-09-07 17:44:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/zb2m_taster_buero_tom', payload '{"action":"press_1","elapsed":15285,"linkquality":255}'
Info 2024-09-07 17:44:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/zb2m_taster_buero_tom/action', payload 'press_1'
Info 2024-09-07 17:44:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/zb2m_taster_buero_tom', payload '{"action":"release_1","elapsed":256,"linkquality":255}'
Info 2024-09-07 17:44:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/zb2m_taster_buero_tom/action', payload 'release_1'
Info 2024-09-07 17:44:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/zb2m_taster_buero_tom', payload '{"action":"press_1","elapsed":4102,"linkquality":255}'
Info 2024-09-07 17:44:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/zb2m_taster_buero_tom/action', payload 'press_1'
Info 2024-09-07 17:44:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/zb2m_taster_buero_tom', payload '{"action":"release_1","elapsed":254,"linkquality":255}'
Info 2024-09-07 17:44:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/zb2m_taster_buero_tom/action', payload 'release_1'`

but my mqtt device does only show the linkquality
[image]

Nerivec, 2024-09-07:
@tluethi71 Looks like something didn't get updated. Try this:
- Stop z2m
- Restart mqtt broker
- Start z2m
- Re-interview the device from z2m frontend

tluethi71, 2024-09-08:
@Nerivec thank you for you answer.  Did not work.

- Stop z2m in Add-ons -> OK
- Restart mqtt broker in Add-ons -> OK
- Start z2m in Add-ons -> OK
- Re-interview  the device from z2m frontend (push the i button) -> did not work 
`Error 2024-09-08 12:12:56z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'zb2m_taster_pendelleuchte' (0x0000000001567624) failed: Error: Interview failed because can not get active endpoints ('0x0000000001567624')'`

and still not showing the status in mqtt.

Nerivec, 2024-09-08:
Did you try removing it and re-pairing it instead? These GP devices have a reset procedure, do that too, that will ensure you have a clean slate.

tluethi71, 2024-09-08:
Hey @Nerivec, yes i did.
- first force remove of the device
- re-paired.  Apeared again in z2m.. 
 `Info 2024-09-08 14:24:16z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/zb2m_taster_buero_tom', payload '{"action":"release_4","elapsed":261,"linkquality":255}'
Info 2024-09-08 14:24:16z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/zb2m_taster_buero_tom/action', payload 'release_4'`

- But still does not show in mqtt.. only link quality (see above)

Nerivec, 2024-09-08:
What do you see if you click on `MQTT INFO` (in your screenshot) for the various `Payload` in Entities/Triggers?

tluethi71, 2024-09-08:
see the pic  and the triggers: 
[image]

triggers:

`Triggers

MQTT discovery data:
Topic: homeassistant/device_automation/0x0000000001567628/action_press_4/config
Payload
automation_type: trigger
device:
  identifiers:
    - zigbee2mqtt_0x0000000001567628
  manufacturer: EnOcean
  model: Pushbutton transmitter module (PTM 215ZE)
  via_device: zigbee2mqtt_bridge_0x00212effff0985cb
  name: zb2m_taster_buero_tom
origin:
  name: Zigbee2MQTT
  sw_version: 1.40.1
  support_url: https://www.zigbee2mqtt.io
payload: press_4
subtype: press_4
topic: zigbee2mqtt/zb2m_taster_buero_tom/action
type: action
platform: mqtt

MQTT discovery data:
Topic: homeassistant/device_automation/0x0000000001567628/action_release_4/config
Payload
automation_type: trigger
device:
  identifiers:
    - zigbee2mqtt_0x0000000001567628
  manufacturer: EnOcean
  model: Pushbutton transmitter module (PTM 215ZE)
  via_device: zigbee2mqtt_bridge_0x00212effff0985cb
  name: zb2m_taster_buero_tom
origin:
  name: Zigbee2MQTT
  sw_version: 1.40.1
  support_url: https://www.zigbee2mqtt.io
payload: release_4
subtype: release_4
topic: zigbee2mqtt/zb2m_taster_buero_tom/action
type: action
platform: mqtt

MQTT discovery data:
Topic: homeassistant/device_automation/0x0000000001567628/action_press_3/config
Payload
automation_type: trigger
device:
  identifiers:
    - zigbee2mqtt_0x0000000001567628
  manufacturer: EnOcean
  model: Pushbutton transmitter module (PTM 215ZE)
  via_device: zigbee2mqtt_bridge_0x00212effff0985cb
  name: zb2m_taster_buero_tom
origin:
  name: Zigbee2MQTT
  sw_version: 1.40.1
  support_url: https://www.zigbee2mqtt.io
payload: press_3
subtype: press_3
topic: zigbee2mqtt/zb2m_taster_buero_tom/action
type: action
platform: mqtt

MQTT discovery data:
Topic: homeassistant/device_automation/0x0000000001567628/action_release_3/config
Payload
automation_type: trigger
device:
  identifiers:
    - zigbee2mqtt_0x0000000001567628
  manufacturer: EnOcean
  model: Pushbutton transmitter module (PTM 215ZE)
  via_device: zigbee2mqtt_bridge_0x00212effff0985cb
  name: zb2m_taster_buero_tom
origin:
  name: Zigbee2MQTT
  sw_version: 1.40.1
  support_url: https://www.zigbee2mqtt.io
payload: release_3
subtype: release_3
topic: zigbee2mqtt/zb2m_taster_buero_tom/action
type: action
platform: mqtt`

does it help?

Koenkk, 2024-09-08:
@tluethi71 what HA version are you using? (can you try updating to the latest)

tluethi71, 2024-09-08:
@Koenkk 
sure
- HA Version: 2024.9.1 (latest)
- z2m configuration.yaml. Do you need the details of the devices too? its a long file..
- `homeassistant:
  legacy_entity_attributes: false
  legacy_triggers: false
advanced:
  network_key:
    - 149
    - 171
    - 188
    - 103
    - 58
    - 96
    - 184
    - 8
    - 229
    - 53
    - 47
    - 80
    - 134
    - 182
    - 153
    - 74
  pan_id: 8309
  ext_pan_id:
    - 23
    - 124
    - 59
    - 213
    - 181
    - 148
    - 22
    - 121
  log_level: debug
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 514
    protocol: udp4
    type: '5424'
  legacy_api: false
  legacy_availability_payload: false
  elapsed: true
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: ae3uxil5Wohxeze5zaM8thuseez9jieNg3biegei1Weiph2ui4esh5ieWai6Aing
  keepalive: 60
  reject_unauthorized: true
  version: 4
serial:
  port: >-
    /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2671447-if00
  adapter: deconz
  disable_led: true
frontend:
  port: 8099
devices:
 .....
blocklist: []
permit_join: false
`

tluethi71, 2024-09-10:
sure.. i did a restart of the whole system

Koenkk, 2024-09-12:
@tluethi71 please upload the full log as a file

tluethi71, 2024-09-12:
@Koenkk see attachment. the full log since yesterday afternoon after restart..
[log.log](https://github.com/user-attachments/files/16984394/log.log)

tluethi71, 2024-09-13:
@Koenkk arrrrghh... now starting with `[2024-09-13 17:10:34] info: 	z2m: Logging to console, file (filename: log.log)`
sorry about that..
[log.log](https://github.com/user-attachments/files/16995953/log.log)

Koenkk, 2024-09-14:
Found the issue, settings `legacy_triggers: true` should fix the issue.

tluethi71, 2024-09-14:
perfect!!! It works perfectly.. I needed also to create an automation to trigger the actions (see https://github.com/Koenkk/zigbee2mqtt/issues/22578#issuecomment-2351170759)