Koenkk/zigbee2mqtt #23815: No units defined for TS0207 / RB-SRAIN01 Tuya rain sensor leading to wrong(?) presentation in Home Assistant

richardgaywood, 2024-09-02:
### What happened?

The newly-added-in-1.40.0 rain sensor, when imported into HA, has no defined units of measurement; hence it looks like this when you view it:

[image]

Obv. it'd be better if this was a line graph!

### How to reproduce it (minimal and precise)

Just set up the device, no other repro steps needed. 

Pretty sure the root cause is in [this commit](https://github.com/Koenkk/zigbee-herdsman-converters/commit/805f5a46cc6bfdf93ac392aebfb6258fa8242fb7#diff-0157bb612c63e92c50f93116e1f50bf8e263030ed26d86caa28479159c2a9f4a):

```json
exposes: [
    e.illuminance().withUnit('lx'),
    e.numeric('illuminance_average_20min', ea.STATE).withUnit('lx').withDescription('Illuminance average for the last 20 minutes'),
    e.numeric('illuminance_maximum_today', ea.STATE).withUnit('lx').withDescription('Illuminance maximum for the last 24 hours'),
    e.binary('cleaning_reminder', ea.STATE, 'ON', 'OFF').withDescription('Cleaning reminder'),
    e.numeric('rain_intensity', ea.STATE).withDescription('Rainfall intensity'),
    e.battery(),
],
```

Note that `rain_intensity` has no `withUnit()` call. I think that's what makes HA default to the unitless mode where it treats the value as discrete categories instead of a continuous number range.

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20210708

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_acea7ed057beed1196ba622e38a92db5

### Setup

Home Assistgant OS add-on, on a Dell OptiPlex Micro 3050

Koenkk, 2024-09-02:
What unit should it be?

sajikur, 2024-09-03:
his device completely changed the exposed sensors in HA during the last update. It was previously recognized as "https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html#tuya-ts0207_water_leak_detector" and in addition to its solar sensors, it exposed the "water_leak". The "rain intensity" gives a value of 2896 if the sensor gets wet for testing which makes little sense for developing automations. Is it therefore possible to expose the old sensors again?

thank you

richardgaywood, 2024-09-03:
The raw sensor itself is outputting values in mV, because it's the "amount of volts necessary to create a short" value that it uses to indicate the sensor is wet. That's a fairly weird value to show to the user in the UI, but I don't have a better idea... It seems that you have to set a unit or Home Assistant won't treat the field as a number (which is also weird to me.)

One option could be as follows: from reading [this review](https://smarthomescene.com/reviews/tuya-zigbee-solar-rain-sensor-rb-srain01-review/) and [the spec sheet](https://developer.tuya.com/en/docs/iot/ztu-module-datasheet?id=Ka45nl4ywgabp), it sounds like this is how it responds normally:

> I discovered that the mV (millivolt) reading changes non-linearly when water is introduced. With a single drop of water, it will immediately spike to 2000mV. Adding 4-5 more drops to the plate brings the value up to 2600mV. Completely submerging the sensor gives me a maximum reading of 3000-3100mV. It also has a base value of 5-7mV when dry that needs to be ignored.

I suspect these voltage values are relative to the 3.3V VCC output pin on the chip. So maybe the best approach is to divide the current reading from the chip by 3.3V and report that as a percentage value, called, I dunno, `rain_sensor_conductivity` or similar. I'd be wary of trying to map the value to intensity bands inside Z2M though, feels like the kind of thing where two sensors might have very different calibrations. 

FWIW, ZHA [seems to call the mV reading](https://github.com/zigpy/zha-device-handlers/issues/3249) `rain_sensor_voltage`, which I guess is maybe a little clearer if you have to put it in mV. 

Also, unrelated, but -- as sajikur just wrote above as I was typing this -- [the commit to Z2M](https://github.com/Koenkk/zigbee-herdsman-converters/commit/805f5a46cc6bfdf93ac392aebfb6258fa8242fb7#diff-0157bb612c63e92c50f93116e1f50bf8e263030ed26d86caa28479159c2a9f4a) doesn't seem to have the binary "water detected" sensor the device used to expose; you can see it in the [sample code here](https://smarthomescene.com/reviews/tuya-zigbee-solar-rain-sensor-rb-srain01-review/) (`water_leak`). Was omitting that intentional?

Koenkk, 2024-09-03:
- Added the unit
- Water leak has already been added in https://github.com/Koenkk/zigbee-herdsman-converters/commit/c7f5aa27d299caf382666f07992c4dba3878a51e

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-03 by Koenkk