Koenkk/zigbee2mqtt #23816: FW upgrade breaks Z2M (SONOFF DONGLE P)

tbrasser, 2024-09-02:
### What happened?

Referenced from reply on slightly different error. 

Facing this after updating fw:

```
[13:19:54] INFO: Preparing to start...
[13:19:54] INFO: Socat not enabled
[13:19:55] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-09-02 13:19:56] info: 	z2m: Logging to console, file (filename: log.log)
[2024-09-02 13:19:56] info: 	z2m: Starting Zigbee2MQTT version 1.40.0 (commit #unknown)
[2024-09-02 13:19:56] info: 	z2m: Starting zigbee-herdsman (0.57.1)
[2024-09-02 13:19:56] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-09-02 13:19:56] info: 	zh:zstack:znp: Serialport opened
[2024-09-02 13:20:03] error: 	z2m: Error while starting zigbee-herdsman
[2024-09-02 13:20:03] error: 	z2m: Failed to start zigbee
[2024-09-02 13:20:03] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-09-02 13:20:03] error: 	z2m: Exiting...
[2024-09-02 13:20:03] error: 	z2m: Error: SRSP - SYS - stackTune after 6000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:300:45
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:26)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1111:13
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:165:13)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:138:29)
    at Zigbee.start (/app/lib/zigbee.ts:64:27)
    at Controller.start (/app/lib/controller.ts:140:27)
    at start (/app/index.js:154:5)
```

flashed the dongle with:

```
sudo podman run --rm --device=/dev/ttyUSB0 --privileged --group-add "$(stat --format=%g /dev/ttyUSB0)" -v /dev/:/dev/ -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip ckware/ti-cc-tool:2024-03-19 -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb
Resolved "ckware/ti-cc-tool" as an alias (/var/cache/containers/short-name-aliases.conf)
Trying to pull docker.io/ckware/ti-cc-tool:2024-03-19...
Getting image source signatures
Copying blob e0183997aaa7 done   | 
Copying blob 4abcf2066143 skipped: already exists  
Copying blob 2aa9be932802 done   | 
Copying blob c3cdf40b8bda skipped: already exists  
Copying blob 09f577b6635e done   | 
Copying blob ae52dd4ac952 done   | 
Copying blob 48d1cb2b9803 done   | 
Copying blob 4377ce580b2c done   | 
Copying blob 4f4fb700ef54 skipped: already exists  
Copying config 88726657f6 done   | 
Writing manifest to image destination
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip
Firmware source: 'CC1352P2_CC2652P_launchpad_coordinator_20240710.zip'
Firmware file:   'CC1352P2_CC2652P_launchpad_coordinator_20240710.hex'
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20240710.hex
Firmware file: Intel Hex
Connecting to target...
CC1350 PG2.1 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:29:DE:23:D8
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xd9dd0124)
```

_Originally posted by @tbrasser in https://github.com/Koenkk/zigbee2mqtt/issues/23785#issuecomment-2324499305_
            

### What did you expect to happen?

Running system

### How to reproduce it (minimal and precise)

Flash with latest FW then try any version of Z2M

### Zigbee2MQTT version

1.40, dev

### Adapter firmware version

2024

### Adapter

Sonoff dongle P

### Setup

Home Assistant Addon

tbrasser, 2024-09-02:
Cleaned up my config.yaml and did a full reboot of the system. Everything working fine now!

WizBangCrash, 2024-09-02:
@tbrasser when you say "cleaned up my config.yaml" was there anything specific you did?

kenschbacher9, 2024-09-04:
> @tbrasser when you say "cleaned up my config.yaml" was there anything specific you did?

@WizBangCrash : Did you get an answer or solve this issue? Unfortunately, I have the same problem and no solution yet.

tbrasser, 2024-09-04:
I removed anything refering to cache/queue persisting.

WizBangCrash, 2024-09-04:
Thanks @tbrasser. I tried similar things. I think it was the cache/queue persisting change that finally got the updated firmware to work. TBH I'm not entirely sure what caused it to finally work as after the first day I stopped doing things logically and was in panic mode 🤣
I had to re-pair all my devices too. 50 of them and some in very inaccessible places.
Lost two days of "Smart Home" and had a family threatening to put me out on the street 😱

Has definitely made me rethink my strategy of using more Zigbee devices though. Not a good experience. Especially the re-pairing as even after devices rejoined the network they would fall out again as I re-paired other devices.

I'm planning to get more than 3 hours sleep tonight now its working again 🥴

ethcz, 2024-09-23:
I experienced the same issue.
I have adapter Sonoff ZBDongle-P. I was running firmware CC1352P2_CC2652P_launchpad_coordinator_20230507 without issues.
After flashing the firmware CC1352P2_CC2652P_launchpad_coordinator_20240710, the Z2M failed starting with the error: 
`error: 	z2m: Error: SRSP - SYS - stackTune after 6000ms`

After flashing 20230507 back, the error was still there. I also deleted  coordinator_backup.json, database.db and state.json, but without any progress.
 
The only thing which helped was shrinking `configuration.yaml` to [minimal configuration example](https://www.zigbee2mqtt.io/guide/configuration/) and then Z2M started to work again. I was then putting back my previous additional configuration lines to  `configuration.yaml`, ending up at the full previous configuration, which was failing right after the flashing of the firmware 20240710 for the first time. 

Also I had to re-pair all devices.

jtig2019, 2024-11-26:
since today zigbee2mqtt no longer starts for me either:
`{"log":"[2024-11-26 09:05:52] \u001b[31merror\u001b[39m: \u0009z2m: Error while starting zigbee-herdsman\n","stream":"stdout","time":"2024-11-26T08:05:52.907684344Z"}
{"log":"[2024-11-26 09:05:52] \u001b[31merror\u001b[39m: \u0009z2m: Failed to start zigbee\n","stream":"stdout","time":"2024-11-26T08:05:52.917411888Z"}
{"log":"[2024-11-26 09:05:52] \u001b[31merror\u001b[39m: \u0009z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible so${"log":"[2024-11-26 09:05:52] \u001b[31merror\u001b[39m: \u0009z2m: Exiting...\n","stream":"stdout","time":"2024-11-26T08:05:52.92165832Z"}
{"log":"[2024-11-26 09:05:53] \u001b[31merror\u001b[39m: \u0009z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)\n","stream":"stdout","${"log":"    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)\n","stream":"stdout","time":"2024-11-26T08:05:5${"log":"    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)\n","stream":"stdout","time":"2024-11-26T08:05:53.362909637Z"}
{"log":"    at Zigbee.start (/app/lib/zigbee.ts:69:27)\n","stream":"stdout","time":"2024-11-26T08:05:53.363060105Z"}
{"log":"    at Controller.start (/app/lib/controller.ts:161:27)\n","stream":"stdout","time":"2024-11-26T08:05:53.36309099Z"}
{"log":"    at start (/app/index.js:154:5)\n","stream":"stdout","time":"2024-11-26T08:05:53.363120521Z"}`

but i have not performed a firmupdate on my sonoff dongle plus p.

i deleted the coordinator_backup.json, database.db and state.json and minimized the config.yaml to the basic config.

unfortunately this did not help, the error still occurs. 

ls /dev/serial/by-id/ shows the adapter with the port that is also stored in config.yaml

could it be a hardware defect of the adapter? what else could i try? try to flash the adapter?

thanks for your help and ideas :)

ethcz, 2024-11-26:
[jtig2019](https://github.com/jtig2019): You are getting `Error: SRSP - SYS - ping after 6000ms`, but this thread discusses `Error: SRSP - SYS - stackTune after 6000ms`. Please try to look into different threads.