Koenkk/zigbee2mqtt #23817: Routers based on Texas Instrument eg CC2652 Constantly Going Offline

harvindhillon, 2024-09-02:
### What happened?

I think this is a follow up to this older ticket 
https://github.com/Koenkk/zigbee2mqtt/issues/13758

Basically I have availability enabled and have set the interval for routers to 35Mins. I have set this as low as 20Mins.
Only TI based routers (I have a few Sonoff Dongle P and SLZB-06 flashed as routers - All same firmware). They periodically go offline.

Restarting Z2M doesnt fix the issue, however restarting the container usually does. If you try to re-pair the TI based routers, it will fail constantly until it succeeds at some point. Once paired, you are briefly able to adjust the radio power before timing out again.

### What did you expect to happen?

To work normally as a router without timeouts

### How to reproduce it (minimal and precise)

Have multiple TI based routers, have availability turned on and wait.

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Docker Container on x86 based Intel NUC

habitats-tech, 2024-09-04:
To my experience this is a 20240710 FW issue. The solution I have found is to just reboot the device (ESP32 or sometimes just restart the Zigbee chipset). It take a couple of minutes for devices to start reporting again.

I also use routers and I have found the issue is with the coordinator (both coordinator and routers use the same HW - SLZB-06P7).

harvindhillon, 2024-09-04:
Actually I dont think it is related to 20240710 because I had this issue on previous releases including beta releases of the firmware. As in the older (stale) ticket, it happens periodically sometimes being fine for days then suddenly dropping every 15mins.

Restarting the Z2M container 99% of the time fixes it so it does look like an issue with Z2M maybe?

habitats-tech, 2024-09-04:
You have a point. I have tried FW 20240716 on SLZB-06P7 with the same issue. Device randomly requires a restart (either Zigbee or ESP32) in order to reconnect to Z2M. However on latest FW, Z2M does not crash, it just keeps going, only requiring a coordinator reboot.

Also updating Z2M to 1.40 has a welcome UI enhancement.

[image]

stavros-k, 2024-10-03:
[image]
Mine says online but last seen 12h ago.
And of course the status does not update.

closed 2024-12-20 by github-actions[bot]