Koenkk/zigbee2mqtt #23818: [Wrong device]: Elivco LSPA9 wrongly detected as Tuya BSD29_1

Trashwarez, 2024-09-02:
### Link

https://www.zigbee2mqtt.io/devices/LSPA9.html

### Model

LSPA9

### Description

Smart Plug

### Vendor

Elivco

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/LSPA9.png

### Database entry

{"id":45,"type":"Router","ieeeAddr":"0x70b3d52b60033c68","nwkAddr":45026,"manufId":4660,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,6,3,4,5,57345,2820,1794],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"manufacturerName":"_TZ3000_okaz9tjs","zclVersion":2,"appVersion":100,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":224,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,3051]}},"genOnOff":{"attributes":{"onOff":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT31BS102","zclVersion":2,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1725296200768}

Trashwarez, 2024-09-09:
As the detection of my device seems kind of broken, I use an external converter to fix it:

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_okaz9tjs'}],
    model: 'LSPA9',
    description: 'Smart plug (with power monitoring)',
    vendor: 'Elivco',
    ota: ota.zigbeeOTA,
    extend: [
        tuya.modernExtend.tuyaOnOff({
            electricalMeasurements: true,
            powerOutageMemory: true,
            indicatorMode: true,
            childLock: true,
            onOffCountdown: true,
        }),
    ],
};

module.exports = definition;
`

alexsotoaguilera, 2024-10-14:
This converter works for me. Thanks a lot

alexsotoaguilera, 2024-10-14:
Indicator mode does nothing. Is working in your devices?

Trashwarez, 2024-10-14:
I made this converter just as a quick fix until someone cares about the bug. Unfortunately nobody did that by now.

LSPA9 been told to work, but doesn't at least on my system. See here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/7684

covers007, 2024-11-19:
Thanks for this converter. It's OK for me

Timieh, 2024-11-30:
Same issue here on 2 new deliverd 
Elivco LSPA9.

Trashwarez, 2025-01-30:
Push!

closed 2025-04-09 by github-actions[bot]