Koenkk/zigbee2mqtt #23820: "Last Seen" is broken in several ways

Ltek, 2024-09-02:
### What happened?

I've only used Z2M for about a year and happens with at least the last 3 releases since I started really using (adding devices) Z2M. This problem occurs with all my 22 devices so its not a device issues : Bosch Motion, Aqara Motion, Tuya Humidity/Temp, Tuya Covers, and more

Last Seen updates on all devices even if the Device Map shows them not connected. Many show Last Seen but never update the device parameters... so clearly Z2M is not "seeing" (communicating) with them otherwise the temp, motion, etc would be updated.

### What did you expect to happen?

1. A user setting to enable Just Scene to update -only- when a parameter updates. This way we know for sure its communicating since devices can/do drop off even when LQI is in the 200s.

2. A way to enable Z2M to write log file with -only- the device parameter updates (like a device activity log), including the 'just seen' with what is triggering 'just seen'.

### How to reproduce it (minimal and precise)

Watch Dashboard for any device and see Last Seen change when no parameter has changed

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

EmberZNet Coordinator revision 7.4.3 [GA]

### Adapter

Ember

### Setup

Homeseer on Windows

Ltek, 2024-09-22:
here's another example of how its broken in another way... looks at the time stamp of Last Seen vs all the others... Last Seen is after 6 pm... its NOON!  All other time stamps are correct

djurny, 2024-12-13:
Hi there,
Just curious, in what timezone do you reside? Do the timestamps make sense when you take your local timezone into account?
Groetjes,

Ltek, 2024-12-13:
> Hi there, Just curious, in what timezone do you reside? Do the timestamps make sense when you take your local timezone into account? Groetjes,

Since all the other device time stamps are reporting correctly, and I dont seen any Z2M time zone settings, I must assume time zone is not the problem

closed 2025-02-21 by github-actions[bot]