Koenkk/zigbee2mqtt #23821: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')

Zayeen, 2024-09-02:
### What happened?

In my setup the Tuya TV02-Zigbee is the cause somehow. If i try to change a setting or it updates the error comes up. 

### How to reproduce it (minimal and precise)

Change a setting in Z2M 

### Zigbee2MQTT version

1.40.0-1

### Adapter firmware version

20210120

### Adapter

zzh!

### Setup

Hass os vm in Proxmox

### Debug log

[2024-09-02 22:01:32] error: 	z2m: EventBus error 'Publish/mqttMessage': Cannot read properties of undefined (reading 'some')
[2024-09-02 22:01:32] debug: 	z2m: TypeError: Cannot read properties of undefined (reading 'some')
    at /app/lib/extension/publish.ts:326:95
    at Array.some (<anonymous>)
    at /app/lib/extension/publish.ts:326:77
    at Array.some (<anonymous>)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:326:61)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)

Koenkk, 2024-09-02:
Can you check if this is fixed?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

jesse-kaufman, 2024-09-02:
> Can you check if this is fixed?
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Looks like it's fixed to me

closed 2024-09-02 by Koenkk

Zayeen, 2024-09-02:
> Can you check if this is fixed?
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Looks like it's fixed with dev branch , thanks.

rickoltan, 2024-09-05:
Installing the dev environment does not work, I get the error "fatal: --unshallow on a complete repository does not make sense" when trying to do a fetch with --unshallow.
Will there be a master release with this correction soon?

silkyclouds, 2024-09-05:
oh nice when will it come to the official home assistant add-on repo ?