Koenkk/zigbee2mqtt #23823: Awox bulbs getting constantly disconnected after the Sonoff Dongle P update to 20240710 + Z2M update to 1.40

pbartek20, 2024-09-02:
### What happened?

Multiple of my Awox TLSR82xx bulbs are getting frequently/almost constant disconnected as soon as I get them back online by pull from the lamp and bring back despite LQI ~=>100
Some of them are more likely to drop (like the pasted below) and actually not recovering, some drop and after 10/20/30 min will came up (either by themselves or by my manually forcing clicks state update) just to after few/10/20 minutes go down again.
For the below ones as I wrote on the beginning I can only bring them back by pulling physically from lamp and put back but after few minutes same happening.

[image]

```
2024-09-02 23:26:32z2m: Failed to ping 'light_bedroom_left1' (attempt 1/1, ZCL command 0xa4c1384053a4b15c/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 53779 - 1 - 216 - 0 - 1 after 10000ms))
2024-09-02 23:27:10z2m: Failed to ping 'light_bedroom_right1' (attempt 1/1, ZCL command 0xa4c1389116369b84/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 47872 - 1 - 218 - 0 - 1 after 10000ms))
2024-09-02 23:28:04z2m: Failed to ping 'light_bedroom_right2' (attempt 1/1, ZCL command 0xa4c13844398f0f77/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 9839 - 1 - 219 - 0 - 1 after 10000ms))
2024-09-02 23:36:43z2m: Failed to ping 'light_bedroom_left1' (attempt 1/1, ZCL command 0xa4c1384053a4b15c/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 53779 - 1 - 255 - 0 - 1 after 10000ms))
2024-09-02 23:37:21z2m: Failed to ping 'light_bedroom_right1' (attempt 1/1, ZCL command 0xa4c1389116369b84/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 47872 - 1 - 2 - 0 - 1 after 10000ms))
2024-09-02 23:38:15z2m: Failed to ping 'light_bedroom_right2' (attempt 1/1, ZCL command 0xa4c13844398f0f77/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 9839 - 1 - 3 - 0 - 1 after 10000ms))
```

I had similar kind of problem with these bulbs around 2 years back but (not remember which exact) one of the adapter newer adapter firmware update, possibly ~1,5h years back, which was in notes mentioned that it fixed the handling of bigger amount of devices environment actually fixed the problem. After that I got only occasional drops. Good working continued  unfortunately until this update hence the problem must lie here I assume.

I already tried rebooting Add-on, Home assistant, the whole Host and the result is same.

### What did you expect to happen?

Some next adapter or Z2M firmware udate will fix it as it was previously working properly

### How to reproduce it (minimal and precise)

Only having big amount of Z2M devices (70 in my situation 43 router capable devices (bulbs or plugs)) or the Awox TLSR82xx bulbs

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle-P

### Setup

Add-on on Home assistant on Proxmox VM (Gowin R48S)

coendaar, 2024-09-03:
Have you tried updating the bulbs firmware? There is warning about network performance in the docs:
https://www.zigbee2mqtt.io/devices/33957.html#warning-degrades-network-performance

My awox 33957 was messing with my network, but after upgrading to firmware 0122052017 using the awox homecontrol app, it's significantly better.

pbartek20, 2024-09-03:
Yes, for every bulb before pairing in Z2M I was adding 2 years back firstly I added them in the app just to update to the latest firmware which is same as you wrote 0122052017.

Like I described in the issue I had problems for some time but one of the adapter updates which had a note that improves the reliability in larger environments almost 100% fixed the issue for me (it was around 1,5 years back, not remember which was the exact version). After that through all newer versions all was working almost flawlessly until last update.

Now after I updated 2 days back the stick to 20240710 and Z2M to 1.40 similar issue from the start is visible again.

pbartek20, 2024-09-06:
Yesterday I downgraded the dongle firmware from 20240710 to 20230507 and since that my network is fully stable as it was before upgrade.
This confirms that the 20240710 adapter firmware is causing the issue with Awox bulbs and overall instability of the network. Many other devices during these days were disconnecting as well (Moes shade motor [AM43-0.45/40-ES-EB], Aquara water leak [SJCGQ11LM] and door/window sensor [MCCGQ11LM]).
For Z2M I'm still on 1.40 so the problem is with 20240710 adapter firmware.

closed 2024-12-20 by github-actions[bot]