Koenkk/zigbee2mqtt #23826: [New device support]:

mullahomes2012, 2024-09-03:
### Link

https://www.zigbee2mqtt.io/devices/HLD503-Z-CT.html

### Database entry

0x00212effff06d161

### Comments

Hi
Im totally new to this. I have this devices. Its made by Shenzhen Homa. The model is  HLD503-Z CT Its listed ass supported in Z2M but in my Z2M it says unsupported. Its llisted in my Z2M as HOMA 1022. Ive added it it to Z2M and using exposes, I can control Brightness, Power and Color Temp. I can control it using Z2M and it comes up in HA but in HA, I wasnt really able to control it.  In the reporting tab, when I set Cluster - LevelCtrl to currentLevel I managed to get HA to be able to show the devices brightness and power but it doesnt always work. I dont get any info about the color temp in HA.  

Sorry Im totally new to this. Please pardon my ignorance. Couldn't find the Database entry so I put in the IEEE address for the coordinator 

### External definition

```shell
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['HOMA1022'],
    model: 'HOMA1022',
    vendor: 'ShenZhen_Homa',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[150,500]}})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

ON Off State, Brightness, Color Temperature

Koenkk, 2024-09-03:
Added, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-03 by Koenkk

mullahomes2012, 2024-09-03:
Sorry for sounding a bit silly. What should I do? Do I use z2m edge as a new addon? Will I have to pair the device to the Z2M edge ? Thanks for your assistance.