Koenkk/zigbee2mqtt #23827: can't flash firmware Electrolama zig-a-zig-ah! (zzh!) with 20240710

ericb48, 2024-09-03:
### What happened?

[image]
hello,
I have a zzh type coordinator that was with an old firmware version. I managed to update it with the 20221226 version but I can't update it with the following versions 20230507 or 20241007.
When I flash the firmware I get the following message in the “Flash programmer 2” application: “unknown record type 3”. See screenshot

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20221226

### Adapter

Electrolama zig-a-zig-ah! (zzh!)

### Setup

docker on rpi 4

chrisrock1984, 2024-09-03:
use the python flasher - this worked for me

https://github.com/wirenboard/wb-cc2652p-flasher

RobLeighton67, 2024-09-03:
Convert to .bin (Intel format) using srecord 1.65 ( https://srecord.sourceforge.net/ ) - Flash Programmer 2 works fine with that format.

closed 2024-09-03 by ericb48