Koenkk/zigbee2mqtt #23835: Going nuts about problem with Mumubiz TYZGTH1CH-D1RF

Birdgage, 2024-09-03:
### What happened?

Well already weeks and lots and lots of hours spent trying to solve this problem, This is the case, I bought several Mumubiz TYZGTH1CH-D1RF devices to control the fans on radiators (heatpump system house). The pairing of the device goes perfect, and all functions and parameters are function. But....after a reboot of Zigbee2MQTT, everything comes online but the on/off function gives an error and will not work, all the other data points are working, such as temp/hum calibration etc etc. Repairing does not solve the problem, only deleting the device, restarting Zigbee2MQTT and pairing the device will work, until the next reboot, and then again only the on/off function is not working. What have I done to solve or understand this problem, first going nuts on google, no usable answers. Build a test docker with a separate build of Zigbee2MQTT, clean install, same problem, same behavior, tried several coordinators (XZG Coordinator Firmware | 20240710 Hardware | CC2652P7) and ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_ on both same problem. Installed a second Zigbee2MQTT in Home Assistant, clean install, same problem. Looked for hours, days, at the logs to try to understand what changes after a reboot. Tried to use ZHA, but ZHA don't recognize the device at all. My profession is problem solver, but I am giving up on this one... Koen? Anyone, who gives the golden tip?

### What did you expect to happen?

That it works as aspects even after reboot.

### How to reproduce it (minimal and precise)

Pair device, check function, stop and start Zigbee2MQTT and check if on/off function works.

### Zigbee2MQTT version

1.40.0-1 

### Adapter firmware version

20240710

### Adapter

UZG-01

### Setup

Ad on Home Assistant, VM on Proxmox/ Zigbee2MQTT CT has same problem

### Debug log

When it is working

[2024-09-03 18:04:15] debug: 	zh:controller: Received payload: clusterID=61184, address=34622, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=200, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":136,"commandIdentifier":6},"payload":{"seq":45581,"dpValues":[{"dp":103,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,90]}}]},"command":{"ID":6,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"activeStatusReport"}}
[2024-09-03 18:04:15] debug: 	z2m: Received Zigbee message from '0xa4c1382611d1db8f', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,90],"type":"Buffer"},"datatype":2,"dp":103}],"seq":45581}' from endpoint 1 with groupID 0
[2024-09-03 18:04:15] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt_2/0xa4c1382611d1db8f', payload '{"auto_settings":null,"child_lock":null,"humidity":90,"humidity_calibration":0,"humidity_sensitivity":1,"linkquality":200,"manual_mode":"ON","power_outage_memory":"off","state":"ON","temperature":19.6,"temperature_calibration":0,"temperature_sensitivity":0.2}'

After Reboot Start/Stop Zigbee2MQTT sending on/off command to device

error 2024-09-03 18:09:22z2m: Publish 'set' 'state' to '0xa4c1382611d1db8f' failed: 'Error: ZCL command 0xa4c1382611d1db8f/0 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 34622 - 0 - 7 - 6 - 11 after 10000ms)'

Additional info from log
 [2024-09-03 18:09:59] debug: 	zh:ezsp:ezsp: Time to watchdog ... 0
[2024-09-03 18:09:59] debug: 	zh:ezsp:ezsp: ==> nop: undefined
[2024-09-03 18:09:59] debug: 	zh:ezsp:ezsp: ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
[2024-09-03 18:09:59] debug: 	zh:ezsp:uart: --> DATA (7,1,0): 5700010500
[2024-09-03 18:09:59] debug: 	zh:ezsp:uart: --> [711521a9512af3707e]
[2024-09-03 18:09:59] debug: 	zh:ezsp:uart: -?- waiting (0)
[2024-09-03 18:09:59] debug: 	zh:ezsp:uart: <-- [1015a1a9512a34f07e]
[2024-09-03 18:09:59] debug: 	zh:ezsp:uart: <-- DATA (1,0,0): 1015a1a9512a34f07e
[2024-09-03 18:09:59] debug: 	zh:ezsp:uart: --> ACK  (2)
[2024-09-03 18:09:59] debug: 	zh:ezsp:uart: --> [82503a7e]
[2024-09-03 18:09:59] debug: 	zh:ezsp:uart: <-- ACK (0): 1015a1a9512a34f07e
[2024-09-03 18:09:59] debug: 	zh:ezsp:ezsp: <== Frame: 5780010500
[2024-09-03 18:09:59] debug: 	zh:ezsp:ezsp: <== 0x5: {"_cls_":"nop","_id_":5,"_isRequest_":false}
[2024-09-03 18:09:59] debug: 	zh:ezsp:uart: -+- waiting (0) success

cdupinco, 2024-10-10:
Hello. I have the same issue.  Only with zibee2mqtt . 
All input data are ok , even state if I push the button on device. 
I got simlar issue with TYZGTH1CH-D1RF 
and  MS24051501 
both are from the same brand : mumubiz

https://fr.aliexpress.com/item/1005007011406699.html?spm=a2g0o.order_list.order_list_main.59.64975e5blU6l7M&gatewayAdapt=glo2fra

Capricious device: After several Delete , pairing.  reboot,   set/state is ok
I have to do it from python application now

Tee86, 2024-11-17:
Guys, can someone please test with me this external converter:

```
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const {deviceEndpoints, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS000F'],
    model: 'TS000F',
    vendor: '_TZ3218_7fiyo3kv',
    description: '1G Relay with Temp Probe',
    fingerprint: [{modelID: 'TS000F', manufacturerName: '_TZ3218_7fiyo3kv'}],
    fromZigbee: [
        {
            cluster: 'manuSpecificTuya',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                const dpValue = tuya.getDataValue(msg.data.dp);
                if (msg.data.dp === 102) {
                    return {temperature: dpValue};
                }
            }
        },
    ],
    toZigbee: [],
    extend: [
        deviceEndpoints({"endpoints":{"0":0,"1":1}}), 
        onOff({"powerOnBehavior":false,"endpointNames":["1"]}),
        tuya.modernExtend.dpTemperature({dp: 0x66, scale: 10}),
    ],
    exposes: [
        e.temperature(),
        e.offset(),
    ],
    meta: {
        tuyaDatapoints: [
            [102, 'temperature', tuya.valueConverter.raw],
            [108, 'offset', tuya.valueConverter.raw],
        ],
    },

    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.bind('genOnOff', coordinatorEndpoint);
    },
};

module.exports = definition;
```
I spent countless hours trying to make this stupid relay work and somehow I've managed with the above.

All of sudden all the exposes are working as they should (not only temp reading and switching) and it survived about 10 z2m restarts till now without any issues, although it throws me an error in the log. Despite that, it seems like that error is forcing it to work correctly. Really weird, but works 100% for me.
Let me know if that works for you too...

pdoidge, 2025-01-01:
I have the same device (_TZ3218_7fiyo3kv) and this config wasn't loading because of an error with the offset variable. I just removed it, it then the temp reading and switch work as intended now, thanks!

Birdgage, 2025-01-05:
> I have the same device (_TZ3218_7fiyo3kv) and this config wasn't loading because of an error with the offset variable. I just removed it, it then the temp reading and switch work as intended now, thanks!

Hi, can you share the js. file with us?
Will be great,
Thx.

pdoidge, 2025-01-19:
I just commented out the two offset lines and it worked, didn't even bother with deleting them. Thanks for figuring this out, it's been useful.

``` js
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const {deviceEndpoints, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS000F'],
    model: 'TS000F',
    vendor: '_TZ3218_7fiyo3kv',
    description: '1G Relay with Temp Probe',
    fingerprint: [{modelID: 'TS000F', manufacturerName: '_TZ3218_7fiyo3kv'}],
    fromZigbee: [
        {
            cluster: 'manuSpecificTuya',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                const dpValue = tuya.getDataValue(msg.data.dp);
                if (msg.data.dp === 102) {
                    return {temperature: dpValue};
                }
            }
        },
    ],
    toZigbee: [],
    extend: [
        deviceEndpoints({"endpoints":{"0":0,"1":1}}), 
        onOff({"powerOnBehavior":false,"endpointNames":["1"]}),
        tuya.modernExtend.dpTemperature({dp: 0x66, scale: 10}),
    ],
    exposes: [
        e.temperature(),
        // e.offset(),
    ],
    meta: {
        tuyaDatapoints: [
            [102, 'temperature', tuya.valueConverter.raw],
            // [108, 'offset', tuya.valueConverter.raw],
        ],
    },

    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.bind('genOnOff', coordinatorEndpoint);
    },
};

module.exports = definition;
```

closed 2025-03-28 by github-actions[bot]