Koenkk/zigbee2mqtt #23839: zhc: Failed to apply calibration to 'voltage' and 'power'

prohand, 2024-09-03:
### What happened?

Hello,

I have this device :

Modèle Zigbee : TS0601
Constructeur Zigbee : _TZE204_yjjdcqsq
Constructeur : [Tuya](https://www.zigbee2mqtt.io/supported-devices/#v=Tuya)
Modèle : [ZTH01](https://www.zigbee2mqtt.io/devices/ZTH01.html#tuya-zth01)

This error appear in the log of z2m : 

Error 2024-09-03 20:50:28zhc: Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-09-03 20:55:43zhc: Failed to apply calibration to 'voltage': 'voltage_precision' is not a number, got string ()

How is it possible to solve this problem?

Thanks

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20240315

### Adapter

zStack3x0

### Setup

Plain

Jobs33, 2024-09-04:
I have the same message for this device. How we can remove this message?

Koenkk, 2024-09-04:
Update your z2m `configuration.yaml`, the calibration should be a string not a number, e.g. change `'5'` -> `5`

closed 2024-09-04 by Koenkk

prohand, 2024-09-04:
The problem was present on another sensor, I modified the lines and everything is ok

Thanks

Jobs33, 2024-09-05:
Hi Koenkk,

Regarding this topic that you are describing;
"Update your z2m configuration.yaml, the calibration should be a string not a number, e.g. change '5' -> 5"
Do you have a manual or example to introduce this code inside configuration.yaml file?

Thanks in advance!

Jobs33, 2024-09-05:
Hello,
Inside configuration.yaml (/homeassistant/zigbee2mqtt/configuration.yaml), I have the following code:

"  '0xbc026efffea7f1cb':
    friendly_name: sensor
    voltage_calibration: 0
    voltage_precision: ''
"
I've modified this code and the message has not disapeared.

extremecarver, 2025-07-17:
I think this bug appears as soon as you enter anything into Settings (specifics). If all fields are empty it is fine, as soon as you fill in one, you need to fill out all.
I had entered 20 for the polling interval - and then got this error. Setting 0 to all other fields solved it.