Koenkk/zigbee2mqtt #23840: Ikea E2123 keeps losing connection

fthiery, 2024-09-03:
### What happened?

Device keeps losing connection after a few hours, often right after i use it (e.g. i long-press -- action works -- but then device becomes unresponsive). To fix it, i have to open the lid, remove the batteries and re-insert them.

I tried pairing on multiple routers (Ikea plug, LIDL plug), even placed the device a few centimeters away from the LIDL plug, and it still loses connection.

NB: Legacy integration is disabled

### What did you expect to happen?

I wish it did not drop...

### How to reproduce it (minimal and precise)

That's the tricky part :( I'm trying to catch something in the logs when it happens; any help to troubleshoot appreciated.

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20240710

### Adapter

zStack3x0 (Sonoff-P)

### Setup

Home Assistant Addon (HAOS on Home Assistant Green)

chrisrock1984, 2024-09-04:
I also have problems with Ikea devices now. for me the sockets keep disappearing

fthiery, 2024-09-05:
I finally catched logs... Which indicate nothing ! When it stops working, i just see nothing anymore in the Z2M logs. The logs are really big, so if anyone has an idea of additional items to check in the logs, please ask

fthiery, 2024-09-05:
I just discovered that if i get the battery level, the device will resume emitting press events, @Koenkk does that ring any bell or is the device just buggy ?

```
PRESS buttons <-- nothing happens
GET BATTERY
[2024-09-05 11:20:45] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Télécommande Soundsystem/get' with data '{"battery":""}'
WORKS BACK
[2024-09-05 11:20:57] debug: 	z2m: Received Zigbee message from 'Télécommande Soundsystem', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":1,"rate":255}' from endpoint 1 with groupID 0
[2024-09-05 11:20:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Télécommande Soundsystem', payload '{"action":"volume_down","battery":84,"linkquality":65,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2600}'
[2024-09-05 11:20:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Télécommande Soundsystem', payload '{"action":"","battery":84,"linkquality":65,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2600}'
[2024-09-05 11:20:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Télécommande Soundsystem/action', payload 'volume_down'
```

Koenkk, 2024-09-05:
@fthiery that doesn't ring any bell, if you do not see any `Received Zigbee message` then the device doesn't send any message. [Sniffing](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) would be needed to debug it further.

Does the issue also occur when re-pairing the device through the coordinator and keeping it close to it? If not, then the issue may be related to one of the routers in your network.

fthiery, 2024-09-05:
> @fthiery that doesn't ring any bell, if you do not see any `Received Zigbee message` then the device doesn't send any message.

I confirm: when the device has reached the problematic state, there is no `Received Zigbee message` at all.

> [Sniffing](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) would be needed to debug it further.

I'll look into this

> Does the issue also occur when re-pairing the device through the coordinator and keeping it close to it? If not, then the issue may be related to one of the routers in your network.

I'll try this next, thanks for the suggestion.

fthiery, 2024-09-10:
For the record, this weekend i did a full mesh "reset" (i.e. stop z2m, kill house mains power for 1 minute, wait 20 min, then restart the coordinator), and so far the problem did not reproduce, although i am ready to sniff the traffic. I guess this seems to int at a router/routing problem...

fthiery, 2024-09-24:
@Koenkk i managed to catch a trace when reproducing the problem, here's a trace captured with filter `_ws.col.def_src == "0xb950" or _ws.col.def_dst == "0xb950"` ; in this trace i was pressing buttons and while the device seems to send and receive data, it seems it does not send action packets; if i request the battery attribute, it comes back and i see the presses.

[trace_state.pcapng.gz](https://github.com/user-attachments/files/17114599/trace_state.pcapng.gz)

What do you think ?

Koenkk, 2024-09-25:
- Whats the network key of this sniff?
- Could you try to wakeup the device and immediately after that do a configure (z2m frontend -> device -> yellow refresh icon). Does that succeed and fix the issue?

fthiery, 2024-09-25:
> * Whats the network key of this sniff?

Sorry, i thought that Wireshark would export decrypted packets. I've connected you on linkedin to DM it to you.

> * Could you try to wakeup the device and immediately after that do a configure (z2m frontend -> device -> yellow refresh icon). Does that succeed and fix the issue?

Will do next time it happens (i.e. run configure instead of querying battery state)

fthiery, 2024-09-25:
Alternatively, 

Here's a normal trace: it shows Zigbee HA requests
[image]

Here's the trace when it's not working; it shows Zigbee ZDP requests 
[image]

fthiery, 2024-10-02:
> * Could you try to wakeup the device and immediately after that do a configure (z2m frontend -> device -> yellow refresh icon). Does that succeed and fix the issue?

Hi, just tried (the device got stuck again); the configure errors out with the following log:

```
error 2024-10-02 13:17:08z2m: Failed to configure 'Télécommande Soundsystem', attempt 3 (Error: Bind 0x540f57fffe1e42b7/1 genPollCtrl from '0x00124b002a2ec36f/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23) at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1118:40) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:857:28 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:799:9) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:531:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:176:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:193:21) at Configure.configure (/app/lib/extension/configure.ts:125:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:56:21))
```

fthiery, 2024-10-02:
@Koenkk i did sniff the traffic again; when the device is stuck, i see a lot of ZDP requests (Network Address Request) from the device; it seems it left the network and is requesting a new address, without success.

When i request the battery level, it seems to get an address ???

[image]

fthiery, 2024-10-04:
> That's not what this ZDP request does. A network address request allows a device to request the network address of another device. It seems to request the network address of the coordinator, which is answered correctly on 1442

What could block the device from receiving a response to it's request ?

> One thing that confuses me, I see the device starts sending requests again (e.g. 1633) but the battery read is later (1671)?

Yes, that's the strange part; i literally do nothing else than requesting the battery level; it seems that asking the coordinator to request the battery level "unblocks" the pending ZDP response.

I'll use a chronometer next time, started when i start the capture, to reference the exact moment when i press the button, so that i can identify the moment in the traces.

>> Could you try to wakeup the device and immediately after that do a configure (z2m frontend -> device -> yellow refresh icon). Does that succeed and fix the issue?
> Hi, just tried (the device got stuck again); the configure errors out with the following log:

What did you have in mind ? What can you do from the fact that it does not work ?

Koenkk, 2024-10-05:
I have no clue tbh, and did not see any other reports about this, could it be that your E2123 is broken? (did you already try with another one)?

fthiery, 2024-12-13:
Still happening, i'm gradually removing devices from my network to see if there is a culprit.

fthiery, 2025-01-12:
For the record, the problem disappeared after removing some devices, notably one Silvercrest smart plug that had bad signal, and heiman sirens. Not sure what fixed it, but it did...

closed 2025-01-12 by fthiery

1aranzant, 2025-04-30:
I have the exact same issue... my symfo gen2 remote stops responding randomly, I deleted a lot of devices but the issue keeps on happening... @fthiery did you fix it for good?

fthiery, 2025-05-02:
> I have the exact same issue... my symfo gen2 remote stops responding randomly, I deleted a lot of devices but the issue keeps on happening... [@fthiery](https://github.com/fthiery) did you fix it for good?

It does not happen anymore, but I'm not sure why not, I did these two things:
- I removed a router (smart plug) that was close to it but that had bad coverage (behind a dishwasher)
- I added a sonoff E flashed as repeater in the same room

I will re-add the smart plug to see if the repeater did the trick