Koenkk/zigbee2mqtt #23843: Zigbee Dongle Port Shutdown and Home Assistant Crashes

Option212, 2024-09-04:
### What happened?

Occasionally, the port on my Zigbee dongle shuts down (one to 5times a day).
When this happens, restarting the service consistently causes Home Assistant to crash, resulting in lost logs.
The workaround I've found is to disconnect (physically) and reconnect the adapter, followed by restarting the Zigbee2MQTT service.

Only one device is paired : ICT module for LINKY to Zigbee 3.0 counter ZLinky_TICLixee.

### Zigbee2MQTT version

1.40.0-1

### Adapter firmware version

7.4.1.0-zbdonglee-115200

### Adapter

ZBdongle-E

### Setup

Raspberry Pi 3 Model B

### Debug log

[log_1.txt](https://github.com/user-attachments/files/16862104/log_1.txt)
[log_error_zigbee2mqtt.txt](https://github.com/user-attachments/files/16862105/log_error_zigbee2mqtt.txt)

nlhomme, 2024-09-07:
I don't have logs to show but I also experience a very similar issue, with the same zigbee dongle, the same zigbee2mqtt release

Option212, 2024-09-11:
Since the release of version 1.40.1 four days ago, I haven't experienced any new incidents or crashes.

Option212, 2024-09-16:
The watchdog functions correctly, and the service restarts in the event of an error. However, I haven’t been able to locate the commit that fixed this issue, either in this repository ZigBee2Mqtt or in the Home Assistant repository.

Option212, 2024-09-20:
"Z2M cant connect to the dongle."
"Z2M problem may originate in some unintentional bindings to this newer HA version"

Related issues : 
https://github.com/Koenkk/zigbee2mqtt/issues/23901 
https://community.home-assistant.io/t/newbie-zigbee2mqtt-errors-after-ha-updtae-zonoff-3-0-plus-e-crc-and-ash-error/768831/19

Option212, 2024-09-22:
Could be a regression in linux kernel 6.6 used by HA. Issues tracked by https://github.com/raspberrypi/linux/issues/6172.

I'm sharing the same conclusion to solve the issue : 
stop z2m add-on and don't let it start at HA start
unplug the dongle
restart host
wait some minutes after restart
plug the dongle
start z2m

Option212, 2024-10-24:
After switching from my Raspberry Pi 3 to a NUC, I no longer have this issue. The problem comes from the operating system of Home Assistant on the Raspberry Pi 3.

nlhomme, 2024-10-25:
I no longer have the issue as well, but I think it was caused by the Home Assistant pluging for the Stream Deck to cause my crash issue. I removed it a few weeks ago and everything run smooth

closed 2025-01-01 by github-actions[bot]