Koenkk/zigbee2mqtt #23844: Can't update Ikea Styrbar 1.0.024 to latest version

drumstick77, 2024-09-04:
### What happened?

I've tried following the instruction on this link --> https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-1692857032, (following update from jkl1337 but the link to download the OTA 2.4.5 doesn't seems to be working. --> http://fw.ota.homesmart.ikea.net/global/GW1.0/01.20.065/bin/10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed
Can anyone please help? Thanks 

### Zigbee2MQTT version

1.40.0-1

### Adapter firmware version

20211217

### Adapter

sonoff

### Setup

Home Assistant

LeeWiu, 2024-09-05:
**Right click** on the link then **Save Link As**

drumstick77, 2024-09-06:
Thank you. I've progressed further now. So I'm following this instruction
[image]

but with the latest update on Z2M, there's already a line for external_converters:[] at the very top, so when I added external_converters on the very bottom of confirmation.yaml, it naturally flagged duplicate.
so I removed the external_converters line on the bottom but add on fails to start.

[image]

drumstick77, 2024-10-16:
Anyone please? I'm still stuck and everytime Z2M gets updated, HA moans. please help. thanks

thomasaarholt, 2024-12-17:
I was able to update from 1.0.024 to 2.4.5 by tapping buttons on the ikea remote while pressing the update option on Z2M / OTA. I am using the standard Z2M config, no external converter.

thomasaarholt, 2024-12-17:
I was then able to update from `2.4.5` to `2.4.16` by doing the same thing again.

chaosmonkey2, 2025-01-07:
> I was then able to update from `2.4.5` to `2.4.16` by doing the same thing again.

Worked just fine. Thanks for the great advice!

chaosmonkey2, 2025-01-07:
> I was able to update from `1.0.024` to `2.4.5` by tapping buttons on the ikea remote while pressing the update option on Z2M / OTA. The intention behind the tapping is to keep the remote "alive" since it is a battery powered device. I am using the standard Z2M config, no external converter.

Worked just fine. Thanks for the great advice!

thomasaarholt, 2025-01-07:
Great! If you have trouble with the actions not working in Z2M, then checkout this comment: https://github.com/Koenkk/zigbee2mqtt/issues/25505#issuecomment-2571664873

You can either turn on legacy support or update to the new way of doing it. I haven’t tested either yet, I’m currently using ZHA.

closed 2025-03-16 by github-actions[bot]