Koenkk/zigbee2mqtt #23845: Irrigation device Woox R7060 off-time not working

miguelpineiroo, 2024-09-04:
### What happened?

I have two irrigation control devices R7060 and I can normally switch them off and on. In the ZigbeeMQTT page of the device, it is said that the device support time.
[On with timed off](https://www.zigbee2mqtt.io/devices/R7060.html#on-with-timed-off)
When setting the state to ON, it might be possible to specify an automatic shutoff after a certain amount of time. To do this add an additional property on_time to the payload which is the time in seconds the state should remain on. Additionally an off_wait_time property can be added to the payload to specify the cooldown time in seconds when the switch will not answer to other on with timed off commands. Support depends on the switch firmware. Some devices might require both on_time and off_wait_time to work Examples : {"state" : "ON", "on_time": 300}, {"state" : "ON", "on_time": 300, "off_wait_time": 120}.

It is not working the "on_time" or "on_time" + "off_wait_time". When I send a MQTT command with them, for example {"state" : "ON", "on_time": 300} or {"state" : "ON", "on_time": 300, "off_wait_time": 120}, it is ignored, no error, but it does not trigger the ON. If I send a command {"state" : "ON"} it works. Under these circumstances, I cannot set a timed off when setting the ON. 

I have a Conbee2 coordinator, the latest versions so far: Z2M 1.40.0, HA OS on Raspberry Pi 2024.8.3 (Core), 2024.08.0 (Supervisor), Operating System 13.1, frontend: 20240809.0. It has happened with all the previous versions since the beginning. It happens the same in my two Woox devices. I bought the original Woox Zigbee hub to update the device firmware, and I temporary paired them with it, but they were already with the latest firmware, so I unpaired and now they are paired again with Z2M, not working the off_wait_time feature. I have an automation at Home Assistant (HA) level, to switch them off after some minutes, but I would prefer to have the native off with the on, to prevent that some HA failure let the irrigation on.

Thanks

### What did you expect to happen?

The on-with-time-off feature to work

### How to reproduce it (minimal and precise)

to publish the MQTT messages described above to switch on with time-off, to see that it does not work

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

26720700

### Adapter

Conbee2

### Setup

Raspberry PI 4B x-64 with HA OS

closed 2024-12-20 by github-actions[bot]

miguelpineiroo, 2024-12-20:
No working yet

pklopotowski, 2025-06-22:
I have exactly the same observation and finally the same problem. Is there any chance to fix it?