Koenkk/zigbee2mqtt #23849: TONGOU TO-Q-SY2-163JZT reporting values out of range

fjluquerodriguez, 2024-09-04:
### What happened?

As documentation shows, for example  for Over-current threshold the value should be between 1 and 64 but device is reporting 65:
```

{
    "current": 0.48,
    "energy": 154.52,
    "indicator_mode": "off/on",
    "linkquality": 76,
    "over_current_breaker": "ON",
    "over_current_threshold": 65,
    "over_voltage_breaker": "ON",
    "over_voltage_threshold": 265,
    "power": 15,
    "power_breaker": "ON",
    "power_outage_memory": "restore",
    "power_threshold": 13,
    "state": "ON",
    "temperature": 37.5,
    "temperature_breaker": "ON",
    "temperature_threshold": 100,
    "under_voltage_breaker": "ON",
    "under_voltage_threshold": 75,
    "voltage": 233,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}
```
This is causing errors in home assistant:

`2024-09-04 20:52:49.647 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.depuradora_over_current_threshold: 65 (range 1.0 - 64.0)`

### What did you expect to happen?

Value shoul be in the range

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20230507

### Adapter

Sonoff zbdongle

### Setup

Docker

phrozen77, 2024-12-08:
I see similar issues on my HA (2024.12.1), Z2M 1.42.0:

```
Invalid value for number.heizkorper_erdgeschoss_under_voltage_threshold: 75 (range 76.0 - 240.0)
Invalid value for number.heizkorper_erdgeschoss_over_current_threshold: 65 (range 1.0 - 64.0)
Invalid value for number.heizkorper_obergeschoss_over_current_threshold: 65 (range 1.0 - 64.0)
Invalid value for number.heizkorper_obergeschoss_under_voltage_threshold: 75 (range 76.0 - 240.0)
```
Since these occur pretty regularly, they're flooding the HA logs - within 1 hour, the log messages are repeated over 1400x.

Excluding those attributes from reporting via Settings -> Filtered publish attributes results in other messages in HA:

```
Logger: homeassistant.helpers.template
Quelle: helpers/template.py:2746
Erstmals aufgetreten: 07:38:38 (60 Vorkommnisse)
Zuletzt protokolliert: 07:41:14

Template variable warning: 'dict object' has no attribute 'over_current_threshold' when rendering '{{ value_json.over_current_threshold }}'
Template variable warning: 'dict object' has no attribute 'under_voltage_threshold' when rendering '{{ value_json.under_voltage_threshold }}'
```
 What gives? :)

phrozen77, 2024-12-08:
To add, this seems to be the same issue as reported in #18051 

This comment asserts, that when changed in Zigbee2MQTT, the values seemingly are "in range" then.

After changing the values for both of my breakers in Z2M, the messages have stopped - however this probably would need to be done after each restart of Zigbee2MQTT.

Since the original issue is already 1.5 years old at this point in time - maybe we can get some attention here @Koenkk ? 

(sorry for tagging you directly, however i wouldn't know who else - maybe it would be wise to adopt a similar system as HA is doing - where some maintainer(s) is/are responsible for a certain integration, here it'd be for one or more devices supported?)

phrozen77, 2025-02-07:
No stale :)

closed 2025-04-17 by github-actions[bot]