Koenkk/zigbee2mqtt #23850: [New device support]: Hue Dymera LCW005 - extend LCW004?

ClimbsRocks, 2024-09-04:
### Link

https://www.philips-hue.com/en-us/p/hue-white-and-color-ambiance-dymera-indoor-and-outdoor-wall-light/046677585235

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x001788010de246df","nwkAddr":11847,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LCW005","epList":[11,12,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LCW005","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20240628","swBuildId":"1.116.5"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":1,"currentX":26493,"currentY":11085,"enhancedCurrentHue":55701,"currentSaturation":254,"colorTemperature":281}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":260,"epId":12,"devId":269,"inClusterList":[0,3,4,5,6,8,64515,768,64513],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20240628","swBuildId":"1.116.5","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1725475764781}

### Comments

I'm hoping this is as simple as extending the LCW004 integration which was discussed here: https://github.com/Koenkk/zigbee2mqtt/issues/21824 to work with LCW005. Side note- that issue can probably be closed now. Here is the link to the LCW004 product on the Supported Devices List: https://www.zigbee2mqtt.io/devices/046677585235.html

When I added the LCW005 device, it shows up as Unsupported, and Disabled, though the Exposes tab seems to show all the information I would expect. 

If it's not as simple as just adding LCW005 to a list next to LCW004 somewhere, let me know and I'll scope out if I can work on the integration, particularly if you give me some pointers on how to start with the existing LCW004 integration. 

[image]

[image]

### External definition

```shell
const {deviceEndpoints} = require('zigbee-herdsman-converters/lib/modernExtend');
const {philipsLight} = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['LCW005'],
    model: 'LCW005',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"11":11,"12":12}}), philipsLight({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}}), philipsLight({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Nothing works as far as I can tell. When I go to Z2M WebUI -> Devices -> Exposes, I see everything there that I would naively expect to see (two lights, brightness, lots of color settings, etc). But, the lights do not seem to respond to changes I make on that page. I can't see the top light, only the bottom light at the moment, so maybe it's related to https://github.com/Koenkk/zigbee2mqtt/issues/22828#issuecomment-2140845468 . But I'm also very new to HA, so I might just be on the wrong page entirely.

Koenkk, 2024-09-05:
Does it work when using:

```js
const definition = {
        zigbeeModel: ['LCW005'],
        model: '046677585235',
        vendor: 'Philips',
        description: 'Hue Dymera indoor and outdoor wall light',
        extend: [
            deviceEndpoints({endpoints: {top: 11, bottom: 12}}),
            philipsLight({colorTemp: {range: [153, 500]}, color: true, endpointNames: ['top', 'bottom']}),
        ],
    }
```

closed 2024-12-20 by github-actions[bot]

lilwangang, 2025-07-29:
Hey man! I’m struggling here, I added the light to the hue app, updated it, etc. 

But I cannot figure out how to get to seperate light entity’s from this light! I can only control the top light and not the bottom. I started using mqtt less than a week ago. I’ve been using zha for 6 months no problem, but this light is giving me problems in mqtt