Koenkk/zigbee2mqtt #23854: [New device support]: Innr RF262 - E27 Spiral Filament Bulb, 550 lm 2200K Warm White

MaKin211, 2024-09-05:
### Link

https://www.amazon.de/innr-Glühbirne-Funktioniert-Assistant-erforderlich/dp/B0CZNPRQDG?th=1

### Database entry

{"id":21,"type":"Router","ieeeAddr":"0xa4c1383515bafca4","nwkAddr":19832,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"RF 262","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"RF 262","manufacturerName":"innr","powerSource":1,"zclVersion":8,"appVersion":23,"stackVersion":54,"hwVersion":1,"dateCode":"20240613","swBuildId":"1.7.16"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff083853","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":23,"stackVersion":54,"hwVersion":1,"dateCode":"20240613","swBuildId":"1.7.16","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1725473430505,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to extend the external definition but am stuck with the limited function of only turning on and off the bulb.

### External definition

```shell
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['RF 262'],
    model: 'RF 262',
    vendor: 'innr',
    description: 'Automatically generated definition',
    extend: [onOff({powerOnBehavior: false})],
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Bulb can only be turned on and off. No setting brightness or setting transition feature.

Koenkk, 2024-09-05:
Added!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `RF 262.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-09-05 by Koenkk

MaKin211, 2024-09-05:
Done. Thanks a lot!