Koenkk/zigbee2mqtt #23855: Z-Dongle E ember periodically error

EugenKomi, 2024-09-05:
### What happened?

Zigbee2MQTT periodically report error  zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT then restart.  

### What did you expect to happen?

The adapter gives an error several times a day, leading to loss of connection and stopping and restarting Zigbee2Mqtt. 

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

7.4.3 [GA]

### Adapter

SonOFF Dongle-E

### Setup

HA on VM Synology DSM 

### Debug log

[MQTT7_39.log](https://github.com/user-attachments/files/16885958/MQTT7_39.log)

EugenKomi, 2024-09-05:
I removed the USB extension cable, switched to ezsp in the settings. Problem still occurs. 

[2024-09-05 12:02:18] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":4} after 4000ms
[2024-09-05 12:02:18] error: 	zh:ezsp:uart: -!- break waiting (4)
[2024-09-05 12:02:18] error: 	zh:ezsp:uart: Can't send DATA frame (3,0,0): 53000134000065ae040106000101000100009a9b0518880b0a00
[2024-09-05 12:02:19] debug: 	zh:ezsp:uart: ->> DATA (3,0,1): 53000134000065ae040106000101000100009a9b0518880b0a00
[2024-09-05 12:02:19] debug: 	zh:ezsp:uart: --> [387d3121a9602a15d7f7904b23aa5493499d4e273176cb7f03f6cc63c3787e]
[2024-09-05 12:02:19] debug: 	zh:ezsp:uart: -?- rewaiting (4)
[2024-09-05 12:02:20] debug: 	zh:ezsp:uart: <-- [c20251a8bd7e]
[2024-09-05 12:02:20] debug: 	zh:ezsp:uart: <-- Error c20251a8bd7e
[2024-09-05 12:02:20] debug: 	zh:ezsp:uart: Uart reseting
[2024-09-05 12:02:20] debug: 	zh:ezsp:uart: --> Write reset
[2024-09-05 12:02:20] debug: 	zh:ezsp:uart: --> [1ac038bc7e]
[2024-09-05 12:02:20] debug: 	zh:ezsp:uart: -?- waiting reset
[2024-09-05 12:02:23] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":4} after 4000ms
[2024-09-05 12:02:23] error: 	zh:ezsp:uart: -!- break rewaiting (4)
[2024-09-05 12:02:23] error: 	zh:ezsp:uart: Can't resend DATA frame (3,0,1): 53000134000065ae040106000101000100009a9b0518880b0a00
[2024-09-05 12:02:23] debug: 	zh:ezsp:ezsp: onSerialReset()
[2024-09-05 12:02:23] debug: 	zh:ezsp:driv: onEzspReset()
[2024-09-05 12:02:23] debug: 	zh:ezsp:driv: Reset connection.
[2024-09-05 12:02:23] debug: 	zh:ezsp:driv: Stopping driver
[2024-09-05 12:02:23] debug: 	zh:ezsp:ezsp: Closing Ezsp
[2024-09-05 12:02:23] debug: 	zh:ezsp:uart: Closing UART
[2024-09-05 12:02:23] debug: 	zh:ezsp:driv: Request error Error: Failure send sendUnicast:{"type":"Buffer","data":[83,0,1,52,0,0,101,174,4,1,6,0,1,1,0,1,0,0,154,155,5,24,136,11,10,0]}
[2024-09-05 12:02:23] debug: 	zh:controller:endpoint: Error: ZCL command 0x282c02bfffed898b/1 genOnOff.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":136,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)
    at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:515:19)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:911:28)
    at Endpoint.defaultResponse (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:610:9)
    at Device.onZclData (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:498:17)
    at Controller.onZclPayload (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:841:13)
[2024-09-05 12:02:23] debug: 	zh:controller:device: Default response to 0x282c02bfffed898b failed (Error: ZCL command 0x282c02bfffed898b/1 genOnOff.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":136,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
[2024-09-05 12:02:30] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
[2024-09-05 12:02:30] debug: 	zh:ezsp:ezsp: onSerialReset()
[2024-09-05 12:02:30] debug: 	zh:ezsp:driv: onEzspReset()
[2024-09-05 12:02:30] debug: 	zh:ezsp:driv: Reset connection.
[2024-09-05 12:02:30] debug: 	zh:ezsp:driv: Stopping driver
[2024-09-05 12:02:30] debug: 	zh:ezsp:ezsp: Closing Ezsp
[2024-09-05 12:02:30] debug: 	zh:ezsp:uart: Closing UART
[2024-09-05 12:02:30] error: 	zh:ezsp:uart: Failed to reset on Error Frame: Error: Reset error: Error: {"sequence":-1} after 10000ms
[2024-09-05 12:02:31] debug: 	zh:ezsp:driv: Startup again.
[2024-09-05 12:02:31] debug: 	zh:ezsp:uart: RTS/CTS config is off, enabling software flow control.
[2024-09-05 12:02:31] debug: 	zh:ezsp:uart: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_186ff44314e2ed11b891eb5162c613ac-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2024-09-05 12:02:53] debug: 	zh:ezsp:uart: Port closed. Error? null
[2024-09-05 12:02:53] debug: 	zh:ezsp:ezsp: onSerialClose()
[2024-09-05 12:02:53] debug: 	zh:ezsp:uart: Serialport opened
[2024-09-05 12:02:53] debug: 	zh:ezsp:uart: Uart reseting
[2024-09-05 12:02:53] debug: 	zh:ezsp:uart: --> Write reset
[2024-09-05 12:02:53] debug: 	zh:ezsp:uart: --> [1ac038bc7e]
[2024-09-05 12:02:53] debug: 	zh:ezsp:uart: -?- waiting reset
[2024-09-05 12:02:54] debug: 	zh:ezsp:driv: Startup again.
[2024-09-05 12:02:54] debug: 	zh:ezsp:uart: RTS/CTS config is off, enabling software flow control.
[2024-09-05 12:02:54] debug: 	zh:ezsp:uart: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_186ff44314e2ed11b891eb5162c613ac-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2024-09-05 12:02:54] error: 	zh:ezsp:ezsp: Connection attempt 1 error: Error: Error Resource temporarily unavailable Cannot lock port
[2024-09-05 12:02:54] debug: 	zh:ezsp:uart: <-- [c1020b0a527e]
[... cut ...]
[2024-09-05 12:02:56] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
[2024-09-05 12:02:56] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
[2024-09-05 12:02:56] info: 	z2m: Disconnecting from MQTT server
[2024-09-05 12:02:56] info: 	z2m: Stopping zigbee-herdsman...
[2024-09-05 12:02:56] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2024-09-05 12:02:57] info: 	z2m: Stopped zigbee-herdsman
[2024-09-05 12:02:57] info: 	z2m: Stopped Zigbee2MQTT

Koenkk, 2024-09-05:
https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-resource-temporarily-unavailable-cannot-lock-port might help

EugenKomi, 2024-09-05:
No response

[image]

Koenkk, 2024-09-05:
Do you have any other services running which access USB devices? I think the error happens because an other process tries to access the port

EugenKomi, 2024-09-05:
> Do you have any other services running which access USB devices? I think the error happens because an other process tries to access the port

I have checked for this first. HA installed on Virtual Mashine, full oficial image. USB port mapped to VM (  and in case of disconnect i have to remap manually in VM settings). 

I have no other stick for check if this stick is faulty.

Koenkk, 2024-09-05:
Can you check on other hardware? e.g. your laptop

EugenKomi, 2024-09-05:
> Can you check on other hardware? e.g. your laptop

Unfortunately, no. I tried flashing with earlier firmware versions. With 7.4.1 Zegbee2MQTT  doesn't start at all.

EugenKomi, 2024-09-06:
> Can you check on other hardware? e.g. your laptop

So, after observations, I see that freezes occur when smart plugs are turned on. I have several temperature and humidity sensors in the system and three smart sockets that are turned on and off by events.
At the same time, I now remember that freezes occurred several times when turning on from the dashboard.

EugenKomi, 2024-09-06:
> Can you check on other hardware? e.g. your laptop

I caught it on video.

https://drive.google.com/file/d/1bIXjSlGxa923AWgXEgEgWkyd4Jod5G6-/view?usp=sharing

Koenkk, 2024-09-06:
Can you try the firmware from [this post](https://github.com/Koenkk/zigbee2mqtt/discussions/23224#discussioncomment-10422347) in combination with the `ember` driver?

EugenKomi, 2024-09-06:
> Can you try the firmware from [this post](https://github.com/Koenkk/zigbee2mqtt/discussions/23224#discussioncomment-10422347) in combination with the `ember` driver?

[2024-09-06 11:29:13] error: 	z2m: Adapter disconnected, stopping
[11:29:21] INFO: Preparing to start...
[11:29:21] INFO: Socat not enabled
[11:29:22] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-09-06 12:03:38] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-09-06 12:03:38] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-09-06 12:03:38] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-09-06 12:03:38] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-09-06 12:03:46] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=106 Len=7].
[2024-09-06 12:04:08] error: 	z2m: Adapter disconnected, stopping

Koenkk, 2024-09-06:
Could you provide the debug log from starting z2m until the crash (resulting from turning on the plugs)?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Koenkk, 2024-09-06:
@Nerivec could you check this one? It's a bit interesting that this causes a crash, there are not too many messages.

Nerivec, 2024-09-06:
Strange indeed, crash after only a couple requests in the queue. It should handle 80+ queue without issue with default concurrency (tested on Dongle-E with a generic Tuya RGB strip).

@EugenKomi From your last logs, it seems to happen only when toggling repeatedly in a short amount of time, correct?
Can you check if the same happens if you set `adapter_concurrent: 1` see [docs](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#advanced-configuration)? Note: you will likely notice a slight delay when triggering.
I see all these smart plugs you have joined at the moment are zb+ble, any chance you have another device model you could join (even temporarily) to see if the same happens?
Since it happens with `ezsp` also, it does sound like some kind of weird interference.

_Strange thing I noticed in the past few weeks, Dongle-E that have the ID format that does not have `V2` in it, seem to be having more troubles... Any chance you can slide off the cover from your adapter (couple of tiny screws on the side) and take a picture of both sides when you have some time?_

EugenKomi, 2024-09-06:
> From your last logs, it seems to happen only when toggling repeatedly in a short amount of time, correct?

No, this effect seems to be cumulative. For example, when I go to bed, my fumigator turns on once an hour, and turns off after an hour. The first turn-on is at 10 p.m. Judging by the logs, the drop happened at 2 a.m. last night. During the day, the dehumidifier turns on about twice an hour. The drop is more frequent.

Nerivec, 2024-09-06:
Ok, so, not a stress problem at all then.

Can you check your previous logs, see if the drops happen roughly around the same time of day (example: mostly around 4:30PM and around 2:00AM)? If not, can you see any kind of pattern if you put together "time of crashes + what's going on in the house + whatever else might affect the network or machine z2m is running on"?

EugenKomi, 2024-09-06:
> Ok, so, not a stress problem at all then.
>
> Can you check your previous logs, see if the drops happen roughly around the same time of day (example: mostly around 4:30PM and around 2:00AM)? If not, [...]

About 10 cicles on-off of smart plugs (all together commands). 

This manual simulation:
https://drive.google.com/file/d/1zuTz2ZBMPXkEEdC8q2k8QYFipDsIFcZ6/view?usp=sharing

I think it's a com port problem. I ordered HamGeek POE Zigbee 3.0 Coordinator, when it arrives I'll connect it via  local network and I think the problem will go away.

EugenKomi, 2024-09-06:
> Ok, so, not a stress problem at all then.
>
> Can you check your previous logs, see if the drops happen roughly around the same time of day (example: mostly around 4:30PM and around 2:00AM)? If not, [...]

Some correlations:
[image]

EugenKomi, 2024-09-09:
Initially I had HA installed in a Docker container on Synology DSM 6.2.4. I bought Dongle-E, but the problem was that I couldn't find the serial port in the device directory. Drivers and all other methods found on the Internet did not help.

Then I tried to install a virtual machine, stop containers with HA.fnd Z2MQTT. After mapping in the virtual machine everything worked, but the adapter hung, as described above.

I bought zigstar uzg-01 (Chenese clon). In the virtual machine it didn't want to work at all. The USB port was mapped, but the Z2MQTT didn't start, the TСP didn't work.

Then I stopped Z2MQTT in the VM and launched in the Docker container. And then everything worked fine without interruptions on the TCP. However, the Synology COM port still does not connect, but it is not necessary.

Well, I flashed Dongle - E to the router, installed HA in the neighboring container, and stopped the virtual machine. In this combination, everything works fine.

Mosquitto installed as Synology package.

stavros-k, 2024-09-20:
There seems to be a few reports around the Dongle E and the same error behavior.
Linking them here, hoping to provide/gather the information and potentially close this or the others to consolidate the issues.
https://github.com/Koenkk/zigbee2mqtt/issues/23861
https://github.com/Koenkk/zigbee2mqtt/issues/23706
https://github.com/Koenkk/zigbee2mqtt/issues/23761

joaofolino, 2024-09-23:
Thank you all for documenting and supporting this.
Read through all about this, because I was experiencing the issue: [the_boom.log](https://github.com/user-attachments/files/17094774/the_boom.log)
 In my case, I came across it while clicking away at this [Moes 4 button switch](https://www.zigbee2mqtt.io/devices/TS0044.html#tuya-ts0044).
I noticed the comment about [trying v8.0.1](https://github.com/Koenkk/zigbee2mqtt/discussions/23224#discussioncomment-10422347). Before trying it, I checked all the version in the reports and they are all 7.4.3 [GA]. I decided to move to [7.4.4](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl) before going the experimental route. I no longer experience this issue, running the 7.4.4 version.
I'm not that confident in this solution but would be interesting to see if it has any effect for others.

Nerivec, 2024-10-01:
Several users affected by this problem have indeed reported that updating to 7.4.4 fixed the problem.
If anyone is still affected, try to update, and report back if it still doesn't work for you.

borgmatthew, 2024-10-03:
The update to `7.4.4` fixed this issue for me. Thank you!

SelcalHF, 2024-10-15:
Another user, I found this thread after switching from RFlink to Zigbee using the Sonoff Zigbee 3 ZB-Dongle-E.
I flashed it to 7.4.3 and wasn't even aware of the firmware links above (7.4.3 was the highest I found).

Had the same problem, with the adapter crashing occasionally, but always when sending a lot of commands at once (ie when setting a scene with 10+ lights, using Home Assistant or Domoticz). Every time the ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT error, followed by adapter disconnected. Exactly as above.

So just to hopefully help others: for me, the update to 7.4.4 also fixed the issue!
Thanks for everybody's efforts.

Scandiravian, 2024-10-22:
I'm experiencing this issue when trying to perform an OTA of my IKEA Styrbar light switches. The issue has been present on both 7.4.4 and 8.0.2

jan0o0sch, 2024-10-24:
I am still having this problem with v7.4.4.
What can I do to help to solve this issue?

Scandiravian, 2024-10-25:
@Nerivec Here's a log from a few days ago where I started z2m, allowed join, and it then crashed when trying to pair with an IKEA Styrbar remote that is right next to the coordinator [log.log](https://github.com/user-attachments/files/17518240/log.log)

I have logs from a couple of situations, where this has happened (like updating and pairing with other devices). Let me know if you'd like me to send those or you need anything else I can provide

jan0o0sch, 2024-10-25:
> @jan0o0sch Can you provide the full `debug` logs (most importantly the few minutes before the error)?

I am sorry I set the log level to "debug" and restarted several times -- also the whole system, but it just wont show the debug enties. As it seems, "info" is the level Z2M prefers.  What am I doing wrong? 

```
advanced:
  log_level: debug
  adapter_concurrent: 1
  adapter_delay: 200
```

Nerivec, 2024-10-25:
Make sure Z2M is using the file you are editing and not another one (the one in the proper data folder). Also, `debug` is only present in the console/file by default (not in mqtt/frontend) to improve performance. https://www.zigbee2mqtt.io/guide/configuration/logging.html

Did you add the `adapter_concurrent: 1` for testing, or did you have that before?

`adapter_delay: 200` has no effect anymore for `ember`, you can remove it.

jan0o0sch, 2024-10-26:
> Make sure Z2M is using the file you are editing and not another one (the one in the proper data folder). Also, `debug` is only present in the console/file by default (not in mqtt/frontend) to [...]

OK, I removed both entries from the configuration and logged the output for ~30min ... resulting in ~20MB of logs -- that's why I zipped it. 

A reoccurring error is:
`error: 	zh:ember:ezsp: The adapter has run out of buffers, causing general malfunction. Remediate network congestion, if present.`

I have many TUYA devices, some of which are considered to be very communicative.

[z2m_logs_2024-10-26.zip](https://github.com/user-attachments/files/17530081/z2m_logs_2024-10-26.zip)

Nerivec, 2024-10-26:
Can you try with this firmware, see if you get any improvement?
[sonoff_zbdonglee_ncp-uart-sw_115200_8.0.2.0.gbl](https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2024.6.2/sonoff_zbdonglee_ncp-uart-sw_115200_8.0.2.0.gbl) / [release notes](https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2)

_Unfortunately, if the above isn't enough (higher firmware configs), we won't be able to do anything about the adapter being overloaded... the only possible solutions would be to get an adapter that has greater capacity (if enough...), or, reducing the load (some Tuya stuff is just too nasty at spamming to be able to achieve any kind of stability)._

Nerivec, 2024-10-26:
Can you provide the stuff in-between `[2024-10-26 16:17:42]` and the error? I'm assuming there wasn't that much time with zero activity.

jan0o0sch, 2024-10-26:
> Can you provide the stuff in-between `[2024-10-26 16:17:42]` and the error? I'm assuming there wasn't that much time with zero activity.

Yes, this is the whole log:

[z2m_logs_2024-10-26_16-07-56.zip](https://github.com/user-attachments/files/17531831/z2m_logs_2024-10-26_16-07-56.zip)

Nerivec, 2024-10-26:
I'm a little confused... Considering the three files follow each other, the logs go from `[2024-10-26 16:14:54]` to `[2024-10-26 16:23:03]`, and towards the very end is when I see `Zigbee2MQTT started!`...
That means your setup takes more than 8min to start (_more because I don't see the actual startup of Z2M in the logs_).
Am I getting that right?!

jan0o0sch, 2024-10-26:
> I'm a little confused... Considering the three files follow each other, the logs go from `[2024-10-26 16:14:54]` to `[2024-10-26 16:23:03]`, and towards the very end is when I see `Zigbee2MQTT [...]

Hi!
Yes, the logs start immediately after the addon tries to start.

Nerivec, 2024-10-26:
The actual startup of Z2M is not in there. Look for the log right before the first one in archive that has `Starting Zigbee2MQTT` at the top.

jan0o0sch, 2024-10-26:
> The actual startup of Z2M is not in there. Look for the log right before the first one in archive that has `Starting Zigbee2MQTT` at the top.

This text appears in the console only. Next line is something like "writing logs to file log.log" and after that, everything should be in the log files.

Nerivec, 2024-10-26:
`[2024-10-26 16:14:54] debug: 	z2m: Received Zigbee message`

That's the first line of the earliest log in the 3 logs you linked, which can only happen several seconds after startup.

jan0o0sch, 2024-10-27:
> `[2024-10-26 16:14:54] debug: z2m: Received Zigbee message`
> 
> That's the first line of the earliest log in the 3 logs you linked, which can only happen several seconds after startup.

Hi Nerivec,
I tried to get the whole log, that you requested. However I encountered a strange behaviour: The first few minutes of the log get overwritten or removed, I can't tell.

So, I started Z2M addon, and watched the log file ... the start was in there at the beginning, so I went away and checked again 30min later. When I downloaded the logs, the first few minutes were missing. 

Next try ... I waited for Z2M to start and saved the first few minutes of the log in a separate editor, then I waited until Z2M ended and downloaded the logs -- again, the first few minutes are missing, but I have the start saved in a separate file called `log_start.txt`. But there is still a gap of 2-3 minutes between `log_start.txt` and `log2.log`.

I don't know why the start entries of the log are overwritten but it is an error in my opinion .. but that's another issue.

Here are the log files:
[z2m_logs_2024-10-27_17-39-40.zip](https://github.com/user-attachments/files/17534848/z2m_logs_2024-10-27_17-39-40.zip)

I hope, you can use these new logs, despite the missing bit ... 

**Thank you very much for your support so far!**

Nerivec, 2024-10-27:
Can you try temporarily powering off all the `MmWave radar` sensors (so they are no longer on the network) and see how the startup goes after that?

Valorix, 2024-10-29:
I have the issue with my Sonof dongle (EmberZNet 7.4.4 [GA]) every time I tried a OTA update, stuck for now.

jan0o0sch, 2024-11-01:
> Can you try temporarily powering off all the `MmWave radar` sensors (so they are no longer on the network) and see how the startup goes after that?

Hi Nerivec,
I disconnected all the presence sensors (MmWave radar), but it still does not work. 
Error message again: error: 	`zh:ember:ezsp: The adapter has run out of buffers, causing general malfunction. Remediate network congestion, if present.`

Here is the log file:
[Uploading log_2024-11-01_12-00-00_wo_radar_sensors.log…]()

Nerivec, 2024-11-01:
It must be a specific device (or model) that is going crazy.
I've never seen the `run out of buffers` issue reported before...

_Link for logs doesn't work._

jan0o0sch, 2024-11-01:
> It must be a specific device (or model) that is going crazy. I've never seen the `run out of buffers` issue reported before...
> 
> _Link for logs doesn't work._

I re-uploaded the logs ... it should work now.

Nerivec, 2024-11-01:
With the mmwave
```
[2024-10-26 16:14:54] STARTED BEFORE THIS (don't have the log)
...
[2024-10-26 16:22:56] info: 	z2m: Started frontend on port 8099
```

Without the mmwave
```
[2024-11-01 12:00:06] info: 	z2m: Starting Zigbee2MQTT version 1.40.2 (commit #unknown)
...
[2024-11-01 12:02:48] info: 	z2m: Started frontend on port 8099
```

They make a huge mess on the network (and for Z2M) for sure. If you have a way to setup a second Z2M, I'd put all these in a separate network to prevent them from messing with the rest.

That device you mentioned is indeed suspicious. It appears to be duplicating reports (x4-5 times each time). Try unplugging it, see how it goes after that.

jan0o0sch, 2024-11-01:
> With the mmwave
> 
> ```
> [2024-10-26 16:14:54] STARTED BEFORE THIS (don't have the log)
> ...
> [2024-10-26 16:22:56] info: 	z2m: Started frontend on port 8099
> ```
> 
> Without the mmwave
> 
> ```
> [2024-11-01 12:00:06] info: 	z2m: Starting Zigbee2MQTT version 1.40.2 (commit #unknown)
> ...
> [2024-11-01 12:02:48] info: 	z2m: Started frontend on port 8099
> ```
> 
> They make a huge mess on the network (and for Z2M) for sure. If you have a way to setup a second Z2M, I'd put all these in a separate network to prevent them from messing with the rest (make sure to use very different network configs, see if [this](https://github.com/Nerivec/ember-zli/wiki/Stack#zigbee2mqtt-onboard-auto-configuration) can help maybe).
> 
> That device you mentioned is indeed suspicious. It appears to be duplicating reports (x4-5 times each time). Try unplugging it, see how it goes after that.

So this is a log again without the mmwave and also without the PwrMon that sent so many messages ... unfortunately, it sill doesn't run ...

log file here:
[log_2024-11-01_17-03-42_wo_radar_sensors_wo_PwrMon.log](https://github.com/user-attachments/files/17601629/log_2024-11-01_17-03-42_wo_radar_sensors_wo_PwrMon.log)

There is still one PwrMonSwitch online with ~480 messages -- I will try to disable that as well.

jan0o0sch, 2024-11-01:
I switched off the second PwrMonSwitch with a lot of communication, but in the end, Z2M still doesn't work ...   :-/

latest log:
[log_2024-11-01_17-36-19_wo_radar_sensors_wo_PwrMon.log](https://github.com/user-attachments/files/17602091/log_2024-11-01_17-36-19_wo_radar_sensors_wo_PwrMon.log)

Nerivec, 2024-11-01:
[2024-11-01 17:36:56] to [2024-11-01 17:39:20]
216 seconds
1399 received zigbee messages
411 from 'Innen/UG/Technik/sa_Thermostat_Fussbodenheizung'
398 from 'Innen/UG/Wohnen/sa_Thermostat_Fussbodenheizung'

Can you see any reason why these two thermostats are hammering Z2M (the others are not nearly as chatty)?

jan0o0sch, 2024-11-01:
> [2024-11-01 17:36:56] to [2024-11-01 17:39:20] 216 seconds 1399 received zigbee messages 411 from 'Innen/UG/Technik/sa_Thermostat_Fussbodenheizung' 398 from [...]

no, I can't ... 

We built a new home. Every room is set up in the exact same way: same Zigbee modules, same valves for heating ... AFAIK there is no reason why these two would need to communicate more than the others.

The thermostats act as routers though ...

nicola-spreafico, 2024-11-09:
Similar problem here: 

I have smlight slzb-06m coordinator, and in the last days it started giving me these errors
[image]

Resources on the VM where Z2M is running are free:
[image]

Did I reach some sort of limit and need to split devices on multiple networks with different coordinators?

jan0o0sch, 2024-11-09:
@nicola-spreafico 

It's not the coordinator, it's the devices that cause the trouble. Do you have TUYA devices in your network?

Nerivec, 2024-11-09:
What setup do you have?
What firmware version?
How many devices do you have on your network? Any known to be heavy spammers (like mmwave, some power monitoring, usually from Tuya)?

Before looking at splitting the setup, I'd say it's more likely you have a misbehaving device on the network that is hammering the adapter.

nicola-spreafico, 2024-11-09:
@jan0o0sch 
Right now I tried to reboot only the coordinator and for the moment is running smoothly, I did not add recently other devices that may have altered the network status

@Nerivec 
Z2M is running inside a dedicated LXC on Proxmox, connected to Home Assistant via MQTT
As coordinator I'm using smlight slzb-06m(core firmware v2.5.8, zigbee firmware 20240510)
Z2M is 1.41.0 commit: cc31ceab with coordinator revision 7.4.1 [GA]
I have a total 89 devices (Router: 45, End devices: 44)

I only have 2-3 Tuya devices which are installed since the beginning of time, never had a issue, all the others are mainly Aqara with a few other brand. I only have 2 presence detector, 15 power meters on DIN rail and about 20 smart plugs for power monitoring

Nerivec, 2024-11-09:
Do you have the logs from when the errors happened (if they are `debug`)?
You don't seem to have a network that would be prone to nasty behaviors (at least as far as spamming), could be just a one-off combination of "bad luck" that triggered this. Or maybe a manual "trigger-happy" activation of some kind (someone playing with a switch a bit too fast)?

Let us know if it happens again.

_FYI: I've seen quite a few issues recently that link back to the use of proxmox in the setup. Can't say if it's related, I've never used it myself, don't know it at all, could just be a coincidence, just thought you might want to know, investigate if there are any known problems with recent versions or something..._

nicola-spreafico, 2024-11-12:
[image]
Only lasted few days, so something is going on... I'm not sure how to debug this, the only idea I have is to split the network into another, but I'm risking encountering all other issue like channel overlap or something else

Nerivec, 2024-11-12:
Do you have the Z2M `debug` log file for that time? It would be good to identify what device(s) triggered this.
You probably should also try to unplug the adapter for a couple of minutes (from power), just in case.

nicola-spreafico, 2024-11-12:
Not entirely sure WHEN the problem started, I notice the problem as I walk in a room and the light does not automatically turn on or it turn on but with a very long delay (like 30-40s). Then i open the Z2M web interface and I notice the error.
Is there a way to find out when the problem started?

For the moment I rebooted again the coordinator and also I rolled back the coordinator firmware (the only change I made recently, don't know if can be a cause for this)

Nerivec, 2024-11-12:
> Is there a way to find out when the problem started?

With the Z2M log files, in the Z2M `data/log` folder; should have a timeline of recent activity, and be able to trace back to when that error first started, and what happens right before it occurs (which could point to the cause).

The coordinator firmware could be the cause (core or zigbee), but those versions you were on have been around for a while, and I haven't seen that issue before, so, it seems unlikely.

nicola-spreafico, 2024-11-12:
I was on firmware 2.3.6 since many months and not once I got that issue, until this morning i was on 2.5.8 (recently update) and now I'm back to 2.3.6. Let's wait again 2-3 days to see if the same problem occur
[image]

Nerivec, 2024-11-15:
Can you confirm exactly (core firmware + zigbee firmware for each):
- what version you had before this started
- what version you had when this started
- what version you reverted to that is currently not showing the issue

Just to make sure I've got the full set of information in one place.
If you can share log files, that would be helpful too (you can find me on Discord if you'd rather not share them here).
If the issue is still gone in a couple days, I'll report it up to SMLight.

[9 more replies]

closed 2025-06-29 by github-actions[bot]