Koenkk/zigbee2mqtt #23858: ZDO error: bind failed with status 'TABLE_FULL' (140)

bernard-dandrea, 2024-09-05:
### What happened?

Since version 1.39.1, problem with binding of LYWSD03MMC-z devices

See https://github.com/pvvx/ZigbeeTLc/issues/114

### What did you expect to happen?

Binding works as before, without error

### How to reproduce it (minimal and precise)

Install an LYWSD03MMC-z and change a value in the reporting tab (eg change min rep interval to new value)

[image]

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

3001-0122

### Adapter

LYWSD03MMC-z

### Setup

Directly with web interface of z2m

### Debug log

[2024-09-05 11:45:18] error: 	z2m: Failed to load external converter file '/var/www/html/plugins/z2m/core/class/../config/converters/Danfoss/icon.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'
Require stack:
- /var/www/html/plugins/z2m/resources/zigbee2mqtt/dist/util/utils.js
- /var/www/html/plugins/z2m/resources/zigbee2mqtt/dist/util/settings.js
- /var/www/html/plugins/z2m/resources/zigbee2mqtt/index.js)
[2024-09-05 11:45:18] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
[2024-09-05 11:45:18] error: 	z2m: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device
[2024-09-05 11:46:24] error: 	z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0xa4c13830c1998801/1 msRelativeHumidity from '0x00124b0029dc06b2/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140))'
[2024-09-05 11:57:55] error: 	z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0xa4c13830c1998801/1 msRelativeHumidity from '0x00124b0029dc06b2/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140))'
[2024-09-05 12:09:16] error: 	z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0xa4c13830c1998801/1 msTemperatureMeasurement from '0x00124b0029dc06b2/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140))'

Koenkk, 2024-09-05:
This is not an error that can be fixed from Z2M, the device binding table is full. Try removing some of the existing binds.

bernard-dandrea, 2024-09-05:
Hello Koen

I would agree with you if I had the same problem with all versions of zigbee2mqtt 

but update works for version 1.39.0 and not for 1.39.1 or 1.40.0

so something must have changed between these versions.

Koenkk, 2024-09-05:
@mrmaximas try to unbind all and see of that helps

bernard-dandrea, 2024-09-06:
> Yes, because earlier versions of z2m did not check the returned status code, so it silently ignored the error

I don't know if there was a silent error with versions prior to 1.9.1 but what is sure is that the update on bindings was done without problem

bernard-dandrea, 2024-09-06:
> @mrmaximas try to unbind all and see of that helps

[image]

if you try to disable the reporting in the Reportingt TAB, you have the error TABLE FULL

[image]

if you unbind in the Binding TAB, it works

you can then change values in Reporting table which by the way reactivate the corresponding bindings in the Bind table

once this is done for all bindings, you cannot update anymore due to the TABLE FULL error

other point: it is not clear to me what is stored in the binding tables of the device.

for LYWSD03MMC-z device, 8 entries are defined in the firmware for binding table which should be enough for the 4 entries in Binding tab + 3 entries found in Reporting tab. Are there other entries that are not displayed ? How can we see the full content of the binding table ?

mrmaximas, 2024-09-06:
unbind and bind again fixed the issue!

roberttco, 2024-10-09:
I too am seeing this and there is nothing to unbind.  There were no bindings to start with and I can't add a binding because I get a table full error.  If this really isn't an issue with Z2M then where can i look to see what is using the table space ?  Z2M version 1.40.2

vinodmishra, 2024-11-13:
The binding table is stored on the zigbee device itself I think. I had to hard reset the device before pairing it again. i.e. shorting RST and GND for 5 seconds before it worked

Extrme, 2024-12-13:
Had the same problem but, only with devices that were flashed with [devbis/z03mmc](https://github.com/devbis/z03mmc) before. @vinodmishra hint worked for me after flashing [pvvx/ZigbeeTLc](https://github.com/pvvx/ZigbeeTLc)

paradohh, 2025-01-19:
The following steps helped me avoid the error  'TABLE_FULL'
1. unbind all (genPollCtrl, PowerCfg, Humidity, Temperature)
2. I configured the necessary parameters in the section "Reporting". In my case "Min rep change" - Temperature 100 и Humidity 100 (like in sonoff snzb-02d) 
3. bind again genPollCtrl, PowerCfg. **Humidity, Temperature have already been chosen.**

closed 2025-03-28 by github-actions[bot]

miklosbagi, 2025-04-23:
> try to unbind all and see of that helps

This helped me, thanks

vinodmishra, 2025-08-20:
This issue keeps coming back for these and I had to reset it time. I have now reverted to the bluetooth custom firmware which uses BTHome and along with my network of bluetooth proxies is very reliable.