Koenkk/zigbee2mqtt #23861: Periodic disconnection of the adapter "ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT."

YoungRus, 2024-09-05:
### What happened?

The Zigbee Sonoff Dongle E adapter periodically disconnects and then reconnects.
The error occurs with both old and new ember firmware versions.

### How to reproduce it (minimal and precise)

Sudden disconnection at random times, then the device reconnects and everything works normally.

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

EmberZNet 7.4.3 [GA]

### Adapter

Sonoff ZBDongle-E 

### Setup

Docker x64 ubuntu server

### Debug log

[2024-09-04 21:03:52] error: 	z2m: Failed to read state of 'Switch Zal' after reconnect (ZCL command 0xa*****/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":10563,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":45},"zclSequence":197,"commandIdentifier":1} timed out after 10000ms))
[2024-09-04 21:04:32] error: 	z2m: Failed to read state of 'Switch San' after reconnect (ZCL command 0xa*****/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":24703,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":50},"zclSequence":199,"commandIdentifier":1} timed out after 10000ms))
[2024-09-04 21:04:39] error: 	z2m: Failed to read state of 'Switch VAn' after reconnect (ZCL command 0xa*****/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":52757,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":56},"zclSequence":200,"commandIdentifier":1} timed out after 10000ms))
[2024-09-04 21:06:54] error: 	z2m: Failed to read state of 'Switch Kuh' after reconnect (ZCL command 0xa*****/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":61550,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":121},"zclSequence":212,"commandIdentifier":1} timed out after 10000ms))
[2024-09-05 13:08:28] error: 	z2m: Failed to read state of 'Switch Kuh' after reconnect (ZCL command 0xa*****/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":61550,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":90},"zclSequence":65,"commandIdentifier":1} timed out after 10000ms))
[2024-09-05 13:08:43] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-09-05 13:08:43] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-09-05 13:08:43] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-09-05 13:08:43] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-09-05 13:08:51] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=174 Len=7].
[2024-09-05 13:09:13] error: 	z2m: Adapter disconnected, stopping

alexborro, 2024-09-08:
I'm also facing a similar issue with an SMLight SLZB-06M running latest firmware (20240510) connected to my HA by the USB/Serial connection.
I already have the `rtscts: false` in the config.

This is the log when it crashes:
```
[2024-09-08 15:10:43] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-09-08 15:10:43] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-09-08 15:10:43] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-09-08 15:10:43] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Total frames: RX=624, TX=776
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   DATA frames : RX=613, TX=156
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   DATA bytes  : RX=14408, TX=4015
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Retry frames: RX=9, TX=3
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=616
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Retry dupes     : RX=9
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-09-08 15:10:43] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=3
[2024-09-08 15:10:51] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=154 Len=7].
[2024-09-08 15:10:51] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_NCP_FATAL_ERROR. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=154 Len=7].
[2024-09-08 15:10:51] error: 	z2m: Publish 'set' 'state' to 'Luz de Teste' failed: 'Error: ZCL command 0x980c33fffea32f91/1 genLevelCtrl.moveToLevelWithOnOff({"level":241,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=40307] Failed to send request with status=FAIL.)'
[2024-09-08 15:11:13] info: 	zh:ember:uart:ash: Port closed.
[2024-09-08 15:11:13] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-09-08 15:11:13] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-09-08 15:11:13] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-09-08 15:11:13] error: 	z2m: Adapter disconnected, stopping
[2024-09-08 15:11:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-09-08 15:11:13] info: 	z2m: Disconnecting from MQTT server
[2024-09-08 15:11:13] info: 	z2m: Stopping zigbee-herdsman...
[2024-09-08 15:11:13] info: 	z2m: Stopped zigbee-herdsman
[2024-09-08 15:11:13] info: 	z2m: Stopped Zigbee2MQTT
```

And this is the boot following the crash:
```
[2024-09-08 15:11:18] info: 	z2m: Logging to console, file (filename: log.log)
[2024-09-08 15:11:18] info: 	z2m: Starting Zigbee2MQTT version 1.40.1 (commit #unknown)
[2024-09-08 15:11:18] info: 	z2m: Starting zigbee-herdsman (0.57.3)
[2024-09-08 15:11:18] info: 	zh:ember: Using default stack config.
[2024-09-08 15:11:18] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-09-08 15:11:18] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-09-08 15:11:18] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-09-08 15:11:18] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-09-08 15:11:18] info: 	zh:ember:uart:ash: Serial port opened
[2024-09-08 15:11:18] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-09-08 15:11:19] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-09-08 15:11:19] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-09-08 15:11:19] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-09-08 15:11:19] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2024-09-08 15:11:19] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.1 [GA]","build":0,"major":7,"minor":4,"patch":1,"special":0,"type":170}
[2024-09-08 15:11:20] info: 	zh:ember: [STACK STATUS] Network up.
[2024-09-08 15:11:20] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2024-09-08 15:11:20] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-09-08 15:11:20] info: 	z2m: zigbee-herdsman started (resumed)
[2024-09-08 15:11:20] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":1,"revision":"7.4.1 [GA]","special":0,"type":170},"type":"EmberZNet"}'

```

YoungRus, 2024-09-10:
I noticed that my adapter didn’t disconnect for 4 days, everything was fine, but now the errors started appearing again. I noticed that the errors are coming from switches without a neutral line, and it happens on all of them. After these errors, the adapter stops and restarts.

```
[2024-09-09 21:43:37] error: 	z2m: Failed to read state of 'vikl kor' after reconnect (ZCL command 0xa4c138aebd2930a1/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":26079,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":143},"zclSequence":3,"commandIdentifier":1} timed out after 10000ms))
[2024-09-09 21:44:01] error: 	z2m: Failed to read state of 'vikl kuh' after reconnect (ZCL command 0xa4c1384af629a0f4/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":61550,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":151},"zclSequence":7,"commandIdentifier":1} timed out after 10000ms))
[2024-09-09 21:44:06] error: 	z2m: Failed to read state of 'vikl kor' after reconnect (ZCL command 0xa4c138aebd2930a1/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":26079,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":157},"zclSequence":10,"commandIdentifier":1} timed out after 10000ms))
[2024-09-09 21:44:13] error: 	z2m: Failed to read state of 'vikl vann' after reconnect (ZCL command 0xa4c138df6af5fc26/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":52757,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":163},"zclSequence":11,"commandIdentifier":1} timed out after 10000ms))
[2024-09-09 21:44:24] error: 	z2m: Failed to read state of 'vikl kor' after reconnect (ZCL command 0xa4c138aebd2930a1/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":26079,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":165},"zclSequence":13,"commandIdentifier":1} timed out after 10000ms))
[2024-09-09 21:45:00] error: 	z2m: Failed to read state of 'vikl zal' after reconnect (ZCL command 0xa4c13858bdaf52df/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":10563,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":184},"zclSequence":16,"commandIdentifier":1} timed out after 10000ms))
[2024-09-09 21:45:18] error: 	z2m: Failed to read state of 'vikl kor' after reconnect (ZCL command 0xa4c138aebd2930a1/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":26079,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":199},"zclSequence":18,"commandIdentifier":1} timed out after 10000ms))
[2024-09-09 21:47:04] error: 	z2m: Publish 'set' 'state' to 'vikl san' failed: 'Error: ZCL command 0xa4c1389a49bbf571/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":24703,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":2,"options":4416,"groupId":0,"sequence":2},"zclSequence":37,"commandIdentifier":11} timed out after 10000ms)'
[2024-09-09 21:48:11] error: 	z2m: Failed to read state of 'vikl zal' after reconnect (ZCL command 0xa4c13858bdaf52df/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":10563,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":39},"zclSequence":48,"commandIdentifier":1} timed out after 10000ms))
[2024-09-09 21:48:41] error: 	z2m: Failed to read state of 'vikl kor' after reconnect (ZCL command 0xa4c138aebd2930a1/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":26079,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":53},"zclSequence":51,"commandIdentifier":1} timed out after 10000ms))
[... cut ...]
[2024-09-09 21:50:56] error: 	zh:controller:device: Read response to 0xa4c13858bdaf52df failed (ZCL command 0xa4c13858bdaf52df/1 genTime.readRsp({"localTime":779233837}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":70,"writeUndiv":false}) failed (~x~> [ZCL to=10563] Failed to send request with status=NETWORK_DOWN.))
[2024-09-09 21:51:12] error: 	z2m: Adapter disconnected, stopping
```

alexborro, 2024-09-16:
I realized that, at least in my case, I can 100% reproduce the issue by triggering a device firmware update, specially for endpoint devices (I think they are slow to comm).

This is an attempt to update the firmware of a Philips Hue Motion Sensor. I had to start the update at least 5 times before completing it (luckily this sensor stores the partially "downloaded" firmware and the next attempt start where the previous has finished). By the way, I was testing a Hue Dimmer just before the reset. 

And I also tried a few times to update 2 devices at the same time and it is impossible.

```
[2024-09-15 17:22:25] info: 	z2m: Update of '0x0017880106f763b6' at 2.33%, ≈ 47 minutes remaining
[2024-09-15 17:22:25] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0017880106f763b6', payload '{"battery":100,"illuminance":null,"illuminance_lux":null,"led_indication":null,"linkquality":148,"motion_sensitivity":null,"occupancy":false,"occupancy_timeout":0,"temperature":null,"update":{"installed_version":1107315168,"latest_version":1107323831,"progress":2.33,"remaining":2843,"state":"updating"},"update_available":null}'
[2024-09-15 17:22:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1', payload '{"action":"on_press","action_duration":null,"battery":2,"linkquality":124,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
[2024-09-15 17:22:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1', payload '{"action":"","action_duration":null,"battery":2,"linkquality":124,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
[2024-09-15 17:22:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1/action', payload 'on_press'
[2024-09-15 17:22:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1', payload '{"action":"on_press_release","action_duration":null,"battery":2,"linkquality":124,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
[2024-09-15 17:22:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1', payload '{"action":"","action_duration":null,"battery":2,"linkquality":124,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
[2024-09-15 17:22:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1/action', payload 'on_press_release'
[2024-09-15 17:22:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bedroom Light', payload '{"brightness":254,"color":{"x":0.4369,"y":0.4041},"color_mode":"color_temp","color_options":null,"color_temp":333,"linkquality":168,"power_on_behavior":null,"state":"ON","update":{"installed_version":268572245,"latest_version":268572245,"state":"idle"},"update_available":null}'
[2024-09-15 17:22:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1', payload '{"action":"off_press","action_duration":null,"battery":2,"linkquality":120,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
[2024-09-15 17:22:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1', payload '{"action":"","action_duration":null,"battery":2,"linkquality":120,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
[2024-09-15 17:22:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1/action', payload 'off_press'
[2024-09-15 17:22:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1', payload '{"action":"off_press_release","action_duration":null,"battery":2,"linkquality":120,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
[2024-09-15 17:22:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1', payload '{"action":"","action_duration":null,"battery":2,"linkquality":120,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
[2024-09-15 17:22:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Hue Dimmer1/action', payload 'off_press_release'
[2024-09-15 17:22:37] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-09-15 17:22:37] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-09-15 17:22:37] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-09-15 17:22:37] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash:   Total frames: RX=1191, TX=1506
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash:   DATA frames : RX=1176, TX=342
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash:   DATA bytes  : RX=25322, TX=16611
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash:   Retry frames: RX=13, TX=2
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=1161
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-09-15 17:22:37] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[... cut ...]
[2024-09-15 17:23:07] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-09-15 17:23:07] error: 	z2m: Adapter disconnected, stopping
[2024-09-15 17:23:07] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-09-15 17:23:07] info: 	z2m: Disconnecting from MQTT server
[2024-09-15 17:23:07] info: 	z2m: Stopping zigbee-herdsman...
[2024-09-15 17:23:07] info: 	z2m: Stopped zigbee-herdsman
[2024-09-15 17:23:07] info: 	z2m: Stopped Zigbee2MQTT

```

stavros-k, 2024-09-20:
I can reproduce this consistently.
If I try to pair 2 RODRET at the same time it will crash 100% of the times.

It even crashes If I do 2 RODRET devices one after the other.

```sh
info 2024-09-20 18:16:08z2m: Zigbee: allowing new devices to join.
info 2024-09-20 18:16:08zh:ember: [STACK STATUS] Network opened.
info 2024-09-20 18:16:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"o1f91-1"}'
info 2024-09-20 18:16:19zh:controller: Interview for '0xecf64cfffe35bb44' started
info 2024-09-20 18:16:19z2m: Starting interview of '0xecf64cfffe35bb44'
info 2024-09-20 18:16:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xecf64cfffe35bb44","ieee_address":"0xecf64cfffe35bb44","status":"started"},"type":"device_interview"}'
info 2024-09-20 18:16:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xecf64cfffe35bb44","ieee_address":"0xecf64cfffe35bb44"},"type":"device_announce"}'
info 2024-09-20 18:16:31zh:controller: Succesfully interviewed '0xecf64cfffe35bb44'
info 2024-09-20 18:16:31z2m: Successfully interviewed '0xecf64cfffe35bb44', device has successfully been paired
info 2024-09-20 18:16:31z2m: Device '0xecf64cfffe35bb44' is supported, identified as: IKEA RODRET wireless dimmer/power switch (E2201)
info 2024-09-20 18:16:32z2m: Configuring '0xecf64cfffe35bb44'
info 2024-09-20 18:16:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"RODRET wireless dimmer/power switch","exposes":[{"access":2,"category":"config","description":"Initiate device identification. This device is asleep by default.You may need to wake it up first before sending the identify command.","label":"Identify","name":"identify","property":"identify","type":"enum","values":["identify"]},{"access":5,"category":"diagnostic","description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"category":"diagnostic","description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["on","off","brightness_move_up","brightness_move_down","brightness_stop"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E2201","options":[{"access":2,"description":"Sets the duration of the identification procedure in seconds (i.e., how long the device would flash).The value ranges from 1 to 30 seconds (default: 3).","label":"Identify timeout","name":"identify_timeout","property":"identify_timeout","type":"numeric","value_max":30,"value_min":1},{"access":2,"description":"Simulate a brightness value. If this device provides a brightness_move_up or brightness_move_down action it is possible to specify the update interval and delta. The action_brightness_delta indicates the delta for each interval. ","features":[{"access":2,"description":"Delta per interval, 20 by default","label":"Delta","name":"delta","property":"delta","type":"numeric","value_min":0},{"access":2,"description":"Interval duration","label":"Interval","name":"interval","property":"interval","type":"numeric","unit":"ms","value_min":0}],"label":"Simulated brightness","name":"simulated_brightness","property":"simulated_brightness","type":"composite"}],"supports_ota":true,"vendor":"IKEA"},"friendly_name":"0xecf64cfffe35bb44","ieee_address":"0xecf64cfffe35bb44","status":"successful","supported":true},"type":"device_interview"}'
info 2024-09-20 18:16:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe35bb44', payload '{"action":null,"battery":83,"linkquality":224,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2024-09-20 18:16:37z2m: Successfully configured '0xecf64cfffe35bb44'
info 2024-09-20 18:17:00z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x8c6fb9fffec98a2a', payload '{"battery":86,"illuminance":6989,"illuminance_lux":5,"linkquality":204,"occupancy":false,"update":{"installed_version":16777303,"latest_version":16777316,"state":"available"},"update_available":null}'
info 2024-09-20 18:17:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xd44867fffe4fa6fe', payload '{"action":null,"battery":84,"linkquality":255,"update":{"installed_version":16777287,"latest_version":16777303,"state":"available"},"update_available":null}'
info 2024-09-20 18:17:31z2m: Update available for '0xd44867fffe4fa6fe'
info 2024-09-20 18:17:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe35bb44', payload '{"action":"brightness_move_down","action_rate":83,"battery":83,"linkquality":224,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2024-09-20 18:17:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe35bb44', payload '{"action":"","battery":83,"linkquality":224,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2024-09-20 18:17:37z2m:mqtt: MQTT publish: topic 'homeassistant/device_automation/0xecf64cfffe35bb44/action_brightness_move_down/config', payload '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0xecf64cfffe35bb44"],"manufacturer":"IKEA","model":"RODRET wireless dimmer/power s [... cut ...]
info 2024-09-20 18:17:53zh:ember:uart:ash: Bad lengths : RX=0
info 2024-09-20 18:17:53zh:ember:uart:ash: Bad ACK numbers : RX=0
info 2024-09-20 18:17:53zh:ember:uart:ash: Out of buffers : RX=0
info 2024-09-20 18:17:53zh:ember:uart:ash: Retry dupes : RX=6
info 2024-09-20 18:17:53zh:ember:uart:ash: Out of sequence : RX=0
info 2024-09-20 18:17:53zh:ember:uart:ash: ACK timeouts : RX=4
error 2024-09-20 18:18:01zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=73 Len=7].
```

Koenkk, 2024-09-20:
Could you try with the 8.0.1 firmware? https://github.com/Koenkk/zigbee2mqtt/discussions/23224#discussioncomment-10422347

stavros-k, 2024-09-20:
I just did.

```
[2024-09-21 00:53:53] info: 	zh:ember: Adapter version info: {"ezsp":14,"revision":"8.0.1 [GA]","build":270,"major":8,"minor":0,"patch":1,"special":0,"type":170}
```

Tried to switch 5 bulbs at the same time. 

```
2024-09-21 00:53:19zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-09-21 00:53:19zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-09-21 00:53:19zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-09-21 00:53:19zh:ember: Adapter fatal error: HOST_FATAL_ERROR
info 2024-09-21 00:53:19zh:ember:uart:ash: ASH COUNTERS since last clear:
info 2024-09-21 00:53:19zh:ember:uart:ash: Total frames: RX=87, TX=128
info 2024-09-21 00:53:19zh:ember:uart:ash: Cancelled : RX=0, TX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: DATA frames : RX=70, TX=47
info 2024-09-21 00:53:19zh:ember:uart:ash: DATA bytes : RX=1255, TX=639
info 2024-09-21 00:53:19zh:ember:uart:ash: Retry frames: RX=15, TX=3
info 2024-09-21 00:53:19zh:ember:uart:ash: ACK frames : RX=0, TX=77
info 2024-09-21 00:53:19zh:ember:uart:ash: NAK frames : RX=0, TX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: nRdy frames : RX=0, TX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: CRC errors : RX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: Comm errors : RX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: Length < minimum: RX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: Length > maximum: RX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: Bad controls : RX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: Bad lengths : RX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: Bad ACK numbers : RX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: Out of buffers : RX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: Retry dupes : RX=15
info 2024-09-21 00:53:19zh:ember:uart:ash: Out of sequence : RX=0
info 2024-09-21 00:53:19zh:ember:uart:ash: ACK timeouts : RX=3
```

Koenkk, 2024-09-21:
Interesting, @Nerivec do you have any clue?

Nerivec, 2024-09-21:
@stavros-k I assume you didn't rename the file before flashing? If so, you didn't pick the "max" version as Koenkk intended.
https://github.com/Nerivec/silabs-firmware-builder/actions/runs/10512841278/artifacts/1843552001
Can you try that one and report back?

YoungRus, 2024-09-22:
> Можете ли вы попробовать прошивку 8.0.1? [#23224 (комментарий)](https://github.com/Koenkk/zigbee2mqtt/discussions/23224#discussioncomment-10422347)

It seems that the errors I’m experiencing might still be due to the devices themselves, although it’s strange that the issue appears specifically with switches that don’t have a neutral line, and it happens with all of them at once.

After reconnecting the devices, I get the following message for all of them:
`zh:ember: [ZDO] Device '10563' is only compliant with revision '21' of the ZigBee specification (current revision: 22).`
This shouldn't affect the device reconnections or stop z2m, right?

Nerivec, 2024-09-22:
@YoungRus Seems some networks are having troubles with message volume/rate. Try the firmware mentioned above, it increases the capabilities of the adapter, and so far has been able to fix that kind of issue.

No, that message is only to inform that the device is not really recent (the ZigBee version it uses is older). 21 shouldn't bring too many problems though.

stavros-k, 2024-09-22:
> @stavros-k I assume you didn't rename the file before flashing? If so, you didn't pick the "max" version as Koenkk intended. https://github.com/Nerivec/silabs-firmware-builder/actions/runs/10512841278/artifacts/1843552001 Can you try that one and report back?

Sorry I thought the goal was to use 8.x.x version.

Tried `max` and boy, oh boy! I could spam all lights at the same time on and off and had ZERO crashes.

stavros-k, 2024-09-22:
I tried to "push" it a bit.
2 OTA updates running fine + tried to start a 3rd.

```
[2024-09-22 16:54:14] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=243 Len=54]
[2024-09-22 16:54:14] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":64,"groupId":0,"sequence":247}], [packetInfo:{"senderShortId":37021,"senderLongId":"0x0000000000000000","bindingIndex":255,"addressIndex":255,"lastHopLqi":0,"lastHopRssi":0,"lastHopTimestamp":25}], [messageContents=015803007c11cd1157000001329600003f]
[2024-09-22 16:54:14] debug: 	zh:controller: Received payload: clusterID=25, address=37021, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=0, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":88,"commandIdentifier":3},"payload":{"fieldControl":0,"manufacturerCode":4476,"imageType":4557,"fileVersion":16777303,"fileOffset":38450,"maximumDataSize":63},"command":{"ID":3,"response":5,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35},{"name":"fileOffset","type":35},{"name":"maximumDataSize","type":32}],"name":"imageBlockRequest"}}
[2024-09-22 16:54:14] debug: 	z2m: Received Zigbee message from 'Office Light Switch', type 'commandImageBlockRequest', cluster 'genOta', data '{"fieldControl":0,"fileOffset":38450,"fileVersion":16777303,"imageType":4557,"manufacturerCode":4476,"maximumDataSize":63}' from endpoint 1 with groupID 0
[2024-09-22 16:54:15] debug: 	zh:ember:uart:ash: <--- [FRAME type=ERROR]
[2024-09-22 16:54:15] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-09-22 16:54:15] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-09-22 16:54:15] error: zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-09-22 16:54:15] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-09-22 16:54:15] debug: 	zh:controller: Adapter disconnected
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Total frames: RX=4831, TX=6272
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   DATA frames : RX=4826, TX=1525
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   DATA bytes  : RX=139387, TX=116681
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Retry frames: RX=3, TX=3
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=4743
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Retry dupes     : RX=3
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-09-22 16:54:15] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=3
```

stavros-k, 2024-09-22:
Single OTA update still crashed it :(

```
[2024-09-22 17:30:12] debug: 	zhc:ota:common: Update at 30.13%, remaining 3818.489138566131 seconds
[2024-09-22 17:30:12] info: 	z2m: Update of 'Living Room Light Switch' at 30.13%, ≈ 64 minutes remaining
[2024-09-22 17:30:12] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Light Switch', payload '{"action":null,"battery":83,"linkquality":0,"update":{"installed_version":16777287,"latest_version":16777303,"progress":30.13,"remaining":3818,"state":"updating"},"update_available":null}'
[2024-09-22 17:30:12] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-22 17:30:12] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=0)
[2024-09-22 17:30:12] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=7](frmRx=7) Added to rxQueue
[2024-09-22 17:30:12] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=0)
[2024-09-22 17:30:12] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=7 Len=26]
[2024-09-22 17:30:12] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=DIRECT], [indexOrDestination=16413], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":190}], [messageTag=19]
[2024-09-22 17:30:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-22 17:30:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=0)
[2024-09-22 17:30:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=0](frmRx=0) Added to rxQueue
[2024-09-22 17:30:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=0)
[2024-09-22 17:30:13] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=7 Len=54]
[2024-09-22 17:30:13] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":21}], [packetInfo:{"senderShortId":16413,"senderLongId":"0x0000000000000000","bindingIndex":255,"addressIndex":255,"lastHopLqi":0,"lastHopRssi":0,"lastHopTimestamp":25}], [messageContents=012703007c11cd1157000001363c01003f]
[2024-09-22 17:30:13] debug: 	zh:controller: Received payload: clusterID=25, address=16413, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=0, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":39,"commandIdentifier":3},"payload":{"fieldControl":0,"manufacturerCode":4476,"imageType":4557,"fileVersion":16777303,"fileOffset":80950,"maximumDataSize":63},"command":{"ID":3,"response":5,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35},{"name":"fileOffset","type":35},{"name":"maximumDataSize","type":32}],"name":"imageBlockRequest"}}
[2024-09-22 17:30:13] debug: 	z2m: Received Zigbee message from 'Living Room Light Switch', type 'commandImageBlockRequest', cluster 'genOta', data '{"fieldControl":0,"fileOffset":80950,"fileVersion":16777303,"imageType":4557,"manufacturerCode":4476,"maximumDataSize":63}' from endpoint 1 with groupID 0
[2024-09-22 17:30:13] debug: 	zhc:ota:common: Request offsets: fileOffset=80950 pageOffset=0 maximumDataSize=63
[2024-09-22 17:30:13] debug: 	zhc:ota:common: Payload offsets: start=80950 end=81000 dataSize=50
[2024-09-22 17:30:13] debug: 	zh:controller:endpoint: CommandResponse 0xecf64cfffe35bb44/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":4557,"fileVersion":16777303,"fileOffset":80950,"dataSize":50,"data":{"type":"Buffer","data":[6,72,244,247,85,249,8,155,16,43,3,208,187,33,3,72,244,247,78,249,18,176,112,189,0,68,64,4,16,40,4,0,48,181,13,70,0,33,145,176,4,70,36,34,7,168,6,145,33,240]}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false})
[2024-09-22 17:30:13] debug: 	zh:controller:endpoint: Request Queue (0xecf64cfffe35bb44/1): send request
[2024-09-22 17:30:13] debug: 	zh:ember: ~~~> [ZCL to=16413 apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":39,"commandIdentifier":5}]
[2024-09-22 17:30:13] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=8 Len=89]
[2024-09-22 17:30:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=1](ackRx=0)
[2024-09-22 17:30:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-09-22 17:30:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=0 frmTx=1)
[2024-09-22 17:30:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=1](frmRx=1) Added to rxQueue
[2024-09-22 17:30:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=1)
[2024-09-22 17:30:13] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=8 Len=10]
[2024-09-22 17:30:13] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=191 messageTag=20 status=OK]
[2024-09-22 17:30:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[... cut ...]
[2024-09-22 17:30:51] debug: 	z2m: Saving state to file /app/data/state.json
[2024-09-22 17:30:51] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-09-22 17:30:51] info: 	z2m: Disconnecting from MQTT server
[2024-09-22 17:30:51] info: 	z2m: Stopping zigbee-herdsman...
[2024-09-22 17:30:51] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-09-22 17:30:51] info: 	z2m: Stopped zigbee-herdsman
[2024-09-22 17:30:51] info: 	z2m: Stopped Zigbee2MQTT

```

Nerivec, 2024-09-22:
Crash on multiple OTAs is not really surprising; it's entirely dependent on the stability of the underlaying network (quality of the routers, number of hops, etc.).

Can you list the devices you have on your network (don't care about IEEE/nwk, just the models)?
How far away from the coordinator is that last device you tried to update? _What are the chances that it's connected through a router, not directly?_
Can you post a screenshot of the network map (try to make it as legible as possible)? At least the path from the coordinator to the device you tried to update last.
You can also try to lower the `adapter_concurrent` option, see how that goes (default is 16, try 8). [docs](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#advanced-configuration)

stavros-k, 2024-09-22:
> Crash on multiple OTAs is not really surprising; it's entirely dependent on the stability of the underlaying network (quality of the routers, number of hops, etc.).

Oh yea, I just wanted to try to stress it a bit.

> Can you list the devices you have on your network (don't care about IEEE/nwk, just the models)? 

```
|     |                                 |            |           |     |
| --- | ------------------------------- | ---------- | --------- | --- |
| 1   | 0x5cc7c1fffe5ddcb8              | IKEA E2001 | E2002     | 122 |
| 2   | 0x8c6fb9fffec98a2a              | IKEA       | E2134     | 37  |
| 3   | 0xd87a3bfffe0fea43              | IKEA       | E2134     | 38  |
| 4   | Office Light Switch             | IKEA       | E2201     | 0   |
| 5   | Kitchen Light Switch            | IKEA       | E2201     | 0   |
| 6   | Hallway Light Switch            | IKEA       | E2201     | 255 |
| 7   | Server Room Light Switch        | IKEA       | E2201     | 0   |
| 8   | Bedroom Light Switch            | IKEA       | E2201     | 0   |
| 9   | Living Room Light Switch        | IKEA       | E2201     | 0   |
| 10  | Entrance Light Switch           | IKEA       | E2201     | 0   |
| 11  | Bathroom Light Switch           | IKEA       | E2201     | 0   |
| 12  | Bedroom Light Bulb              | IKEA       | LED2109G6 | 14  |
| 13  | Entrance Light Bulb             | IKEA       | LED2109G6 | 15  |
| 14  | Kitchen Light Bulb              | IKEA       | LED2109G6 | 16  |
| 15  | Living Room Light Bulb          | IKEA       | LED2109G6 | 0   |
| 16  | Office Light Bulb               | IKEA       | LED2109G6 | 0   |
| 17  | Server Room Light Bulb          | IKEA       | LED2109G6 | 0   |
| 18  | Bathroom Mirror Light Bulb      | IKEA       | LED2109G6 | 0   |
| 19  | Bathroom Light Bulb             | IKEA       | LED2109G6 | 28  |
| 20  | Hallway Light Bulb              | IKEA       | LED2111G6 | 13  |
| 21  | some-long-random-id             | Aqara      | MCCGQ11LM | 104 |
| 22  | 0x00158d000ae4e67f              | Aqara      | MCCGQ11LM | 9   |
| 23  | Living Room Door (Right Side)   | Aqara      | MCCGQ11LM | 193 |
| 24  | Server Room Door (Right Side)   | Aqara      | MCCGQ11LM | 103 |
| 25  | Server Room Door (Left Side)    | Aqara      | MCCGQ11LM | 21  |
| 26  | Living Room Door (Left Side)    | Aqara      | MCCGQ11LM | 123 |
| 27  | Office Side Window (Left Side)  | Aqara      | MCCGQ11LM | 103 |
| 28  | 0x00158d000ae4cd81              | Aqara      | MCCGQ11LM | 128 |
| 29  | Kitchen Window (Left Side)      | Aqara      | MCCGQ11LM | 56  |
| 30  | Office Side Window (Right Side) | Aqara      | MCCGQ11LM | 203 |
| 31  | Office Rear Window (Left Side)  | Aqara      | MCCGQ11LM | 11  |
| 32  | Office Rear Window (Right Side) | Aqara      | MCCGQ11LM | 11  |
| 33  | Main Door                       | Aqara      | MCCGQ11LM | 12  |
| 34  | Kitchen Window (Right Side)     | Aqara      | MCCGQ11LM | 200 |
| 35  | Bedroom Door (Right Side)       | Aqara      | MCCGQ11LM | 71  |
| 36  | Bedroom Door (Left Side)        | Aqara      | MCCGQ11LM | 7   |
| 37  | Magic Cube                      | Aqara      | MFKZQ01LM | 108 |
| 38  | Entrance Motion Sensor          | Aqara      | RTCGQ11LM | 30  |
| 39  | Kitchen Motion Sensor           | Aqara      | RTCGQ11LM | 25  |
```

> How far away from the coordinator is that last device you tried to update? 

Like 1-2meters.

> _What are the chances that it's connected through a router, not directly?_ 

So the coordinator is on my server on the rack ~1-2meters away.
Above the rack there is a bulb, the switch I was upgrading is at my desk (1-2meters from the rack)
And right above the desk is a bulb.

> Can you post a screenshot of the network map (try to make it as legible as possible)? 
> At least the path from the coordinator to the device you tried to update last. 

The device I was upgrading is the `Living Room Light Switch`
Which seems to be connected via the `Server Room Light Bulb`
[image]

> You can also try to lower the `adapter_concurrent` option, see how that goes (default is 16, try 8). [docs](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#advanced-configuration)

I'll try that soon

stavros-k, 2024-09-22:
Single update stopped at 66.5% with the same error again. 
It was direct connected to the coordinator. I'll now try changing the adapter_concurrent

stavros-k, 2024-09-22:
Concurrency of 8 
```
error 2024-09-22 20:41:37zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-09-22 20:41:37zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-09-22 20:41:37zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-09-22 20:41:37zh:ember: Adapter fatal error: HOST_FATAL_ERROR
info 2024-09-22 20:41:37zh:ember:uart:ash: ASH COUNTERS since last clear:
info 2024-09-22 20:41:37zh:ember:uart:ash: Total frames: RX=2250, TX=3001
info 2024-09-22 20:41:37zh:ember:uart:ash: Cancelled : RX=0, TX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: DATA frames : RX=2245, TX=752
info 2024-09-22 20:41:37zh:ember:uart:ash: DATA bytes : RX=65862, TX=62421
info 2024-09-22 20:41:37zh:ember:uart:ash: Retry frames: RX=3, TX=3
info 2024-09-22 20:41:37zh:ember:uart:ash: ACK frames : RX=0, TX=2245
info 2024-09-22 20:41:37zh:ember:uart:ash: NAK frames : RX=0, TX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: nRdy frames : RX=0, TX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: CRC errors : RX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: Comm errors : RX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: Length < minimum: RX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: Length > maximum: RX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: Bad controls : RX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: Bad lengths : RX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: Bad ACK numbers : RX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: Out of buffers : RX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: Retry dupes : RX=3
info 2024-09-22 20:41:37zh:ember:uart:ash: Out of sequence : RX=0
info 2024-09-22 20:41:37zh:ember:uart:ash: ACK timeouts : RX=3
error 2024-09-22 20:41:45zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=238 Len=10].
info 2024-09-22 20:42:07zh:ember:uart:ash: Port closed.
info 2024-09-22 20:42:07zh:ember:uart:ash: ======== ASH stopped ========
info 2024-09-22 20:42:07zh:ember:ezsp: ======== EZSP stopped ========
info 2024-09-22 20:42:07zh:ember: ======== Ember Adapter Stopped ========
error 2024-09-22 20:42:07z2m: Adapter disconnected, stopping
```

alexborro, 2024-09-23:
This is the same behavior I have. It looks like an OTA is the "easiest" way to trigger this issue - you need a device to update, or course.
I also tried placing sensors just besides the coordinator (20cm from the coordinator antenna) and it doesn't help, so I'm sure how the rest of the network influences in this issue.

Nerivec, 2024-09-23:
@alexborro Does your network map also look similar to the one above (many devices directly on the coordinator)?

alexborro, 2024-09-23:
> @alexborro Does your network map also look similar to the one above (many devices directly on the coordinator)?

I don't think so, because my coordinator needs to be placed in a corner of my house and then I need to rely on routers.
This is a map of the network and you can see almost all end-devices are connected to a router, not to the coordinator.
[image]

EMAKASS, 2024-09-25:
I updated the firmware on my Sonoff adapter to **Ember 7.4.4 GA**. It's been **3 days**, and so far, everything is **okay**. There are some minor reconnections with the switches, but no disconnections so far. I’ll keep monitoring and report back later.

Nerivec, 2024-09-25:
@stavros-k Could you try with [7.4.4](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/), see if you can make it crash like you did before? _I don't see anything in the release notes from silabs that would explain this._ If it works, can you also confirm with [8.0.2](https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2) since I'd expect them to be somewhat aligned (at least on bug fixes...).

stavros-k, 2024-09-25:
With a 7.4.4 is the first time I managed to complete a single OTA update.
I'll try now 8.0.2.

Nerivec, 2024-09-26:
Thanks for the detailed feedback!
Looks like something in 7.4.4 is making your OTAs happier... too bad silabs release notes don't really point to something in that regard...

stavros-k, 2024-09-26:
> Thanks for the detailed feedback! Looks like something in 7.4.4 is making your OTAs happier... too bad silabs release notes don't really point to something in that regard...

I do have an order coming for an SMLIGHT, but happy I got the sonoff working in the mean time!
Thanks for the help

alexborro, 2024-09-26:
Nice that you mentioned the SMLight. I have the 06M and it is running the latest SMLight firmware, which is "20240510".
I guess it is the 7.4.1 because the build data matches. 
And I'm assuming that if I want to flash a new one, I have to manually download and flash from https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/slzb-06m or ask SMLight to make it available throught 06M OTA Interface, right?

EMAKASS, 2024-09-30:
Worked for 7 days without disconnections, I can say that the **7.4.4** **firmware** solved my reconnection issue.
The only thing I noticed was after I changed the antenna to a 6 dBi one, I started getting a lot of errors with my switches. I had to revert back to the stock antenna.

adeepn, 2024-10-07:
Even with 7.4.4 with increased buffers we sometimes fall in this error.

Nerivec, 2024-10-07:
> 7.4.4 with increased buffers

You mean custom firmware? If yes, what increased values exactly?

Note: That error is not necessarily a firmware issue, but could just be generic network failure depending on the scenario.

adeepn, 2024-10-07:
> > 7.4.4 with increased buffers
> 
> You mean custom firmware? If yes, what increased values exactly?

```
+  SL_IOSTREAM_USART_VCOM_RX_BUFFER_SIZE: 64
+  SL_ZIGBEE_NEIGHBOR_TABLE_SIZE: 26
+  SL_ZIGBEE_SOURCE_ROUTE_TABLE_SIZE: 200
```
 
> Note: That error is not necessarily a firmware issue, but could just be generic network failure depending on the scenario.

Yes, this is most likely an indication of a busy radio channel. the problem is that this is not handled as a normal situation, but puts the z2m into a reboot cycle.

Nerivec, 2024-10-07:
Whenever an ERROR frame is received from the adapter, there is no other choice but to reboot the adapter as it is in "failed" state. `ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT` is not a "normal situation" as far as the adapter is concerned.

Since there is no indication in silabs notes as to why 7.4.4 fixes it for most users, I can't be sure exactly what's the source of the issue.

PS: The mentioned "increased values" would not work for 7.4.4 as these are the new names for 8.0.0 and above.

adeepn, 2024-10-07:
> Whenever an ERROR frame is received from the adapter, there is no other choice but to reboot the adapter as it is in "failed" state. `ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT` is not a "normal situation" as far as the adapter is concerned.

If this error can occur due to network status - it is not normal to terminate z2m. it is normal to handle the error, report it and work on (by restarting the adapter if needed).

> Since there is no indication in silabs notes as to why 7.4.4 fixes it for most users, I can't be sure exactly what's the source of the issue.

If refer to the semantic name of the error, it is a timeout error for sending/waiting messages. the normal reaction is to resend/wait them again. it seems logical to me.
but of course requires additional research or an official response from the silabs

> 
> PS: The mentioned "increased values" would not work for 7.4.4 as these are the new names for 8.0.0 and above.

ps. As for your message about "wrong names", I left them that way on purpose to make it easier for you to understand. sorry for my bad english, I'll try to explain more next time.

Nerivec, 2024-10-07:
This is not an error that should occur. When errors like this are thrown by the adapter, something is definitely wrong (beyond issues related to network state), they are reserved for "something made the adapter crash".
In this case, it seems the adapter is missing ACKs (from previous logs, I don't see Z2M missing any, so, it must be something about the firmware, this is supported by the fact that 7.4.4 seemingly handles this better than previous versions), which means it goes out of sync and can't continue normal operation.
It must be linked to very specific actions though, as most users don't have this problem (_possibly specific devices, some are just too terrible for normal operation_). I've put the test Dongle-E I have under extreme stress (for its capabilities) several times, and never had this in 7.4.x or 8.0.2...

_Rebooting the adapter requires rebooting much of herdsman to properly sync-up various states. Rebooting the adapter internally to the driver (like it was in the past), brings other issues because Z2M doesn't have that proper sync (this is not supported by the API, at least for now).
In effect, an adapter reboot amounts to pretty much a Z2M reboot, since it accounts for 90% of the logic involved. So, until a possible future where Z2M handles adapter-level restarts, it is much cleaner to reboot Z2M and avoid sync issues altogether._

adeepn, 2024-10-08:
> this is supported by the fact that 7.4.4 seemingly handles this better than previous

I can only add that the bigger impact on the error was from the option SL_IOSTREAM_USART_VCOM_RX_BUFFER_SIZE: 64
with the old value of 32 the behavior of firmware 7.4.4 was no different from 7.4.3.
Now I have an OTA update sometimes with this error, otherwise the adapter stopped resetting by itself in everyday mode.

CodeForLove83, 2024-10-08:
I can confirm that happens very often even with 7.4.4. I can trigger deterministically through OTA updates. I've a network of more than 70 Legrand/BTicino devices. All of them trigger the same issue during the update. 

I'm available to make any test if you need.

alexborro, 2024-10-09:
I'm a little bit lost in this topic and since I'm only using Z2M for 1 month, I'm a little bit lost in firmware's responsibilities. 
But if I understand correctly, the firmware built in the repo https://github.com/darkxst/silabs-firmware-builder uses the sources provided by Silabs, right ? And for me it seems quite clear it is a bug in the chipset firmware - or at least some some parameter not well tuned. Is it possible to contact Silabs somehow about it ?
Please let me know if I'm wrong in my assumptions.

Nerivec, 2024-10-09:
The behaviors reported are a bit inconsistent, since adeepn reports it doesn't work with "regular" 7.4.4 (need to customize firmware parameters), while for others it does work fine.
For networks with devices that have always been problematic with OTA (Legrand, BTcino, etc.), it's also a bit harder to determine what's what.

I'll contact silabs to get a better understanding of what changed regarding this in 7.4.4, since it doesn't appear to be in the release notes.

CodeForLove83, 2024-10-09:
Hi @Nerivec, is there anything I can do?
If you need I can do more testing.
Thanks

adeepn, 2024-10-10:
> The behaviors reported are a bit inconsistent, since adeepn reports it doesn't work with "regular" 7.4.4 (need to customize firmware parameters), while for others it does work fine. For networks [...]

i can provide source code with defines to compare in pm. also we will try to build with new sdk and i'll return with results.

donbernhardo, 2024-11-13:
I had the exact same issues with "ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT" until I flashed the [8.0.2 Release with the stability improvements.](https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2).

There is a recent knowledge base article from Silicon Labs on "[Guidelines for Large/Dense Networks with EmberZNet](https://community.silabs.com/s/article/guidelines-for-large-dense-networks-with-emberznet-pro?language=en_US)" with lots of tips and best practise for stable networks. On the end are also some recommended parameters one can try to improve performance. Some of them have been modified in the FW build linked above.

EMAKASS, 2024-11-19:
The issue reappeared after a few more devices were added to the system.

Valorix, 2024-11-20:
I have the same issue, unable to make OTA update since a month and I have to deal with adapter crashes for now. Sometimes it crashes without restarting so I have my heating running during off time. It start to be annoying. Firmware `EmberZNet 7.4.4 [GA]`

EMAKASS, 2024-11-20:
Ctqxf

> У меня та же проблема, не могу сделать обновление OTA уже месяц, и мне приходится иметь дело с крахами адаптера. Иногда он крашится без перезапуска, поэтому у меня работает отопление в выключенное время. Это начинает раздражать. Прошивка`EmberZNet 7.4.4 [GA]`

I just updated the Aqara thermostat head via OTA. It took 3-4 attempts, and the adapter also went into error mode, but the firmware update was successful.

rentonmark, 2024-11-21:
Hello, I'm also experiencing OTA update issues with Z2m using my ZBDongle-e. 
- I've tried both firmware versions 7.4.4 and 8.0.2. 
- There's no Wi-Fi conflict as I've tested with it disabled.
- Using a 50cm USB 3.1 extension cable).

I'm getting this error: 

```
zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error zh:ember: Adapter fatal error: HOST_FATAL_ERROR
info zh:ember:uart:ash: ASH COUNTERS since last clear:
info zh:ember:uart:ash: Total frames: RX=1957, TX=2497
info zh:ember:uart:ash: Cancelled : RX=0, TX=0
info zh:ember:uart:ash: DATA frames : RX=1952, TX=538
info zh:ember:uart:ash: DATA bytes : RX=58140, TX=18284
info zh:ember:uart:ash: Retry frames: RX=4, TX=3
info zh:ember:uart:ash: ACK frames : RX=0, TX=1956
info zh:ember:uart:ash: NAK frames : RX=0, TX=0
info zh:ember:uart:ash: nRdy frames : RX=0, TX=0
info zh:ember:uart:ash: CRC errors : RX=0
info zh:ember:uart:ash: Comm errors : RX=0
info zh:ember:uart:ash: Length < minimum: RX=0
info zh:ember:uart:ash: Length > maximum: RX=0
info zh:ember:uart:ash: Bad controls : RX=0
info zh:ember:uart:ash: Bad lengths : RX=0
info zh:ember:uart:ash: Bad ACK numbers : RX=0
info zh:ember:uart:ash: Out of buffers : RX=0
info zh:ember:uart:ash: Retry dupes : RX=4
info zh:ember:uart:ash: Out of sequence : RX=0
info zh:ember:uart:ash: ACK timeouts : RX=3
error zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=50 Len=10].
error zh:ember:ezsp: ERROR Transaction failure; status=ASH_NCP_FATAL_ERROR. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=50 Len=10].
info zh:ember:uart:ash: Port closed.
info zh:ember:uart:ash: ======== ASH stopped ========
info zh:ember:ezsp: ======== EZSP stopped ========
info zh:ember: ======== Ember Adapter Stopped ========
error z2m: Adapter disconnected, stopping
```

Valorix, 2024-11-27:
I've juste upgrade from `RPi 3B+` to `RPi 5 8go` and no more issues for now

Mikei386, 2024-12-02:
Having exactly the same problem.

I'm running Z2m inside Docker on an i5 x64 Debian system. Tried both Firmware Versions 7.4.4 and 8.0.2.
Tried to update my 5 "IKEA INSPELNING Smart plugs". 

Had to try it up to 10 times until one update would go to 100% without crashing.
Same with one Hue Bulb I tired.

Since I have 2 Dongles, I might start a parallel running Network with ZHA and one of those plugs, to see if it's behaving any differently .

Mikei386, 2024-12-03:
I tried updating the firmware on one plug with zha in a second zigbee network and it worked at the first try. But that could be because there is only this device in the network at the moment. I then tried another plug in Z2m and it errored out again at about 96% (see screenshots)

[image]
[image]

Not sure if this even helps in any way…

Nerivec, 2024-12-03:
Can I get a clearer picture of users still affected by this please?
- Adapter model
- Firmware version
- For you, is it happening only during firmware updates, or during normal operation?
  - If during normal operation, how reasonably certain are you that your network contains or does not contain "spammy devices" (Tuya, power meters, mmWaves...)?

Also, if anyone affected has updated to current dev branch (2.0.0), can you try tweaking the new OTA setting (`image_block_response_delay`) that throttles messages to see if it affects this bug at all? See the preview docs here: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/docs/guide/usage/ota_updates.md#advanced-configuration
Try large increases at first to see if the crash still happens (triple, double).

antst, 2024-12-03:
1) #24807 one also belongs here

2) @Nerivec , I experience this only during OTA. I run your latest build of 8.0.2 .

Here my current list of device types: (already slightly different from one in thread above)
```
["DONGLE-E","SONOFF"]
["LCT010","Signify Netherlands B.V."]
["LWA004","Signify Netherlands B.V."]
["LWO001","Signify Netherlands B.V."]
["PLUG EU EM T","LEDVANCE"]
["RB 286 C","innr"]
["RWL021","Philips"]
["Remote Control N2","IKEA of Sweden"]
["Smart Wall Switch ","Calex "]
["TRADFRI SHORTCUT Button","IKEA of Sweden"]
["TRADFRI bulb E14 W op/ch 400lm","IKEA of Sweden"]
["TRADFRI bulb E14 WS 470lm","IKEA of Sweden"]
["TRADFRI bulb E27 WW G95 CL 470lm","IKEA of Sweden"]
["TRADFRI on/off switch","IKEA of Sweden"]
["TRADFRI remote control","IKEA of Sweden"]
["TRADFRIbulbE27WWclear250lm","IKEA of Sweden"]
["TS011F","_TZ3000_1obwwnmq"]
["TS0501A","_TZ3000_7dcddnye"]
["TS0502A","_TZ3000_oborybow"]
["TS0505A","_TZ3000_gek6snaj"]
["TS0505A","_TZ3000_v7fkcekx"]
["TS0505B","_TZ3210_hxtfthp5"]
["TS0505B","_TZ3210_r0xgkft5"]
["TY0202","_TZ1800_fcdjzz3s"]
["ZBT-CCTLight-D0106","lk"]
["ZBT-CCTLight-M3500107","lk"]
["eT093WRO","D5X84YU\u0000"]
["eTRV0100","Danfoss\u0000"]
["lumi.ctrl_neutral2","LUMI"]
["lumi.plug.aeu001","Aqara"]
["lumi.plug.mmeu01","LUMI"]
["lumi.remote.b1acn01","LUMI"]
["lumi.remote.b286acn01","LUMI"]
["lumi.remote.b28ac1","LUMI"]
["lumi.sensor_magnet.aq2","LUMI"]
["lumi.sensor_motion.aq2","LUMI"]
["lumi.switch.l0agl1","LUMI"]
["lumi.switch.l2aeu1","LUMI"]
["lumi.weather","LUMI"]
```

Mikei386, 2024-12-03:
> * Adapter model

Sonoff ZBDongle-E USB Zigbee 3.0 USB Dongle Plus,EFR32MG21 + CH9102F/CP2102N

> * Firmware version

Tried 7.4.4 and 8.0.2

> * For you, is it happening only during firmware updates, or during normal operation?

As far as I can tell only when running a firmware update. But I don't babysit it all the time so can't say it 100% certainly. 

>  * If during normal operation, how reasonably certain are you that your network contains or does not contain "spammy devices" (Tuya, power meters, mmWaves...)?

I have some IKEA sockets, that are metering power (5x INSPELNING) and another metering device  (1x Frient EMIZB-141)

> * List of device models on the network
    ["Frient", "EMIZB-141"]
    ["Custom devices (DiY)", "LYWSD03MMC"]
    ["Aqara", "WSDCGQ11LM"]
    ["IKEA", "E2013"]
    ["IKEA", "E2206"]
    ["IKEA", "E2202"]
    ["IKEA", "E2134"]
    ["IKEA", "E2204"]
    ["IKEA", "E2204"]
    ["IKEA", "E2206"]
    ["IKEA", "E2202"]
    ["IKEA", "E2013"]
    ["IKEA", "E2013"]
    ["Philips", "9290012573A"]
    ["Philips", "929002240401"]
    ["Philips", "8718696485880"]
    ["Custom devices (DiY)", "LYWSD03MMC"]
    ["IKEA", "E2206"]
    ["IKEA", "E2206"]
    ["Custom devices (DiY)", "LYWSD03MMC"]
    ["Custom devices (DiY)", "LYWSD03MMC"]

> Also, if anyone affected has updated to current dev branch (2.0.0), can you try tweaking the new OTA setting (`image_block_response_delay`) that throttles messages to see if it affects this bug at [...]

I have not tried any different firmware or Zigbee2MQTT Version.
I tried using ZHA inside HomeAssistant with the same result, crashing when trying to update a device firmware.
I had luck updating a device firmware by retrying it about 15 to 20 times...

antst, 2024-12-03:
this is overlap with me.

["Aqara", "WSDCGQ11LM"]
["Philips", "9290012573A"]

BTW,  Philips, potentially, might be candidate. This is 2nd gen lamps, if I recall right, which must be pretty old.
And it looks like 1st gen lamps (I had couple of them, 10+years old), seems to stopped to be compatible with modern zigbee. At least at some point this year they stopped to pair with setup I have. Neither zstack nor ember don't even notice their attempts to pair anymore. I didn't investigate further, but just use it as excuse to replace those couple of lamps with modern ones.

And we hardly can blame those aqara sensors for spamming. In my opinions, they rather have problem of to unfrequented updates :)

Nerivec, 2024-12-03:
@antst or @Mikei386 Can you isolate each of these in a test network, see if the issue appears when one of them is present, and you try to update another device? Either that, or powering them off completely (i.e. unplug, one model at a time), then trying to update in your current network.

Mikei386, 2024-12-03:
> @antst or @Mikei386 Can you isolate each of these in a test network, see if the issue appears when one of them is present, and you try to update another device? Either that, or powering them off completely (i.e. unplug, one model at a time), then trying to update in your current network.

I can try that, but I don't think it will do the trick since I already had an isolated network with only 2 IKEA Inspelning Plugs as members. Although it was an ZHA controlled network inside HomeAssistent which crashed too trying to update tose Plugs...

I will remove those two components and try again....

Mikei386, 2024-12-03:
Didn't  help...

>Error 2024-12-03 22:42:07zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
Error 2024-12-03 22:42:07zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
Error 2024-12-03 22:42:07zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
Error 2024-12-03 22:42:07zh:ember: Adapter fatal error: HOST_FATAL_ERROR
Error 2024-12-03 22:42:08z2m: Adapter disconnected, stopping

BTW: Only removing the Battery from the Aqara Sensor doesn't do anything. It still shows as "online" with the last values in Z2M, so I removed it completely....

antst, 2024-12-03:
Actually, I noticed crash without OTA just minutes ago.

antst, 2024-12-03:
```
[2024-12-03 22:57:35] info: 	z2m: Device '0xc49886000006e63c' joined
[2024-12-03 22:57:35] info: 	z2m: Starting interview of '0xc49886000006e63c'
[2024-12-03 22:57:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xc49886000006e63c","ieee_address":"0xc49886000006e63c"},"type":"device_joined"}'
[2024-12-03 22:57:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xc49886000006e63c","ieee_address":"0xc49886000006e63c","status":"started"},"type":"device_interview"}'
[2024-12-03 22:57:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xc49886000006e63c', payload '{"last_seen":"2024-12-03T22:57:35+01:00","update":{"installed_version":-1,"latest_version":-1}}'
[2024-12-03 22:57:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xc4988600000701dd', payload '{"last_seen":"2024-12-03T22:57:35+01:00","update":{"installed_version":-1,"latest_version":-1}}'
[2024-12-03 22:57:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/dining_wall_switch', payload '{"action":null,"device_temperature":20,"flip_indicator_light":null,"last_seen":"2024-12-03T22:57:35+01:00","led_disabled_night":null,"linkquality":192,"mode_switch":null,"operation_mode_left":"decoupled","operation_mode_right":"control_relay","power_outage_count":1,"power_outage_memory":null,"state":"ON","state_left":"ON","state_right":"OFF"}'
[2024-12-03 22:57:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/kitchen_wall_switch', payload '{"action":null,"device_temperature":22,"flip_indicator_light":null,"last_seen":"2024-12-03T22:57:35+01:00","led_disabled_night":null,"linkquality":112,"mode_switch":"quick_mode","operation_mode_left":"decoupled","operation_mode_right":"control_relay","power_outage_count":5,"power_outage_memory":null,"state":"ON","state_left":"ON","state_right":"ON"}'
[2024-12-03 22:57:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xc49886000006e63c', payload '{"last_seen":"2024-12-03T22:57:35+01:00","update":{"installed_version":-1,"latest_version":-1}}'
[2024-12-03 22:57:36] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xc49886000006e63c","ieee_address":"0xc49886000006e63c"},"type":"device_announce"}'
[2024-12-03 22:57:36] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xc4988600000701dd', payload '{"last_seen":"2024-12-03T22:57:36+01:00","update":{"installed_version":-1,"latest_version":-1}}'
[2024-12-03 22:57:36] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xc4988600000701dd","ieee_address":"0xc4988600000701dd"},"type":"device_announce"}'
[2024-12-03 22:57:43] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-12-03 22:57:43] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-12-03 22:57:43] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-12-03 22:57:43] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Total frames: RX=7869, TX=9269
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   DATA frames : RX=7836, TX=1401
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   DATA bytes  : RX=288495, TX=36309
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Retry frames: RX=27, TX=5
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   ACK frames  : RX=4, TX=7862
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Retry dupes     : RX=27
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-12-03 22:57:43] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=5
[2024-12-03 22:57:50] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=119 Len=10].
[2024-12-03 22:57:50] error: 	zh:controller: Interview failed for '0xc49886000006e63c with error 'Error: Interview failed because can not get active endpoints ('0xc49886000006e63c')'
[2024-12-03 22:57:50] error: 	z2m: Failed to interview '0xc49886000006e63c', device has not successfully been paired
[2024-12-03 22:57:50] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/light/0xc49886000006e63c/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/0xc49886000006e63c/set","device":{"identifiers":["zigbee2mqtt_0xc49886000006e63c"],"manufacturer":"Innr","model":"E27 bulb RGBW  [... cut ...] ject_id":"0xc49886000006e63c","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"schema":"json","state_topic":"zigbee2mqtt/0xc49886000006e63c","supported_color_modes":["xy","color_temp"],"unique_id":"0xc49886000006e63c_light_zigbee2mqtt"}'
[2024-12-03 22:57:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xc49886000006e63c","ieee_address":"0xc49886000006e63c","status":"failed"},"type":"device_interview"}'

```

Nerivec, 2024-12-04:
`ember` and 7.4.x/8.0.x have been tested with networks of over 300 devices (and with OTA running!), so it would seem this issue has a particular source, it shouldn't be just about "stress". It definitely looks like a packet (or several) being missed by the adapter.
The presence of IKEA and/or Aqara for users with this issue is worth noting though, since IKEA has some troubles with EmberZNet (I know, strange, since it uses EmberZNet too...), and Aqara has general troubles (not meant to be used without their hub, mostly not compliant)...
Also worth noting, for proxmox users, it has some general issues which probably do not help... we've had several weird behaviors linked to it recently (passthrough issues, NIC issues...).

Could you run a network scan (both channels, and existing networks) to check what kind of environment you have around you? https://github.com/Nerivec/ember-zli/wiki/Stack#channels-usage--rssi-11-26
Also, it would be good to have a debug log of Z2M along with a [sniff](https://github.com/Nerivec/ember-zli/wiki/Sniff) of a couple of crashes to see exactly what's happening, and if the exact same sequence of events happens every time.

EMAKASS, 2024-12-04:
I don’t understand how this works. I **updated** to **version 1.42.0**, and now there are no more reboots. It’s been running smoothly for **3 days**, whereas before it **restarted** at least once a day. If it continues to work without issues, could this be related to changes on the **Zigbee2MQTT** side?

Nerivec, 2024-12-04:
@EMAKASS No changes for `ember` in 1.42. Did you update anything else in your setup around the same time? Any other change? _Can be "as simple" as some kind of nearby interference (USB, 2.4GHz, power...) that's no longer having an effect..._

EMAKASS, 2024-12-04:
> @EMAKASSНикаких изменений `ember`в 1.42. Вы что-нибудь еще обновляли в своей настройке примерно в то же время? Есть ли еще какие-то изменения? _Может быть "так просто", как какие-то помехи поблизости (USB, 2,4 ГГц, питание...), которые больше не оказывают эффекта..._

No, it randomly rebooted again after lasting 3 days. So yeah, it’s not related to the Z2M firmware.

[121 more replies]

closed 2025-01-25 by Nerivec