Koenkk/zigbee2mqtt #23865: No device is connecting anymore after restart

Wuehli, 2024-09-05:
### What happened?

I just finished my move from deconz to z2m (on Pi4, Lastest PiOS lite x64, Sonoff USB Dongle Plus with lastest firmware Z-Stack_3.x.0_coordinator_20240710, 
Zigbee2MQTT-Version
    [1.40.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.0) commit: [13084d22](https://github.com/Koenkk/zigbee2mqtt/commit/13084d22)

Coordinator-Typ
    zStack3x0

Coordinator-Version
    20240710

Coordinator IEEE Adresse
    0x00124b0029dc1135

Frontend-Version
    0.7.4

Zigbee Herdsman Konverter Version
    20.8.4

Zigbee Herdsman Version
    0.57.1

After a shutdown (to change the case) no device is connected anymore, all battery powered devices show online, all mains powered offline, last seen is about the time i shut down for all 31 devices currently connected.

There are no errors in the Log except for each mains powered device:
z2m: Failed to ping 'ZB-ODPL-10' (attempt 1/1, ZCL command 0xf0d1b8be240836db/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))

### What did you expect to happen?

it still works after reboot 

### How to reproduce it (minimal and precise)

shut down the pi

### Zigbee2MQTT version

1.40

### Adapter firmware version

20240710

### Adapter

Sonoff USB Dongle Plus

### Setup

Raspberry Pi 4, PiOs lite x64

Wuehli, 2024-09-05:
after re-pairing a device this device is working again

nicolasvila, 2024-09-06:
Did you power of completely the RPi4 to make it work again (I mean unplug the power adapter) 
I have a similar issue with the same firmware and a different stick and Zigbee2MQTT 1.38.1

Wuehli, 2024-09-06:
To generate the initial error, I first shut down the Pi and then disconnected the power supply.
I also tried everything from restarting only z2m, reboot and unplug the power to make it work again, but nothing helped.

After that i re-paired some devices to get the basics back to work.
I then also tried again to restarting only z2m, reboot and unplug the power to see what is happening.

The current status is somehow different, most of the devices i haven't re-paired so far are still shown als offline or last seen when i did the initial shutdown, a few (1 or 2) seem to work again without re-pairing. But also at least one device that has been re-paired is again not working (but other identical types of this devices do).

UberDudePL, 2024-09-19:
I'm experiencing the same issue with multiple devices showing the same error message. The strange thing is, I didn't change anything. Yesterday, all my devices went offline unexpectedly. I remotely closed a cover, took a nap, and when I woke up, everything had stopped working. I checked the logs but found no indication of what caused it—there was no power outage or anything unusual.

Some devices eventually came back online, and I managed to repair a few others, leaving only 18 offline. However, today, most of the devices are offline again, including the ones I fixed yesterday. The only potential factor I can think of is that I powered down my server for a couple of hours the day before this all started.

closed 2024-12-19 by github-actions[bot]