Koenkk/zigbee2mqtt #23869: Firmware 20240710 repair Devices every day

Barneybaer84, 2024-09-06:
### What happened?

Since the firmware Update to my Sonsoff zigbee 3.0-P , I have devices that are offline every day and need to be repaired. This is frustrating 

### Zigbee2MQTT version

1.40.0 

### Adapter firmware version

20240710

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus

### Setup

X86 Home Assistant

rursache, 2024-09-07:
same issue here, i downgraded the adapter to `20221226` but it still crashed. so i also downgraded `zigbee2mqtt` to `1.39.1` and i plan to go as low as `1.38.0`

my zigbee network was flawless for 2 years until these 2-3 days when both `zigbee2mqtt` and adapter fw got updated

Koenkk, 2024-09-07:
Could you provide the debug log from starting z2m until the device drops?

jymorel, 2024-09-07:
Same issue here after firmware and z2m updates. Especially sonoff temp and humidity sensor (snzb-02), but not only. Re-pairing is not enough. Need to delete the device and re-pair again

Barneybaer84, 2024-09-07:
I have update the firmware at 05.09. but my next log ist at 06.09.
In the log you can ignore Alarm Sirene, Alarm Sirene defekt and Nachttisch Schatz. All other devices are always online. 
[06.09.24_log.zip](https://github.com/user-attachments/files/16919262/06.09.24_log.zip)

rursache, 2024-09-07:
for me it seems like restarting the zigbee2mqtt container fixes it but i wouldn't like to have a cronjob to restart the container every hour

Koenkk, 2024-09-09:
@Barneybaer84 does the issue also happen when you keep the devices really close to the coordinator? Z2M tries to reach it but fails, could indicate an issue with one of the routers

@rursache does the issue still happen with the availability feature disabled? Did the 20221226 firmware work?

rursache, 2024-09-09:
@Koenkk I always had "Availability" enabled as I want to see that my devices are available. i can try with it off next week but I don't want to keep it disabled. the latest working fw was 'CC2652RB_coordinator_20221226' (almost 2 years, butter smooth and stable) but now downgrading to it does not fix it!

Barneybaer84, 2024-09-10:
After 5 days with device drops and update z2m to 1.40.1, my Zigbee network works perfecly again.

Skeletorjus, 2024-09-10:
I'm facing the same, but I'm not so sure that 20240710 is causing this - I had the same with 20230507 (https://github.com/Koenkk/zigbee2mqtt/issues/23329#issuecomment-2227268818).
Updated to 20240410 five days ago and am still having the same thing happening with the network going offline. The way I usually notice it is because some of my bulbs (Namrons, to be exact) starts panicking and goes into some flashing disco mode 😆 

I can't remember when it started, but I'm tempted to say that my issues started around 1.38.0.

Log from a couple of minutes before the crash and the crash itself, pretty much the same as https://github.com/Koenkk/zigbee2mqtt/issues/23869#issuecomment-2336713323.
[z2m_error.txt](https://github.com/user-attachments/files/16949781/z2m_error.txt)

Zigbee2MQTT 1.40.1 commit: 403d3c0
Sonoff Plus P
zStack3x0
20240710.

jymorel, 2024-09-17:
everything ok for a few days after re-pairing several devices

rursache, 2024-09-17:
i still have to restart the zigbee2mqtt docker container every hour otherwise the coordinator and all the routers go offline, bringing everything down.

rursache, 2024-09-19:
as a walkaround and fed up with the sloppy cronjob, i made this HomeAssistant automation to restart the zigbee2mqtt docker container when my philips hue light goes offline (`unavailable` in HASS):

```
alias: Fix Zigbee2MQTT
description: ""
trigger:
  - platform: state
    entity_id:
      - light.living_room_philips_hue_color
    from: null
    to: unavailable
    for:
      hours: 0
      minutes: 0
      seconds: 5
condition: []
action:
  - action: shell_command.restart_zigbee2mqtt
    metadata: {}
    data: {}
mode: single
```

please note that you need to create an entry for `shell_command.restart_zigbee2mqtt` in your HASS `configuration.yaml` file like this:
```
shell_command:
  restart_zigbee2mqtt: >
    'nohup curl -X POST URL $1 > /dev/null 2>&1 &' 
```

make sure to replace `URL` with your portainer or whatever else webhook you have and `light.living_room_philips_hue_color` with your zigbee entity from HASS

now when the light goes offline (the simpler way of detecting when the entire zigbee network is down) it will restart the z2m container bringing everything up and running in under 15 seconds

i really can't wait for a proper fix tho!

Skeletorjus, 2024-09-22:
Very likely a red herring, but do any of you utilize bindings?
My network has been crashing on and off for a while (as described in https://github.com/Koenkk/zigbee2mqtt/issues/23869#issuecomment-2341846601), but it hasn't been too bad lately.

I have two [IKEA Parasoll](https://www.zigbee2mqtt.io/devices/E2013.html)-devices that both have been bound to their own bulbs ([Namron 3802952](https://www.zigbee2mqtt.io/devices/3802962.html)). The contact sensors have been out of use for a while due to empty batteries. Today I replaced the batteries, and as soon as I did a couple of tests to ensure that the bidings to the bulbs worked, my whole network went down. Had to replug the Sonoff and restart Zigbee2MQTT. Did this multiple times.

I have removed the bindings, and for the time being the network seems stable.

rursache, 2024-09-22:
being super frustrated with the lack of support or work being done to fix this, i bought a new `ZigStar UZG-01` (`CC2652P7 `) which arrived with FW `20230507`. i switched the old `slaesh CC2652RB` with the `UZG-01` and my zigbee network has been stable ever since. 72h so far, had crashes every 20min-8h. so far 0 drops or crashes. will flash the `slaesh` as a router and use it like that.

i think the new firmware ruins the coordinator somehow but it's just my guess. i tried flashing the `slaesh CC2652RB` with each firmware starting with 2022 until the latest, none fixed it. a new device did. well 🤷🏻‍♂️

@Skeletorjus i don't use bindings, don't think it's related

Koenkk, 2024-09-22:
Could you see if the `99240914` firmware fixes it? If yes, then try the next fws (e.g. `99240915`) until the problem reappears. [fws.zip](https://github.com/user-attachments/files/17091049/fws.zip)

rursache, 2024-09-22:
i do not own a sonoff device, just two slash-es. 

i'm also not interested in fiddling again after i finally have a functioning network

thogens, 2024-09-23:
I seem to have the same issue, but need to investigate further.
So far the only way for me to fix the crash was to restart HA, but that takes 5mins at least.
Next time I'll check if a simple restart of the Zigbee2Mqtt Addon will do the trick.
If yes -> @rursache : I find your script promising, but I don't know how to restart the Addon in my case. I'm not running it on a dedicated portainer, but as std. HA Addon... Any idea?

rursache, 2024-09-23:
@thogens i'm sorry, i run homeassistant, zigbee2mqtt, mosquitto and all my services in docker. not sure if you could reload an HASS addon if you have HASS OS installed

Barneybaer84, 2024-09-24:
I don't flash the test fw but i have switch the USB 3.0 to 2.0 with an USB 2.0 cable and i have no device disconnect since two days.
I hope it stays that way.

Barneybaer84, 2024-09-26:
Day 4 after switch to USB 2.0 2 devices are offline :( I will test the fws.zip

Luca1996O, 2024-09-28:
Same issue here, with ZBDongle-P latest firmware version and ZigBee2MQTT latest version (1.40.1).

Barneybaer84, 2024-09-28:
With test fw 99240914 same problem. I have downgraded to 20230507, re-pair same devices and now it works for 4 days. No device goes offline anymore.

stewepylon, 2024-11-11:
In my environment, latency was caused by the automatic update checks. Many BTicino devices are reporting hundreds of messages due to firmware version mismatches, which seems to be slowing everything down.

fsedarkalex, 2024-11-11:
I have just disabled the OTA check to test if this fixes the device "drops"

thogens, 2024-11-15:
Yesterday again the whole network crashed and I know that I disabled the automatic OTA just some hours earlier. 
Currently I have some sort auf auto-detection mechanism in place that automatically reboots HA in such a case, where nothing works anymore in the Zigbee net.
Downside is that this takes 15mins. 
But after rebooting HA, everything works fine again...

Roemer, 2025-01-20:
I updated my SLZB-06 coordinator to 20240710 and it got totally unreliable. Devices were suddenly removed, notifications on button presses were missed and such things. I then downgraded to 20221226 and everything is rock solid now.

thogens, 2025-01-20:
Same here, 20221226 is much better! Only thing I noticed is: don’t turn on „last seen“. Once I did that, several devices were all of a sudden „gone“. They weren’t even listed in the devices table any more. Interestingly: re pairing worked and device name was working again.

thogens, 2025-01-20:
Ok, I was just "assuming" something... "Last seen" was the last thing I enabled just recently and all of the sudden, the problems start kicking in, even with the old firmware (after it was working great for 3 weeks or so).
I now disabled "last seen" and will report, if problems are coming back again.
It could very well be that this was just coincidence and that there is a different underlying problem. (some buffer that gets deleted when flashing firmware and fills up over time until it "crashes" or so.

jafar-atili, 2025-01-21:
I had the same issue, Z2M got upgraded and my network crashed every other day (offline devices and errors in Z2M UI)
Moving to ZHA fixed this, the network is solid for 5 months now.

I'll keep an eye on this one so I know when to switch back to Z2M.

Luca1996O, 2025-01-22:
> I had the same issue, Z2M got upgraded and my network crashed every other day (offline devices and errors in Z2M UI) Moving to ZHA fixed this, the network is solid for 5 months now.
> 
> I'll keep an eye on this one so I know when to switch back to Z2M.

Did you have to re-pair each device manually, or did you find a way to migrate them?

jafar-atili, 2025-01-22:
> Did you have to re-pair each device manually, or did you find a way to migrate them?

Yep

thogens, 2025-01-22:
Yep == re-pair or yep == migrate ?

jafar-atili, 2025-01-22:
Re-paired everything

closed 2025-03-31 by github-actions[bot]