Koenkk/zigbee2mqtt #23870: AM43-0.45_40-ES-EB Device uses batteries, but shows up as powered device

rsacherer, 2024-09-06:
### What happened?

Using two of these devices to test them out:
https://www.zigbee2mqtt.io/devices/AM43-0.45_40-ES-EB.html

Working good except that they show up as electricity powered, while they are actually battery powered. They seem to have big batteries as they work quite some time until I have to charge, but they come also with a solar charger, so I would like to see if the solar charging is actuall working fine, but as the battery and level are not shown/detected, I have no idea about the charge state.

### What did you expect to happen?

see battery and charge states

### How to reproduce it (minimal and precise)

Just pairing should show it incorrectly. I am available and open for testing etc.

### Zigbee2MQTT version

 1.37.1 commit: ea39d86

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

docker setup for home assistant

teixemf, 2024-09-06:
Same issue here.
Also, the response to zigbee commands is very slow. It takes around 2 to 4 seconds to react to Open/Stop/Close commands.
All my other zigbee devices react fast. @rsacherer does your device reatcs slow to like mine?

pbartek20, 2024-09-06:
After I saw this thread I checked and noticed that for me are also showing as grid powered not battery.
If I remember correctly in previous Z2M versions they were showing as battery powered but there was no battery % level showed. Not sure if it changed in Z2M 1.40 or before.

@teixemf for me response time is ok, after clicking not more than 0,5 sec when it starting to close/open.

rsacherer, 2024-09-06:
Hi, I always thought the blinds take a bit to react, a couple of seconds sounds right, moving the I have time to think if the commands have been sent and then it starts reacting. 

That actually does not even register as an annoyance, I have opening and closing mostly automated. But the battery % level not showing bothers me quite a bit.

rsacherer, 2024-09-06:
There seem to be a couple of different devices with a very similar look (maybe all are the same?) which seem to export (at least on the devices site) battery state:

https://www.zigbee2mqtt.io/devices/ZNJLBL01LM.html
https://www.zigbee2mqtt.io/devices/TS0301.html

hossamnagy, 2024-12-08:
Same issue here, also  per the manual , the solar panel has a luminance sensor, that can bu used to adjust the blind .  I couldn’t find that reported via z2m. Appreciate if this can be supported .

to0freak, 2025-01-08:
Same issue with battery level, react speed and no light sensor

dovy6, 2025-02-02:
Is it possible to change these to report battery level, or do they just not expose that? Is there any information I can provide that could help determine this information?

dovy6, 2025-04-04:
Keeping this open

ceedeebee907, 2025-04-20:
would be keen to know if we can see the illuminance sensor too

closed 2026-03-16 by github-actions[bot]

andrei-lazarov, 2026-04-18:
I checked `_TZE200_zah67ekd` in Tuya app / dev portal. It does not report battery state/percentage. 

[image]

[image]

Data: [_TZE200_zah67ekd.md](https://github.com/user-attachments/files/26852738/_TZE200_zah67ekd.md)