Koenkk/zigbee2mqtt #23871: Tuya Plug _TZ3000_o1jzcxou Energy Monitoring not working

roteRakete66, 2024-09-06:
### What happened?

I have a couple of those [Bseed ZigBee Socket](https://www.bseed.com/products/bseed-zigbee-eu-wall-sockets-power-outlets-with-energy-monitoring-kids-protection?variant=42657798291611), which became device support from this request https://github.com/Koenkk/zigbee2mqtt/issues/10675 and it doesn't work entirely as expected.
All energy measurements show 'Null'.
I have already performed an OTA update on one of them and adjusted the rms in the haElectricalMeasurement, then each value was 0.

Here is the database entry of the updated one:
```
{"id":23,"type":"Router","ieeeAddr":"0xa4c138f2bdd7fd31","nwkAddr":60964,"manufId":4417,
"manufName":"_TZ3000_o1jzcxou","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],
"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],
"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"S^k.\u0012","65506":54,"65508":0,"65534":0,
"modelId":"TS011F","manufacturerName":"_TZ3000_o1jzcxou","powerSource":1,"zclVersion":3,"appVersion":192,
"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,
"acCurrentMultiplier":1,"rmsVoltage":0,"activePower":0,"rmsCurrent":0}},"seMetering":{"attributes":{"divisor":100,
"multiplier":1,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,
"offWaitTime":0,"tuyaBacklightMode":0,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},
"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,
"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x048727fffe1e6d6a","endpointID":1},
{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x048727fffe1e6d6a","endpointID":1},{"cluster":1794,
"type":"endpoint","deviceIeeeAddress":"0x048727fffe1e6d6a","endpointID":1}],"configuredReportings":[{"cluster":2820,
"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,
"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,
"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},
"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],
"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,
"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1725619247142}
```

And here one of the other:
```
{"id":22,"type":"Router","ieeeAddr":"0xa4c138697aea3afe","nwkAddr":17695,"manufId":4417,
"manufName":"_TZ3000_o1jzcxou","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],
"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],
"clusters":{"genBasic":{"attributes":{"65503":"��k.\u0013�\u0003\u0000\u0000\u0012","65506":48,"65508":0,"65534":0,
"modelId":"TS011F","manufacturerName":"_TZ3000_o1jzcxou","powerSource":1,"zclVersion":3,"appVersion":67,
"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,
"switchType":0}},"genOnOff":{"attributes":{"onOff":1,"tuyaBacklightMode":0}}},"binds":[{"cluster":6,"type":"endpoint",
"deviceIeeeAddress":"0x048727fffe1e6d6a","endpointID":1},{"cluster":2820,"type":"endpoint",
"deviceIeeeAddress":"0x048727fffe1e6d6a","endpointID":1},{"cluster":1794,"type":"endpoint",
"deviceIeeeAddress":"0x048727fffe1e6d6a","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,
"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],
"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,
"meta":{},"lastSeen":1725619295323}
```

If someone could help me, that would be great.
If you need more information, feel free to ask.

### Zigbee2MQTT version

1.40.0-dev commit: 6dade8b

### Adapter firmware version

7.4.1 [GA]

### Adapter

SkyConnect

### Setup

Add-on on Home Assistant OS on RPi5

OlegPitsenko, 2024-09-07:
Energy monitoring also does not work:
_TZ3000_cphmq0q7

roteRakete66, 2024-09-15:
It would be great if someone could take a look at it.
Actually, I only bought these because of the monitoring function.

Thanks

Koenkk, 2024-09-16:
Could you provide the debug log when pressing the yellow configure button in the frontend?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

roteRakete66, 2024-09-17:
Sure, I hope you meant this:

the updated one:
`
z2m: Failed to configure 'Teststecker', attempt 1 (Error: ZCL command 0xa4c138f2bdd7fd31/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":60964,"apsFrame":{"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":58},"zclSequence":136,"commandIdentifier":7} timed out after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:198:23) at EmberOneWaitress.startWaitingFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:222:50) at /app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2526:55 at processTicksAndRejections (node:internal/process/task_queues:95:5) at runNextTicks (node:internal/process/task_queues:64:3) at processImmediate (node:internal/timers:447:9) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:911:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:648:9))
`

and one of the other:
`
z2m: Failed to configure 'Büro Router', attempt 4 (Error: Bind 0xa4c138697aea3afe/1 haElectricalMeasurement from '0x048727fffe1e6d6a/1' failed ([ZDO] Failed response for '17695' cluster '32801' Status 'INVALID_EP'.) at EmberOneWaitress.resolveZDO (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:139:35) at EmberAdapter.onZDOResponse (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:556:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspIncomingMessageHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5249:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:782:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21))
`

Oh, and by the way, thanks for reply.

Koenkk, 2024-09-19:
For the first one, could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/3eeb90dd20a3bf6f09e76880abfc9c74): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if configure works now.

roteRakete66, 2024-09-19:
First of all, thank you for taking a look at this.
I wasn’t sure if the external converter loaded correctly because there was no `CUSTOM` in the description, so I added the following entry to `whiteLabel`:
`tuya.whitelabel('Bseed', 'TS011F_plug', 'Smart plug (with power monitoring)', ['_TZ3000_o1jzcxou'])`
The vendor is now Bseed, so it seems to have loaded. Configure works without errors now.
However, all measurements are still 0.

Koenkk, 2024-09-20:
Do you get a value when manually reading it?

[image]

roteRakete66, 2024-09-21:
Yes, when I read it manually, I also get only 0 everywhere.

Koenkk, 2024-09-21:
Then I guess your plug is broken

roteRakete66, 2024-10-17:
Finally, I hear that someone else has the same problem. Now I'm not alone anymore, thank you for that.
I wanted to say that it's not just one device that's broken but around 20, ... I don't think so.

But well, would it maybe help if we asked the vendor for some info?

Or should we take a closer look at the non-updated devices I mentioned above and posted info about?

crazyserver, 2024-10-17:
I think there could be 2 different problems:
1. There is some bug on Z2m that make the energy information unavailable on Z2m... 
2. The device is detecting monitoring and it does not have any. So Z2M should not show it.

Unfortunatelly I don't have any other ZB system to try the device (ie, tuya smart life).

And I'm sorry to say it may be the last one, because I've heard in many other forums the vendor sent the one without Energy monitoring...

Checking my order on aliexpress it does not say anything about monitor (but I'm sure it was written there before) so maybe the seller changed the description once they realised...

roteRakete66, 2024-10-17:
> Unfortunatelly I don't have any other ZB system to try the device (ie, tuya smart life).

That’s a good idea, I had the same one, but I also don’t have a gateway that I could test with the Tuya Smart Life app. Maybe I’ll order one.
> Also I don't have any OTA option in Z2M

It’s strange, mine all have that.
> I think there could be 2 different problems:
> 1. There is some bug on Z2m that make the energy information unavailable on Z2m...
> 2. The device is detecting monitoring and it does not have any. So Z2M should not show it.

Yes, you’re right, I fear the second possibility might be true, which would be bad. Then I would have to buy new ones. Let's hope not.

rakalex, 2024-10-28:
I have this one https://www.aliexpress.com/item/1005005646484680.html?spm=a2g0o.order_list.order_list_main.71.6db81802STfjTW. 
[image]

But as you can see on ali, this plug doesn't have an energy monitor. All else works fine, no issues.

closed 2025-01-05 by github-actions[bot]