Koenkk/zigbee2mqtt #23874: A lot of messages: "EventBus error 'OnEvent/deviceMessage': ZCL command ..... timed out after 10000ms"

Danieldiazi, 2024-09-06:
### What happened?

Hi,

With this device:

- Model: ZWT199/ZWT100-BH
- Zigbee Model TS0601
- Manufactured: _TZE204_lzriup1j 

[image]

I get a lot of "EventBus error" messages every second.

Thanks in advance for your support!

### What did you expect to happen?

Don't have any of this messages

### How to reproduce it (minimal and precise)

Only to pair this device in zigbee2mqtt and you will see those messages.

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20240510

### Adapter

SLZB-06M

### Setup

Docker container

### Debug log

`error 2024-09-06 14:16:41z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xXXXXXXXXX/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":30621,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":237},"zclSequence":40,"commandIdentifier":11} timed out after 10000ms)
error 2024-09-06 14:16:53z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xXXXXXXXXX/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":30621,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":125},"zclSequence":98,"commandIdentifier":11} timed out after 10000ms)
error 2024-09-06 14:17:24z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xXXXXXXXXX/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":30621,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":139},"zclSequence":177,"commandIdentifier":11} timed out after 10000ms)
error 2024-09-06 14:17:34z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xXXXXXXXXX/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":30621,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":192},"zclSequence":188,"commandIdentifier":11} timed out after 10000ms)
error 2024-09-06 14:17:55z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xXXXXXXXXX/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":30621,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":135},"zclSequence":128,"commandIdentifier":11} timed out after 10000ms)
error 2024-09-06 14:18:51z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xXXXXXXXXX/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":30621,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":47},"zclSequence":133,"commandIdentifier":11} timed out after 10000ms)
error 2024-09-06 14:19:11z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xXXXXXXXXX/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":30621,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":223},"zclSequence":61,"commandIdentifier":11} timed out after 10000ms)`

arkadi222, 2024-09-15:
same here.  z2m 1.40.1 on docker unraid, skyconnect, emberznet

arkadi222, 2024-09-15:
after comment 'external_converters' in configuration.yaml and restart z2m there are no that messages in log

Danieldiazi, 2024-09-15:
> after comment 'external_converters' in configuration.yaml and restart z2m there are no that messages in log

In my case, I don't use external_converters option.

Koenkk, 2024-09-16:
It looks like the device is not reachable somehow, does it still happen when re-pairing it close to the coordinator?

Danieldiazi, 2024-09-16:
There are less than 1.5 meters between them.
They are in the same room. 1.5 meters is enough or maybe  should I connect them closer together?

Danieldiazi, 2024-09-25:
> It looks like the device is not reachable somehow, does it still happen when re-pairing it close to the coordinator?

Hi, I've tried to repair the device near the coordinator and the problem is the same. Any advice?

mrtmsgh, 2024-10-01:
Hi, I'm having the same problem with a similar device:

Model: ZWT198/ZWT100-BH
ZigBee Model: TS0601
Device:  _TZE200_viy9ihs7

ZigBee adapter: SLZB-06M
ZigBee adapter firmware version: 20240510
Coordinator type: EmberZNet
Coordinator revision: 7.4.1 [GA]
ZigBee2MQTT version: 1.40.1
zigbee-herdsman-converters version: 20.12.1
zigbee-herdsman version: 0.57.3

After successfully pairing it, every ten seconds or so, I'm seeing a bunch of messages like these:

[20241001:174759] error: 	z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xb43a31fffe3d296a/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":143})
[20241001:174809] error: 	z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xb43a31fffe3d296a/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":229})
[20241001:174835] error: 	z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xb43a31fffe3d296a/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":73})

Thanks

Danieldiazi, 2024-10-05:
In order to add more information, I can confirm the problem exists with the newest 1.40.2 version.

Koenkk, 2024-10-06:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/4b4c582f8a01fd71108acce6a5e2db15): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

domome, 2024-10-07:
Hello,
It seems to be working in my case, there are no errors in the Z2M and HA logs.
Let's see what others say
Thanks

Koenkk, 2024-10-07:
@domome double check that all the features are still working

domome, 2024-10-07:
> @domome double check that all the features are still working

Yes, everything is working properly.

Vangreen, 2024-10-07:
Same here, now everything works ok, no more error logs. Thanks!

Danieldiazi, 2024-10-07:
Same here too! It works great!! Any erros in the log file! 

I wonder if it is possible to know what the problem was...

Thanks for your support @Koenkk !!

mrtmsgh, 2024-10-09:
The issue with my ZWT198 thermostat is all sorted out now. There are no more error messages in the logs, and it’s working perfectly. Huge thanks for all the hard work and support.

domome, 2024-10-09:
> @domome double check that all the features are still working

Hello
Doing an automation in HA I just checked that it seems that the Auto and Manual function are interchanged. 
When Manual mode is set, the clock symbol appears on the thermostat (which is the automatic mode) and when Auto is set, a hand appears on the thermostat (which is the manual mode). 
Can anyone else do this check?
Thank you.

arkadi222, 2024-10-09:
@domome you can add comment here :)
https://github.com/Koenkk/zigbee2mqtt/issues/23998
please also confirm your manufacturer number : _TZE204_lzriup1j

arkadi222, 2024-10-09:
temporary you can do what I did :)
https://github.com/Koenkk/zigbee2mqtt/issues/23998#issuecomment-2399162938

domome, 2024-10-09:
> @domome you can add comment here :) #23998 please also confirm your manufacturer number : _TZE204_lzriup1j

It is indeed that model.

I was looking at doing a test with something like: 
[4, 'preset',
     if (device.manufacturerName === '_TZE200_viy9ihs7') **or (device.manufacturerName === '_TZE204_lzriup1j')**

Danieldiazi, 2024-10-09:
> Auto and Manual function are interchanged.

Same here

mrtmsgh, 2024-10-09:
The TZE200_viy9ihs7 has the same issue with the presets. I just discovered this after reading the latest comments here.

There’s a comment in [#21353](https://github.com/Koenkk/zigbee2mqtt/issues/21353#issuecomment-1938328429) that explains the order of presets for both models, and it seems that in the code, both are reversed.

WolforOwl, 2024-10-26:
Hi All, i have similar issue but with 
Zigbee Model TS0601
Zigbee Manufacturer _TZE200_g9a3awaj
Model [ZWT07]

Would this fix work on this model by adding external converter and changing model line?

bachmarc, 2024-11-18:
HI, 
I have here a hand full of Sensors looking the same but identified as 
Zigbee Model TS0601
Zigbee Manufactur _TZE204_upagmta9 

which show same behavior and are very coupling only after several retries. Usually they look connected but send no metrics or parts of it
in ZIGBEE logs the error "EventBus error 'OnEvent/deviceMessage': ZCL command ..... timed out after 10000ms" is a frequent guest

Maybe the debug log helps from coupling? 

`info 2024-11-18 18:00:07z2m: Accepting joining not in blocklist device '0xa4c13837935c5b8c'
info 2024-11-18 18:00:07z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13837935c5b8c', payload '{"last_seen":"2024-11-18T18:00:08+01:00","linkquality":236}'
info 2024-11-18 18:00:07zh:controller: Interview for '0xa4c13837935c5b8c' started
info 2024-11-18 18:00:07z2m: Device '0xa4c13837935c5b8c' joined
info 2024-11-18 18:00:07z2m: Starting interview of '0xa4c13837935c5b8c'
info 2024-11-18 18:00:07z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13837935c5b8c","ieee_address":"0xa4c13837935c5b8c"},"type":"device_joined"}'
info 2024-11-18 18:00:07z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13837935c5b8c","ieee_address":"0xa4c13837935c5b8c","status":"started"},"type":"device_interview"}'
info 2024-11-18 18:00:07z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13837935c5b8c', payload '{"last_seen":"2024-11-18T18:00:08+01:00","linkquality":236}'
info 2024-11-18 18:00:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13837935c5b8c","ieee_address":"0xa4c13837935c5b8c"},"type":"device_announce"}'
error 2024-11-18 18:00:18z2m: EventBus error 'OnEvent/deviceAnnounce': ZCL command 0xa4c13837935c5b8c/1 manuSpecificTuya.dataQuery({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":30062,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":240},"zclSequence":49,"commandIdentifier":11} timed out after 10000ms)
info 2024-11-18 18:00:41z2m: Accepting joining not in blocklist device '0xa4c13837935c5b8c'
info 2024-11-18 18:00:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13837935c5b8c', payload '{"last_seen":"2024-11-18T18:00:42+01:00","linkquality":236}'
info 2024-11-18 18:00:41z2m: Accepting joining not in blocklist device '0xa4c13837935c5b8c'
info 2024-11-18 18:00:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13837935c5b8c', payload '{"last_seen":"2024-11-18T18:00:42+01:00","linkquality":236}'
info 2024-11-18 18:00:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13837935c5b8c', payload '{"last_seen":"2024-11-18T18:00:42+01:00","linkquality":236}'
info 2024-11-18 18:00:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13837935c5b8c","ieee_address":"0xa4c13837935c5b8c"},"type":"device_announce"}'
info 2024-11-18 18:00:41zh:controller: Succesfully interviewed '0xa4c13837935c5b8c'
info 2024-11-18 18:00:41z2m: Successfully interviewed '0xa4c13837935c5b8c', device has successfully been paired
info 2024-11-18 18:00:41z2m: Device '0xa4c13837935c5b8c' is supported, identified as: Tuya Temperature and humidity sensor (ZTH05)
info 2024-11-18 18:00:41z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c13837935c5b8c/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c13837935c5b8c/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c13837935c5b8c"],"manufacturer":"Tuya","model":"Temperature and humidity sensor (ZTH05)","name":"0xa4c13837935c5b8c","via_device":"zigbee2mqtt_bridge_0x8c65a3fffe5485e4"},"device_class":"temperature","enabled_by_default":true,"object_id":"0xa4c13837935c5b8c_temperature","origin":{"name":"Zigbee2MQTT","sw":"1.41.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c13837935c5b8c","unique_id":"0xa4c13837935c5b8c_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
info 2024-11-18 18:00:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature and humidity sensor","exposes":[{"access":1,"description":"Measured temperature value","label":"Temperature","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","label":"Humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"State of the battery","label":"Battery state","name":"battery_state","property":"battery_state","type":"enum","values":["low","medium","high"]},{"access":3,"description":"Temperature unit","label":"Temperature unit","name":"temperature_unit","property":"temperature_unit","type":"enum","values":["celsius","fahrenheit"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZTH05","options":[{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report [... cut ...] dity sensor (ZTH05)","name":"0xa4c13837935c5b8c","via_device":"zigbee2mqtt_bridge_0x8c65a3fffe5485e4"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","name":"Last seen","object_id":"0xa4c13837935c5b8c_last_seen","origin":{"name":"Zigbee2MQTT","sw":"1.41.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0xa4c13837935c5b8c","unique_id":"0xa4c13837935c5b8c_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'`

AppleFlapp, 2025-01-02:
Same here with ZTH01 temperature sensor.

Zigbee Model: TS0601
Zigbee Manufacturer: _TZE204_yjjdcqsq

Zigbee2MQTT version 1.42.0

Message keeps popping up even after the devices is removed from the Z2M configuration.

2025-01-02 19:17:30z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xa4c138d6e76a7d2f/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 29665 - 1 - 44 - 61184 - 11 after 10000ms)
error 2025-01-02 19:17:53z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xa4c138d6e76a7d2f/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 29665 - 1 - 45 - 61184 - 11 after 10000ms)
error 2025-01-02 19:18:17z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0xa4c138d6e76a7d2f/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 29665 - 1 - 46 - 61184 - 11 after 10000ms)

closed 2025-03-13 by github-actions[bot]

saper-2, 2025-08-12:
I started to get those errors too 
```EventBus error 'OnEvent/deviceAnnounce': ZCL 0x...../1 command manuSpecificTuya.dataQuery({}, ....```

Check batteries... 
One of my batteries leaked inside while the clock/sensor was showing last bar on battery gauge - I thought batts are still ok...  
But when I pressed button the screen started to dimming/blinking and showed garbage....

ansantos, 2025-09-15:
I have this same error with this device: https://www.zigbee2mqtt.io/devices/PJ-1203A.html#tuya-pj-1203a

HAOS 16.2 running on RPi 4.
Zigbee2MQTT 2.6.1-1
Sonoff Zigbee dongle - E with Ember 7.4.5.0

zhc:tuya: Failed respond to version response '0xa4c1384e31082f59' (Error: ZCL command 0xa4c1384e31082f59/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":29572,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":179},"zclSequence":133,"commandIdentifier":11} timed out after 10000ms))