Koenkk/zigbee2mqtt #23876: Error when clicking the button - Loginovo ZG-101ZL

rursache, 2024-09-06:
### What happened?

**Zigbee Model** TS004F
**Zigbee Manufacturer** _TZ3000_ja5osu5g
**Manufacturer** Loginovo
**Model** ZG-101ZL

Every time I press the button, this error comes up. It works fine and it triggers my automations in HASS but it logs this error:
```
z2m: Failed to configure 'BUTTON_NAME', attempt 1 (Error: ZCL command 0xa4c138d47e3ef5e3/1 genBasic.read([4,0,1,5,7,65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:916:26) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:443:29) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/moes.ts:548:13) at Configure.configure (/app/lib/extension/configure.ts:125:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

### Zigbee2MQTT version

1.40.0 commit: 6fa1dab

### Adapter firmware version

20240710 and 20221226

### Adapter

Slaesh's CC2652RB

### Setup

Docker

guillaumematheron, 2024-09-12:
I solved this by pressing the reset button next to the battery. Zigbee2mqtt indicated the device left the network, and after re-joining everything worked as before.

kartone, 2024-10-09:
Same issue here. Actions and automations seem to work without issue but it's annoying seeing the error messages every now and then.

numminorih-git, 2024-10-22:
+1 all is work but time to time (without button press) in log I see error: 	z2m: Failed to configure 'click_name', attempt 1
Zigbee2MQTT version [1.40.2]
Coordinator type zStack3x0
Coordinator revision 20240716

rursache, 2024-12-22:
still not fixed.

rursache, 2025-02-21:
still not fixed.

numminorih-git, 2025-02-21:
version: 2.1.1-1 not fixed. Same issues.

carlosnh96, 2025-03-20:
I have the same problem and it has not been solved with the latest versions.

rursache, 2025-05-20:
not fixed

closed 2025-07-27 by github-actions[bot]