Koenkk/zigbee2mqtt #23880: Aqara ceiling light T1M (CL-L02D) - Color (X/Y) & Color (HS) not working

joe-peet, 2024-09-07:
### What happened?

I got the Aqara Ceiling light T1M (CL-L02D - lumi.light.acn032) yesterday and want to change the color of the second color LED stripe. Unfortunately, nothing happens with Color (X/Y) or Color (HS) when I click on a color. The following error message appears in Zigbee2mqtt or at the broker:

22m: Publish 'get' 'color' to
'OxXXXXXXXXXXXXX' failed: 'Error: ZCL
command OxXXXXXXXXXXXXX/2 lightingColorCtrl.read(["colorMode", " curr
entX", "currentY" "enhancedCurrentHue"," currentSaturation", "colorTemperature"], {"timeout":10000,"disableResponse":false
"disableRecovery":false,"disableDefaultR esponse":true, "direction":0,"reservedBits
":O,"writeUndiv":false}) failed (Status
'UNSUPPORTED_ATTRIBUTE")'

I hope someone can help me, please. It's really unfortunate that it isn't working right now.

### What did you expect to happen?

It should change the color.

### Zigbee2MQTT version

1.40.0-dev commit: 0d46354

### Adapter firmware version

7.3.1.0 build 0

### Adapter

Sonoff ZBDongle-E

### Setup

Z2M as a Add-on in Home Assistant

joe-peet, 2024-09-07:
Fun fact: https://github.com/Koenkk/zigbee2mqtt/issues/21626#issuecomment-2094641233
[geduxas](https://github.com/geduxas) found it working in the Home-Assistant UI. In the Zigbee2mqtt UI it's not working.

The two different approaches have the same structure:

```
Info 2024-09-07 10:05:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"brightness_rgb":254,"brightness_white":150,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_mode_rgb":"xy","color_mode_white":"color_temp","color_rgb":{"hue":233,"saturation":83,"x":0.198,"y":0.173},"color_temp":370,"color_temp_rgb":153,"color_temp_white":250,"color_white":{"x":0.3804,"y":0.3767},"device_temperature":50,"linkquality":255,"power_outage_count":27,"state":"OFF","state_rgb":"ON","state_white":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-09-07 10:05:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX/white', payload '{"brightness":150,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"state":"ON"}'
Info 2024-09-07 10:05:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX/rgb', payload '{"brightness":254,"color":{"hue":233,"saturation":83,"x":0.198,"y":0.173},"color_mode":"xy","color_temp":153,"state":"ON"}'
Info 2024-09-07 10:05:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"brightness_rgb":254,"brightness_white":150,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_mode_rgb":"xy","color_mode_white":"color_temp","color_rgb":{"hue":310,"saturation":66,"x":0.371,"y":0.239},"color_temp":370,"color_temp_rgb":370,"color_temp_white":250,"color_white":{"x":0.3804,"y":0.3767},"device_temperature":50,"linkquality":255,"power_outage_count":27,"state":"OFF","state_rgb":"ON","state_white":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-09-07 10:05:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX/white', payload '{"brightness":150,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"state":"ON"}'
Info 2024-09-07 10:05:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX/rgb', payload '{"brightness":254,"color":{"hue":310,"saturation":66,"x":0.371,"y":0.239},"color_mode":"xy","color_temp":370,"state":"ON"}'
Info 2024-09-07 10:05:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"brightness_rgb":254,"brightness_white":150,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_mode_rgb":"xy","color_mode_white":"color_temp","color_rgb":{"hue":310,"saturation":66,"x":0.371,"y":0.239},"color_temp":370,"color_temp_rgb":370,"color_temp_white":250,"color_white":{"x":0.3804,"y":0.3767},"device_temperature":50,"linkquality":255,"power_outage_count":27,"state":"OFF","state_rgb":"ON","state_white":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-09-07 10:05:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX/white', payload '{"brightness":150,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"state":"ON"}'
Info 2024-09-07 10:05:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX/rgb', payload '{"brightness":254,"color":{"hue":310,"saturation":66,"x":0.371,"y":0.239},"color_mode":"xy","color_temp":370,"state":"ON"}'
Info 2024-09-07 10:05:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX', payload '{"brightness_rgb":254,"brightness_white":150,"color":{"h":310,"hue":310,"s":66,"saturation":66,"x":0.371,"y":0.239},"color_mode":"xy","color_mode_rgb":"xy","color_mode_white":"color_temp","color_rgb":{"hue":310,"saturation":66,"x":0.371,"y":0.239},"color_temp":370,"color_temp_rgb":370,"color_temp_white":250,"color_white":{"x":0.3804,"y":0.3767},"device_temperature":50,"linkquality":255,"power_outage_count":27,"state":"OFF","state_rgb":"ON","state_white":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2024-09-07 10:05:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX/white', payload '{"brightness":150,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"state":"ON"}'
Info 2024-09-07 10:05:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xXXXXXXXXXXXXXXXX/rgb', payload '{"brightness":254,"color":{"hue":310,"saturation":66,"x":0.371,"y":0.239},"color_mode":"xy","color_temp":370,"state":"ON"}'
```

technikwandel, 2024-10-14:
That's a good question. It's important because, as others have mentioned, the first thing you often do after configuring your new device is to test it in Zigbee2MQTT. Later, you might test it through the Home Assistant UI. If it doesn't work in Zigbee2MQTT during your initial test, you typically won't try it in the Home Assistant configuration. Including this information in the device documentation would be very helpful.

closed 2024-12-22 by github-actions[bot]