Koenkk/zigbee2mqtt #23881: 1.40.0 - Unstable light entities control

FredyReady, 2024-09-07:
### What happened?

Upgraded from 1.39.0 to 1.40.0 and since then all my light entities are acting strange. The behaviour is random and so far I have not recognized any pattern.

Some light do not turn on on command, I receive error messages within Z2M UI. Some lights turn on but with different parameters (intensity, colours). Some lights do not response to any commands (On, off, change intensity or colour).

The issue seems to not differentiate between hardware versions as it occurs on light bulbs from IKEA and LED controllers from Gledopto (I do not own any other brands of lights).

It does not happen all the time but few hours spread across the day.

Restart some times helps and some times it does not.

It does not matter if I control the lights from dashboard, automation or directly from Z2M UI.

Managed to catch the errors when control of light group failed after timeout.

[image]
[image]

### What did you expect to happen?

Responsive behaviour of light entities.

### How to reproduce it (minimal and precise)

Update to 1.40.0. and try to control light entities - random behaviour.

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

20221226

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Add-on on Home Assistant OS on Intel NUC

kwlorenz, 2024-09-07:
I have had nearly the same problem with light entities (Philips Hue Lights, Philips Hue Smart Plug, Ikea lights) and version 1.40; not possible to control lights anymore due to timeout errors.
Unfortunately I don not have the error message any more, went back to 1.39.1 and had to restart mqtt-broker. Than everything turns normal

Koenkk, 2024-09-07:
It looks like your coordinator crashed, I recommend to update your coordinator to the latest version first (https://www.zigbee2mqtt.io/guide/adapters/zstack.html)

FredyReady, 2024-09-08:
Thanks Koenkk for taking a look at this much appreciated!
I am out of home for now but once I am back I will perform the update and see if it solves the issue.

I'll keep the thread open for others or until I can confirm the solution viability, then I'll close it.

Thanks ✌️

kwlorenz, 2024-09-08:
Updated the stick to the newest firmware yesterday and installed 1.40.1 afterwards. No problems in the moment.
Thanks very much...

closed 2024-10-11 by FredyReady