Koenkk/zigbee2mqtt #23883: HTTPS not working

ZTHawk, 2024-09-07:
### What happened?

I have Zigbee2MQTT as a HomeAssistant addon.

* My settings:
[image]

* My 'configuration.yaml' (located in HA/zigbee2mqtt) looks like this:
```
frontend:
  port: 8099
  ssl_cert: /ssl/zigbee2mqtt.crt
  ssl_key: /ssl/zigbee2mqtt.key
```
The SSL files are additionally located here: HA/zigbee2mqtt/ssl/

I get is this error "SSL_ERROR_RX_RECORD_TOO_LONG" when I try to access "https://<HA-IP>:8099".
The SSL certificate is correctly configured.

I tried for "ssl_cert" to use a full chain certificate as well.

PS:
Using an invalid path (eg: /ssl/zigbee2mqtt.crt.invalid) does not show any errors in log.

### What did you expect to happen?

Accessing website "https://<HA-IP>:8099" shows the Zigbee2MQTT frontend.

### Zigbee2MQTT version

1.40.0

### Adapter firmware version

EmberZNet 7.4.2 [GA]

### Adapter

SONOFF Dongle Plus V2

### Setup

HA Addon and HA in proxmox

MiguelCosta, 2024-09-07:
If your Home Assistant is using https and if you are using Zigbee2MQTT as a HomeAssistant addon, I think you should use the follow URL to access to the Z2M: 

`https://YourDns.com/45df7312_zigbee2mqtt`

(note: you should replace https://YourDns.com/ by the correct value)

ZTHawk, 2024-09-08:
This is of course working, because this is using the HTTPS connection of HA.
In fact you do not need to spedify any SSL cert/key in Zigbee2MQTT for that to be working.
I want to have Zigbee2MQTT to use its own HTTPS connection.

What I mean exactly, is that for example Nodered can be accessed via https://myDNS/a0d7b954_nodered but also via https://192.168.0.102:1880 (192.168.0.102 = HA IP)

So I want Zigbee2MQTT to be accessible via https://192.168.0.102:8099 (192.168.0.102 = HA IP). This is what I understand from the documentation.

ZTHawk, 2024-12-12:
Still applicable (v1.42.0)

ZTHawk, 2025-02-02:
still applicable (v2.1.0)
I think it is related to #23648

ZTHawk, 2025-04-04:
still applicable

ZTHawk, 2025-04-06:
Update v2.2.1 (maybe 2.2.0) fixed this issue.
Bug created a new one with ingress.

Site cannot be displayed. I get "502: Bad Gateway" and a popup:
```
Zigbee2MQTT

The add-on seems to not be ready, it might still be starting. Do you want to try again?
```

I am expecting the site to show up.

Reverting to 2.1.3-1 as the ingress feature is very important.

ZTHawk, 2025-05-02:
The actuall issue is fixed. The new issue discrived is now here: #27297

closed 2025-05-02 by ZTHawk