Koenkk/zigbee2mqtt #23887: TS011F_plug_3 not reporting power in the last versions (issue started in the version since 1.39.1)

MiguelCosta, 2024-09-07:
### What happened?

My Blitzwolf BW-SHP13 socket has power measurement, but in the latest versions it has stopped working. It's always the last value that has been reported or if you re-pair, the value changes to null.
The kwh sensor is working, but the current energy (w) is not reported.

The issue started after upgrade to 1.39.1 version.

[image]

[image]

[image]

### What did you expect to happen?

The sensors `Current`, `Power` and `Voltage` should be reported.

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20210708

### Adapter

Sonoff ZBDongle-P

### Setup

Docker

MiguelCosta, 2024-09-07:
More users with the same issue: https://github.com/Koenkk/zigbee2mqtt/issues/22673#issuecomment-2286743440

Koenkk, 2024-09-08:
- Could you provide a screenshot of the device about page?
- Could you provide the debug log when starting z2m + 5 mins?
See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

MiguelCosta, 2024-09-09:
@Koenkk the logs have this error, I think it is related:

```
[2024-09-09 21:52:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-09-09 21:52:28] debug: 	zhc:utils: Polling 'measurement' for '0x84ba20fffe71029d' at an interval of 10
[2024-09-09 21:52:28] debug: 	zhc:utils: Polling 'measurement' for '0x540f57fffe8ebd91' at an interval of 60
[2024-09-09 21:52:28] debug: 	zhc:utils: Polling 'measurement' for '0x540f57fffe8eb862' at an interval of 10
[2024-09-09 21:52:28] error: 	z2m: Failed to call 'OnEvent' 'start' (Error: 'measurement_poll_interval' is not a number, got string ()
    at toNumber (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:634:15)
    at Object.onEventPoll (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:45:25)
    at Object.onEventMeasurementPoll (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:186:11)
    at Object.onEvent (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:5105:56)
    at OnEvent.callOnEvent (/app/lib/extension/onEvent.ts:48:37)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at OnEvent.start (/app/lib/extension/onEvent.ts:12:13)
    at Controller.callExtensions (/app/lib/controller.ts:377:17)
    at Controller.start (/app/lib/controller.ts:196:9)
    at start (/app/index.js:154:5))
```

All the logs here: 
[log.log](https://github.com/user-attachments/files/16936534/log.log)

> Could you provide a screenshot of the device about page?

[image]

MiguelCosta, 2024-09-09:
@Koenkk in the in the version `1.39.1`, the `zigbee-herdsman-converters` lib was upgraded from `19.62.0` to `19.72.0`  version and maybe the bug started with this change: https://github.com/Koenkk/zigbee-herdsman-converters/commit/6f5707bff5f79bc8c6c7dc1b78c9b4d8a4d0f607

can you help?

MiguelCosta, 2024-09-09:
Mitigated 💪 

This field was empty, I added a value value here to work and now it is working 🙏

[image]

MiguelCosta, 2024-09-09:
@Koenkk I think this issue occurs because the UI is creating a configuration with a empty string

[image]

hotbirdx, 2024-09-11:
@Koenkk I can confirm that the issue is the same for the TS0121 / BlitzWolf BW-SHP13 device. If I change in the GUI to a value eg. 60 it works.

Koenkk, 2024-09-12:
Fixed the issue, when set to `""`, the default poll interval will be used now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-12 by Koenkk